espressif/esp-idf #11488: Boot from OTA_1`with image size cross over 16M boundary

chewhs00, 2023-05-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-439-gc25d92074e

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board with esp32-s3-wroom-2-n32r8v

### Power Supply used.

USB

### What is the expected behavior?

Given my app image size of 2531728, I expect it to boot from OTA_1 when I use otatool.py to switch it to that partition. The partition table I have:
```
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvsfs            WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 ota_0            OTA app          00 10 00010000 00e00000
I (97) boot:  4 ota_1            OTA app          00 11 00e10000 00e00000
I (104) boot:  5 system           Unknown data     01 82 01c10000 00040000
I (112) boot:  6 nvs              WiFi data        01 02 01c50000 000b0000
I (119) boot:  7 user             Unknown data     01 82 01d00000 00300000
I (127) boot: End of partition table
```

### What is the actual behavior?

Instead it boots from OTA_0. If the image size is below 2M to fit in 16M boundary, then it boots fine from OTA_1
```
I (32) boot: ESP-IDF v4.4.4-439-gc25d92074e 2nd stage bootloader
I (33) boot: compile time 16:37:39
I (33) boot: chip revision: v0.1
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : SLOW READ
I (46) boot.esp32s3: SPI Flash Size : 32MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvsfs            WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 ota_0            OTA app          00 10 00010000 00e00000
I (97) boot:  4 ota_1            OTA app          00 11 00e10000 00e00000
I (104) boot:  5 system           Unknown data     01 82 01c10000 00040000
I (112) boot:  6 nvs              WiFi data        01 02 01c50000 000b0000
I (119) boot:  7 user             Unknown data     01 82 01d00000 00300000
I (127) boot: End of partition table
I (131) esp_image: segment 0: paddr=00e10020 vaddr=3c0c0020 size=1aa27ch (1745532) map
I (554) esp_image: segment 1: paddr=00fba2a4 vaddr=3fc970e0 size=051f0h ( 20976) load
I (560) esp_image: segment 2: paddr=00fbf49c vaddr=40374000 size=00b7ch (  2940) load
I (561) esp_image: segment 3: paddr=00fc0020 vaddr=42000020 size=b4c34h (740404) map
E (744) esp_image: invalid segment length 0x30a10110
E (744) boot: OTA app partition slot 1 is not bootable
I (745) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=1a99b4h (1743284) map
I (1167) esp_image: segment 1: paddr=001b99dc vaddr=3fc94f20 size=049dch ( 18908) load
I (1172) esp_image: segment 2: paddr=001be3c0 vaddr=40374000 size=01c58h (  7256) load
I (1175) esp_image: segment 3: paddr=001c0020 vaddr=42000020 size=aae80h (700032) map
I (1348) esp_image: segment 4: paddr=0026aea8 vaddr=40375c58 size=0f2c4h ( 62148) load
I (1372) boot: Loaded app from partition at offset 0x10000
I (1372) boot: Disabling RNG early entropy source...
I (1385) opi psram: vendor id : 0x0d (AP)
I (1385) opi psram: dev id    : 0x02 (generation 3)
I (1385) opi psram: density   : 0x03 (64 Mbit)
I (1389) opi psram: good-die  : 0x01 (Pass)
I (1394) opi psram: Latency   : 0x01 (Fixed)
I (1399) opi psram: VCC       : 0x00 (1.8V)
I (1404) opi psram: SRF       : 0x01 (Fast Refresh)
I (1409) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (1415) opi psram: BurstLen  : 0x01 (32 Byte)
I (1420) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (1426) opi psram: DriveStrength: 0x00 (1/1)
I (1432) spiram: Found 64MBit SPI RAM device
I (1436) spiram: SPI RAM mode: sram 80m
I (1440) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (1447) cpu_start: Pro cpu up.
I (1451) cpu_start: Starting app cpu, entry point is 0x40375470

```

### Steps to reproduce.

1. Create a partition table with OTA_1 near 16M address
2. Create and build an application image with size that will cross 16M address boundary when flashing it at OTA_1 using esptool 4.5.1
3. Use otatool to switch booting from  OTA_1
.

mythbuster5, 2023-05-26:
This is because cache cannot map over 16MB address on v4.4, it will support this on 5.1

mythbuster5, 2023-07-19:
related docs:  https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html#bit-address-flash-chips

closed 2024-12-17 by Alvin1Zhang