espressif/esp-idf #11489: Bluetooth HFP client occasionally crashes (StoreProhibited) trying to connect

danjulio, 2023-05-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Attempted Bluetooth HFP client connection should not fail regardless of the success or not of the connection.

### What is the actual behavior?

Application has been paired and will successfully connect to a phone (iPhone).  Application attempt to connect using esp_hf_client_connect with a known good address every 60 seconds when disconnected (e.g. phone is away).  Most of the time the connection attempt fails normally (CONNECTION_STATE_EVT returns disconnected).  Occasionally the application crashes after initiating the connection attempt.

I have tried increasing the stack size for both Bluedroid and the callback mechanism (+512 bytes each).  I have also tried to be careful to not block in the callbacks.

### Steps to reproduce.

The code is tied to custom hardware but I am happy to share parts of it and possibly see if I can make your hfp client demo fail this way too if you think that would help.

1. Pair phone with application
2. Run application with phone far enough away so that connection attempts fail

Attached log file shows two crashes in series.  There is an initial connection to the phone in the first run and then the phone was removed and the application attempted to reconnect every 60 seconds many times before the crash.  In the second run the phone never successfully connected.

Sometimes the crash occurs on the first attempt to connect (if phone is not present).

### Debug Logs.

```plain
I (27) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (27) boot: compile time 12:15:36
I (27) boot: chip revision: v3.0
I (30) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (40) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 16MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00300000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=575fch (357884) map
I (199) esp_image: segment 1: paddr=00067624 vaddr=3ffbdb60 size=04a58h ( 19032) load
[... 487 lines cut ...]
I ran xtensa-esp32-elf-addr2line on one of the backtraces to get this cleaner output:

```
hci_rd_rem_ext_feats_cmd_handler
??:?
??
??:0
ke_task_schedule
ke_task.c:?
??
??:0
??
??:0
r_rw_schedule
??:?
btdm_controller_task
??:?
vPortTaskWrapper
``` 

Also, before I did some cleanup on the bluetooth management code I did occasionally also see this message in the log file right before the crash.   But I have not seen it since cleaning up the code.

```
E (10836) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x41c
```

xiongweichao, 2023-05-24:
Hi @danjulio ,

This [commit](https://github.com/espressif/esp-idf/commit/f9bfdd0b1e51ea36fe3213e1c6fbdd4396bb035a) has fixed the issue.

Thanks

xiongweichao, 2023-05-25:
Hi @danjulio ,

May I ask whether you have tested it with release/v5.0? Please let me know if the problem still exists.

Thanks

danjulio, 2023-05-25:
Hello @xiongweichao ,

Thank you for getting back to me so quickly.  From the commit I saw that you applied it to v4.4 so I updated my local copy to include the fix.   I have been testing that today and will test more tomorrow.  I do not know if this version number is associated with the new library you mentioned or not but this is what my code now displays.

```I (684) BTDM_INIT: BT controller compile version [97e0759]```

Please let me know if I am in error about the fix being in the latest v4.4.

I will have to change some other items in the project to move to v5 so that may take a bit if necessary.

Thanks again.

xiongweichao, 2023-05-26:
Hi @danjulio ,

`I (684) BTDM_INIT: BT controller compile version [97e0759]`
This library already contains the fix.

Thanks

danjulio, 2023-05-26:
Thank you for confirming.

I have tested for the past 24 hours in various conditions with no failures.  I think this issue is resolved with the new library.  Thank you.

closed 2023-05-26 by danjulio