espressif/esp-idf #11491: SDKCONFIG_DEFAULTS options doesn't seem to get applied to component

hurricanefrog, 2023-05-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Not applicable

### Power Supply used.

USB

### What is the expected behavior?

I have a (empty, configuration-only) component which is only there to specify the hardware revision used. This is a component so other components can depend on it. (This appears cleaner to me than to specify the hardware in the main component).

I have several default sdkconfig files which only specify the revision used.

In the CI pipeline, I loop over the hardware configurations/sdkconfig default files to build each config in a separate directory.

I expect that when I (re)configure the project using a specific `SDKCONFIG_DEFAULTS`, the options will be reflected in the generated sdkconfig.h.

### What is the actual behavior?

The option is not reflected in sdkconfig.h. Only the default option is present.

This might be because the option is in a (sub)component?

### Steps to reproduce.

1. Create an empy project `idf.py  create-project minimal`
2. Create a config-only component:
```
#components/hw_config/CMakeLists.txt
## config-only
idf_component_register()
---
#components/hw_config/Kconfig.projbuild
menu "Hardware Config"

    choice HARDWARE_REVISION
        prompt "ESP PCB hardware revision"

        config HARDWARE_REVISION_5_0_0
            bool "Hardware 5.0.0"

        config HARDWARE_REVISION_5_0_1
            bool "Hardware 5.0.1"

        config HARDWARE_REVISION_5_0_2
            bool "Hardware 5.0.2"
    endchoice

    config HARDWARE_REVISION
        hex
        default 0x050000 if HARDWARE_REVISION_5_0_0
        default 0x050001 if HARDWARE_REVISION_5_0_1
        default 0x050002 if HARDWARE_REVISION_5_0_2
endmenu
```

4. Generate sdkconfig default files:
```
# grep REVISION def_*
def_5_0_0:CONFIG_HARDWARE_REVISION_5_0_0=y
def_5_0_1:CONFIG_HARDWARE_REVISION_5_0_1=y
def_5_0_2:CONFIG_HARDWARE_REVISION_5_0_2=y
```
5. Try to configure each config with this build script.
```
# build.sh
for rev in 5_0_0 5_0_1 5_0_2; do {
  builddir=hw_${rev};
  export SDKCONFIG_DEFAULTS="${PWD}/def_${rev}";
  idf.py -B ${builddir} reconfigure;
  grep -q CONFIG_HARDWARE_REVISION_${rev} ${builddir}/config/sdkconfig.h || { echo "idf.py failed configuring correctly" >&2; exit 1; };
}; done
```

The `grep` should succeed if the option is applied correctly. However, only the first iteration succeeds. `hw_5_0_1/config/sdkconfig.h` also contains the `CONFIG_HARDWARE_REVISION_5_0_0` option.

### Debug Logs.

```plain
n/a
```

### More Information.

Am I using `SDKCONFIG_DEFAULTS` wrong?

igrr, 2023-05-24:
What you are trying to do should work, however you need to use a different `sdkconfig` file for each build. `sdkconfig` is only re-generated if it doesn't exist, therefore as you are reusing the `sdkconfig` file between builds you are getting the same settings.

Changing the `idf.py` call like this is one way to make it work:
```
idf.py -B ${builddir} -DSDKCONFIG=${builddir}/sdkconfig reconfigure;
```
This places `sdkconfig` file into the build directory, so each build has its own unique `sdkconfig` file.

Or, simply delete `sdkconfig` file after each build.

More information can be found in the README of this example: https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config

Alternatively, you may check out https://github.com/espressif/idf-build-apps — it is a tool which can simplify the process of building one app in a variety of configurations.

hurricanefrog, 2023-05-25:
Ah, thanks. 👍  Should have realized that, it makes sense now that you point it out.

Build works for me now.

closed 2023-05-25 by hurricanefrog