espressif/esp-idf #11492: NimBLE cannot send notification to multiple devices

ascillato, 2023-05-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Release v5.0.2

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

USB

### What is the expected behavior?

esp-idf NimBLE accepts multiple connections which can send commands to the ESP32, but esp-idf crashes when trying to send data to multiple devices. The API in the esp-idf docs, accepts a handle number which is `0` for the first devices connected (that is working fine), 1 for the second device, and so on. So, when issuing the command `ble_gatts_notify_custom(ble_conn_hdl, notify_char_attr_hdl, om)` it crashes when trying to send data to `ble_conn_hdl > 0`

The bug seems to be in the `esp_vhci_host_send_packet` at `.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:1242`

This was already reported in https://github.com/espressif/esp-idf/issues/8945 but never fixed.

Is there any workaround to this bug?

Thanks a lot for your time.

### What is the actual behavior?

Actual behaviour is crashing when trying to send data to any BLE device that has a handle different than zero

### Steps to reproduce.

The steps and code were already reported in the closed issue: https://github.com/espressif/esp-idf/issues/8945

### Debug Logs.

```plain
The debug logs were already reported in the closed issue: https://github.com/espressif/esp-idf/issues/8945
```

chipweinberger, 2023-05-25:
> it crashes

you should include a backtrace in your issue report

chipweinberger, 2023-05-25:
> "So currently, nimble doesn't support notification from one server to multiple clients. This will require major code changes, we will take this as feature request as it will take time."

^ @IshaESP 

Oh no. I had no idea about this restriction!! This is a huge limitation. How much work will it take to add support?

If we cannot add support, I may need to switch to bluedroid. I am not sure if I have enough spare internal ram, though.

IshaESP, 2023-05-25:
Hi @chipweinberger ,
You can refer to the examples/bluetooth/nimble/ble_spp/spp_server example. We have demonstrated sending of notifications to multiple connected clients.

If this doesn't help, please send the debug logs.
To enable debug logs:
1. idf.py menuconfig
2. Component config → Bluetooth → NimBLE Options → NimBLE Host log verbosity → Debug logs
3. Component config → Log output → Default log verbosity → Debug

Thank You.

chipweinberger, 2023-05-25:
Thats great! When did ESP-IDF add support for sending of notifications to multiple connected clients? In v5.0?

ascillato, 2023-05-25:
I have been testing the examples/bluetooth/nimble/ble_spp/spp_server and I'm having the same issue. As soon as I sent data from the ESP32 to a more than 1 device it crashes with:

```
vhci.c 134
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008d131  PS      : 0x00060c35  A0      : 0x8008d330  A1      : 0x3ffbffe0  
A2      : 0x00000002  A3      : 0x3ffb54d4  A4      : 0x40017620  A5      : 0x3ffb8140  
A6      : 0x3ffd88f4  A7      : 0x00000013  A8      : 0x8008d131  A9      : 0x3ffbff50  
A10     : 0x0000000b  A11     : 0x0000000b  A12     : 0x0000000c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x400f86d7  EPC2    : 0x00000000  EPC3    : 0x4008d131  EPC4    : 0x4008d131

Backtrace: 0x4008d12e:0x3ffbffe0 0x4008d32d:0x3ffc0000 0x4013697b:0x3ffc0020 0x4008d440:0x3ffc0040 0x400848c5:0x3ffc0060 0x40081caa:0x3ffe1130 0x4008d0b9:0x3ffe1150 0x401477a8:0x3ffe1170 0x400e6972:0x3ffe11a0 0x400e62fd:0x3ffe11c0 0x400dee41:0x3ffe12e0 0x400e4da9:0x3ffe1300 0x400e4df5:0x3ffe1330 0x400e38fc:0x3ffe1350 0x400e1299:0x3ffe1380 0x400e1119:0x3ffe13b0 0x400e04c9:0x3ffe13e0 0x400d344d:0x3ffe1400 0x400d663d:0x3ffe1430 0x400d6857:0x3ffe1490

  #0  0x4008d12e:0x3ffbffe0 in vhci_recv_end at vhci.c:?
  #1  0x4008d32d:0x3ffc0000 in r_vhci_isr at ??:?
  #2  0x4013697b:0x3ffc0020 in r_eif_isr at ??:?
  #3  0x4008d440:0x3ffc0040 in r_eif_isr_wrapper at intc.c:?
  #4  0x400848c5:0x3ffc0060 in _xt_lowint1 at /home/ascillato/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118
  #5  0x40081caa:0x3ffe1130 in cause_sw_intr at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:883
      (inlined by) cause_sw_intr_to_core_wrapper at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:894
  #6  0x4008d0b9:0x3ffe1150 in vhci_set_interrupt at vhci.c:?
  #7  0x401477a8:0x3ffe1170 in API_vhci_host_send_packet at ??:?
  #8  0x400e6972:0x3ffe11a0 in esp_vhci_host_send_packet at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:1242
  #9  0x400e62fd:0x3ffe11c0 in ble_hci_trans_hs_acl_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:125
  #10 0x400dee41:0x3ffe12e0 in ble_hs_tx_data at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:740
  #11 0x400e4da9:0x3ffe1300 in ble_hs_hci_acl_tx_now at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:551
  #12 0x400e4df5:0x3ffe1330 in ble_hs_hci_acl_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:609
  #13 0x400e38fc:0x3ffe1350 in ble_l2cap_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap.c:453
  #14 0x400e1299:0x3ffe1380 in ble_att_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_cmd.c:76
  #15 0x400e1119:0x3ffe13b0 in ble_att_clt_tx_notify at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:905
  #16 0x400e04c9:0x3ffe13e0 in ble_gatts_notify_custom at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4192
  #17 0x400d344d:0x3ffe1400 in sendBleString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at src/ble_proxy.cpp:451
```

ascillato, 2023-05-25:
The same but with logs to Debug level

```
D (63176) NimBLE: ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=11 data=
D (63176) NimBLE: 0x07 
D (63176) NimBLE: 0x00 
D (63186) NimBLE: 0x04 m
D (63186) NimBLE: 0x00 
D (63186) NimBLE: 0x10 m
D (63186) NimBLE: 0x0a 
D (63196) NimBLE: 0x00 m
D (63196) NimBLE: 0xff 
D (63196) NimBLE: 0xff 0m
D (63196) NimBLE: 0x00 
D (63206) NimBLE: 0x28 0m
D (63206) NimBLE: 

D (63206) NimBLE: host tx hci data; handle=1 length=26

D (63216) NimBLE: ble_hs_hci_acl_tx(): 
D (63216) NimBLE: 0x01 [0m
D (63216) NimBLE: 0x00 
D (63216) NimBLE: 0x1a [0m
D (63226) NimBLE: 0x00 
D (63226) NimBLE: 0x16 [0m
D (63226) NimBLE: 0x00 
D (63236) NimBLE: 0x04 [0m
D (63236) NimBLE: 0x00 
D (63236) NimBLE: 0x11 [0m
D (63236) NimBLE: 0x14 
D (63246) NimBLE: 0x0a 
D (63246) NimBLE: 0x00 
D (63246) NimBLE: 0xff 
D (63246) NimBLE: 0xff 
D (63256) NimBLE: 0x9e 
D (63256) NimBLE: 0xca 
D (63256) NimBLE: 0xdc 
D (63256) NimBLE: 0x24 m
D (63266) NimBLE: 0x0e 
D (63266) NimBLE: 0xe5 m
D (63266) NimBLE: 0xa9 
D (63266) NimBLE: 0xe0 m
D (63276) NimBLE: 0x93 
D (63276) NimBLE: 0xf3 0m
D (63276) NimBLE: 0xa3 
D (63276) NimBLE: 0xb5 0m
D (63286) NimBLE: 0x01 
D (63286) NimBLE: 0x00 0m
D (63286) NimBLE: 0x40 
D (63286) NimBLE: 0x6e [0m
D (63296) NimBLE: 

32mI (63296) NimBLE: GATT procedure initiated: notify; 0m
I (63296) NimBLE: att_handle=14
m
D (63306) NimBLE: host tx hci data; handle=0 length=141

D (63306) NimBLE: ble_hs_hci_acl_tx(): 
D (63316) NimBLE: 0x00 
D (63316) NimBLE: 0x00 
D (63316) NimBLE: 0x8d m
D (63316) NimBLE: 0x00 
D (63326) NimBLE: 0x89 m
D (63326) NimBLE: 0x00 
D (63326) NimBLE: 0x04 0m
D (63326) NimBLE: 0x00 
D (63336) NimBLE: 0x1b 0m
D (63336) NimBLE: 0x0e 
D (63336) NimBLE: 0x00 0m
D (63336) NimBLE: 0x7b 
[... 164 lines cut ...]
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x400ff43b  EPC2    : 0x00000000  EPC3    : 0x4008d649  EPC4    : 0x4008d649

Backtrace: 0x4008d646:0x3ffc00d0 0x4008d845:0x3ffc00f0 0x4013fb4f:0x3ffc0110 0x4008d958:0x3ffc0130 0x40084d25:0x3ffc0150 0x40081dbe:0x3ffe1110 0x4008d5d1:0x3ffe1130 0x401509a0:0x3ffe1150 0x400e7c96:0x3ffe1180 0x400e75c9:0x3ffe11a0 0x400dfc89:0x3ffe12c0 0x400e5f29:0x3ffe12e0 0x400e5f75:0x3ffe1330 0x400e4a1c:0x3ffe1350 0x400e216d:0x3ffe1380 0x400e1fed:0x3ffe13b0 0x400e1383:0x3ffe13e0 0x400d40b1:0x3ffe1400 0x400d729d:0x3ffe1430 0x400d74b7:0x3ffe1490

  #0  0x4008d646:0x3ffc00d0 in vhci_recv_end at vhci.c:?
  #1  0x4008d845:0x3ffc00f0 in r_vhci_isr at ??:?
  #2  0x4013fb4f:0x3ffc0110 in r_eif_isr at ??:?
  #3  0x4008d958:0x3ffc0130 in r_eif_isr_wrapper at intc.c:?
  #4  0x40084d25:0x3ffc0150 in _xt_lowint1 at /home/ascillato/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118
  #5  0x40081dbe:0x3ffe1110 in cause_sw_intr at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:883
      (inlined by) cause_sw_intr_to_core_wrapper at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:894
  #6  0x4008d5d1:0x3ffe1130 in vhci_set_interrupt at vhci.c:?
  #7  0x401509a0:0x3ffe1150 in API_vhci_host_send_packet at ??:?
  #8  0x400e7c96:0x3ffe1180 in esp_vhci_host_send_packet at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:1242
  #9  0x400e75c9:0x3ffe11a0 in ble_hci_trans_hs_acl_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:125
  #10 0x400dfc89:0x3ffe12c0 in ble_hs_tx_data at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:740
  #11 0x400e5f29:0x3ffe12e0 in ble_hs_hci_acl_tx_now at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:551
  #12 0x400e5f75:0x3ffe1330 in ble_hs_hci_acl_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:609
  #13 0x400e4a1c:0x3ffe1350 in ble_l2cap_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap.c:453
  #14 0x400e216d:0x3ffe1380 in ble_att_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_cmd.c:76
  #15 0x400e1fed:0x3ffe13b0 in ble_att_clt_tx_notify at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:905
  #16 0x400e1383:0x3ffe13e0 in ble_gatts_notify_custom at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4192
```

ascillato, 2023-05-25:
These logs may be more useful.

First chunk is with only one device (send and receive works fine)
Second chunk is when the second device connects and crash when the ESP sends data to it.

```
One device and sending Data:

I (73696) NimBLE: GATT procedure initiated: notify; 
I (73696) NimBLE: att_handle=14

D (73696) NimBLE: host tx hci data; handle=0 length=91

D (73696) NimBLE: ble_hs_hci_acl_tx():
D (73696) NimBLE: 0x00 
D (73706) NimBLE: 0x00
D (73706) NimBLE: 0x5b 
D (73706) NimBLE: 0x00 
D (73706) NimBLE: 0x57 
D (73716) NimBLE: 0x00
D (73716) NimBLE: 0x04 
D (73716) NimBLE: 0x00
D (73716) NimBLE: 0x1b 
D (73726) NimBLE: 0x0e 
D (73726) NimBLE: 0x00 
D (73726) NimBLE: 0x7b 
D (73726) NimBLE: 0x22 
D (73736) NimBLE: 0x74 
D (73736) NimBLE: 0x79 
D (73736) NimBLE: 0x70 
D (73736) NimBLE: 0x65
D (73746) NimBLE: 0x22 
D (73746) NimBLE: 0x3a
D (73746) NimBLE: 0x22 
D (73746) NimBLE: 0x74
D (73756) NimBLE: 0x65 
D (73756) NimBLE: 0x6d
D (73756) NimBLE: 0x70 
D (73756) NimBLE: 0x22
D (73766) NimBLE: 0x2c
D (73766) NimBLE: 0x22
D (73766) NimBLE: 0x69 
D (73766) NimBLE: 0x64
D (73776) NimBLE: 0x22 
D (73776) NimBLE: 0x3a
D (73776) NimBLE: 0x22 
D (73776) NimBLE: 0x37
D (73786) NimBLE: 0x43 
D (73786) NimBLE: 0x39
D (73786) NimBLE: 0x45 
D (73786) NimBLE: 0x42 
D (73796) NimBLE: 0x44 
D (73796) NimBLE: 0x33 
D (73796) NimBLE: 0x32 
D (73796) NimBLE: 0x32 
D (73806) NimBLE: 0x39 
D (73806) NimBLE: 0x42 
D (73806) NimBLE: 0x43 
D (73806) NimBLE: 0x22 
D (73816) NimBLE: 0x2c
D (73816) NimBLE: 0x22 
D (73816) NimBLE: 0x6f 
D (73816) NimBLE: 0x75 
D (73826) NimBLE: 0x74 
D (73826) NimBLE: 0x6c 
D (73826) NimBLE: 0x65 
D (73826) NimBLE: 0x74 
D (73836) NimBLE: 0x5f 
D (73836) NimBLE: 0x74 
D (73836) NimBLE: 0x65 
[... 893 lines cut ...]
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
Core  0 was running in ISR context:
EPC1    : 0x400ff0e3  EPC2    : 0x00000000  EPC3    : 0x4008d2cd  EPC4    : 0x4008d2cd

Backtrace: 0x4008d2ca:0x3ffc00d0 0x4008d4c9:0x3ffc00f0 0x4013f7f3:0x3ffc0110 0x4008d5dc:0x3ffc0130 0x400849a9:0x3ffc0150 0x40081d7e:0x3ffe1120 0x4008d255:0x3ffe1140 0x40150644:0x3ffe1160 0x400e7aca:0x3ffe1190 0x400e73fd:0x3ffe11b0 0x400dfabd:0x3ffe12d0 0x400e5d5d:0x3ffe12f0 0x400e5da9:0x3ffe1340 0x400e4850:0x3ffe1360 0x400e1fa1:0x3ffe1390 0x400e1e21:0x3ffe13c0 0x400e11b7:0x3ffe13f0 0x400d4081:0x3ffe1410 0x400d6fd5:0x3ffe1440 0x400d7513:0x3ffe1490

  #0  0x4008d2ca:0x3ffc00d0 in vhci_recv_end at vhci.c:?
  #1  0x4008d4c9:0x3ffc00f0 in r_vhci_isr at ??:?
  #2  0x4013f7f3:0x3ffc0110 in r_eif_isr at ??:?
  #3  0x4008d5dc:0x3ffc0130 in r_eif_isr_wrapper at intc.c:?
  #4  0x400849a9:0x3ffc0150 in _xt_lowint1 at /home/ascillato/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118
  #5  0x40081d7e:0x3ffe1120 in cause_sw_intr at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:883
      (inlined by) cause_sw_intr_to_core_wrapper at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:894
  #6  0x4008d255:0x3ffe1140 in vhci_set_interrupt at vhci.c:?
  #7  0x40150644:0x3ffe1160 in API_vhci_host_send_packet at ??:?
  #8  0x400e7aca:0x3ffe1190 in esp_vhci_host_send_packet at /home/ascillato/.platformio/packages/framework-espidf/components/bt/controller/esp32/bt.c:1242
  #9  0x400e73fd:0x3ffe11b0 in ble_hci_trans_hs_acl_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c:125
  #10 0x400dfabd:0x3ffe12d0 in ble_hs_tx_data at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:740
  #11 0x400e5d5d:0x3ffe12f0 in ble_hs_hci_acl_tx_now at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:551
  #12 0x400e5da9:0x3ffe1340 in ble_hs_hci_acl_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci.c:609
  #13 0x400e4850:0x3ffe1360 in ble_l2cap_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap.c:453
  #14 0x400e1fa1:0x3ffe1390 in ble_att_tx at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_cmd.c:76
  #15 0x400e1e21:0x3ffe13c0 in ble_att_clt_tx_notify at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_att_clt.c:905
  #16 0x400e11b7:0x3ffe13f0 in ble_gatts_notify_custom at /home/ascillato/.platformio/packages/framework-espidf/components/bt/host/nimble/nimble/nimble/host/src/ble_gattc.c:4192
```

ascillato, 2023-05-25:
@IshaESP

Hi, thanks for your responses. From the previous logs, as soon as `handle` is greater than zero, `ble_gatts_notify_custom` crashes. Is there any workaround to this?

ascillato, 2023-05-25:
I made several tests trying to find a workaround to this issue. Something interesting is that if I change the code that sends data to devices to be:

```cpp
        for (int i = 0; i <= CONFIG_BT_NIMBLE_MAX_CONNECTIONS; i++) {
            if (connection_handle[i] != -1) {
                err = ble_gatts_notify_custom(connection_handle[i], ble_spp_svc_gatt_read_val_handle, om);
                break;
            }
        }
```

So, it sends data only to the first device connected, there is no crash. So, if I connect 2 devices, only the first one will receive data from the ESP32. Then If I disconnect the first device, the connection_handle will be 1, but it is going to be the only device connected and the data starts to be sent to that device without issue.

I tried to add a vtaskdelay between ble_gatts_notify_custom but it still crashes.

ascillato, 2023-05-25:
Found the issue!!!!

I changed the logic to not use a for loop with ble_gatts_notify_custom and send messages to different devices in different moments and it works perfectly.

Closing as resolved. No issue on the SDK!!!!

closed 2023-05-25 by ascillato

chipweinberger, 2023-05-26:
@ascillato please can you share your code change? thanks

ascillato, 2023-05-27:
I just made a custom routine which takes the handle as parameter, sends the message and exit without trying to do a for loop over all connected devices. The main loop in is the one in charge of sending to all connected devices. In my case, calling `ble_gatts_notify_custom(connection_handle[i], ble_spp_svc_gatt_read_val_handle, om);` one after another was not working in my program. Hope this helps in your program.

chipweinberger, 2023-05-27:
If you are not looping through the devices, how are you sending to multiple devices? Don't you need to loop through them?

ascillato, 2023-05-30:
Hi, sorry for not explaining correctly. I'm not using a for loop. My program sends different messages to different clients. I hope that when you try the code with latest SDK you won't have issues. It is working fine sending notifications to multiple cennected devices.