espressif/esp-idf #11496: idf_component_set_property with EMBED_TXTFILES can not use?

WangShuoran, 2023-05-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

When I use `target_add_binary_data(mqtt_ssl.elf "main/mqtt_eclipseprojects_io.pem" TEXT)` at project cmakelists.txt, the project is build ok.

### What is the actual behavior?

But I what to use components cmakelists.txt to tell builder with those code:
`idf_component_set_property(${COMPONENT_NAME} EMBED_TXTFILES ${project_dir}/main/mqtt_eclipseprojects_io.pem)`
[image]

I find an error:
[image]

I what to use idf_component_set_property set components cmakelists.txt to use pem. What I should do?

### Steps to reproduce.

use example mqtt/ssl and replace cmakelists.txt content with behevior.

### Debug Logs.

```plain
Successfully created esp32s3 image.
Generated C:/Espressif/frameworks/esp-idf-v4.4.3/examples/protocols/mqtt/ssl/build/bootloader/bootloader.bin
[102/102] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.3\examples\protocols\mqtt\ssl\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.3/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Espressif/frameworks/esp-idf-v4.4.3/examples/protocols/mqtt/ssl/build/bootloader/bootloader.bin"
Bootloader binary size 0x5100 bytes. 0x2f00 bytes (37%) free.
[1060/1062] Linking CXX executable mqtt_ssl.elf
FAILED: mqtt_ssl.elf 
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\mqtt_ssl.elf.rsp -o mqtt_ssl.elf  && cd ."
c:/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/main/libmain.a(app_main.c.obj):(.literal.mqtt_app_start+0x4): undefined reference to `_binary_mqtt_eclipseprojects_io_pem_start'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

WangShuoran, 2023-05-26:
When I check OTA example, I find in simple_ota_example use cmakelists:
```
idf_build_get_property(project_dir PROJECT_DIR)
idf_component_register(SRCS "simple_ota_example.c"
                    INCLUDE_DIRS "."
                    EMBED_TXTFILES ${project_dir}/server_certs/ca_cert.pem)
```

but I try to use last code in components cmakelists:
```
idf_component_register(SRCS "src/xxxx.c"
                            "src/xxx.c"
                    INCLUDE_DIRS "include"
                    REQUIRES mqtt xxx xxx)

idf_build_get_property(project_dir PROJECT_DIR)
idf_component_set_property(${COMPONENT_NAME} EMBED_TXTFILES ${project_dir}/certs/mqtt0_ca_cert.pem)
```
I find error with last I say:
[image]

WangShuoran, 2023-05-30:
target_add_binary_data only use in project cmakelists.txt. I want to use idf_component_set_property() after idf_component_register() because I can use I customize CMake micro (or variable) to open Compilation Options.

Use customize CMake micro (or variable) can let my projects components rasier reuse.

igrr, 2023-05-30:
> target_add_binary_data only use in project cmakelists.txt

You should be able to call `target_add_binary_data` as well. Are you seeing any issue with that? Could you try doing this in `main/CMakeLists.txt`, after idf_component_register?
```cmake
target_add_binary_data(${COMPONENT_LIB} "${CMAKE_CURRENT_LIST_DIR}/mqtt_eclipseprojects_io.pem" TEXT)
```

Here is another place where we do the same in IDF:
https://github.com/espressif/esp-idf/blob/903af13e847cd301e476d8b16b4ee1c21b30b5c6/components/mbedtls/CMakeLists.txt#LL70C14-L70C14

closed 2023-05-31 by WangShuoran