espressif/esp-idf #11501: TWI / i2c stops communicating after deep sleep

embenix, 2023-05-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32c3

### Power Supply used.

USB

### What is the expected behavior?

I am using ESP32C3 in one of my projects to read the sensor data from SHT3x sensor. After reading the data, I publish this data on the TWAI (CAN) bus as well as on the terminal (idf.py monitor) and then put the unit to deep sleep with timer set for 10-30 seconds wake up time. Every time unit wakes up, it initializes everything, reads data, publishes it on the CAN bus and goes to sleep. And this loop just goes on. For debugging reasons, I keep the USB connected and I can see the data coming out every 10-30 seconds.

[image]

### What is the actual behavior?

After a couple of loops, the unit fails to read the data from SHT3x sensor. No matter what I do with the SHT3x sensor, as long as ESP32C3 module is powered up, it always fails to read sensor data, as if the conversion is still under process. And the only way to fix the problem is to hard reset the ESP32C3 module. Check the SHT3x response after requesting for the data.

[image]

### Steps to reproduce.

Use any ESP32C3 based dev board, connect SHT3x sensor I2C pins to GPIO5 as SDA and GPIO6 as SCL pin. I am using the following library from github for SHT3x.
https://github.com/gschorcht/sht3x-esp-idf

### More Information.

I have tried my code with both the variants i.e. ESP32-C3-MINI-1 and ESP32-C3-WROOM-02. The problem is persistent. And highly likely the problem lies within the chip and not the framework.

Sotatek-TuHoang, 2023-09-14:
@embenix  Excuse me, but have you found the cause of this error?
I also used the same sht3x library, ESP32-S3 also got this error.

embenix, 2023-09-14:
> @embenix Excuse me, but have you found the cause of this error? I also used the same sht3x library, ESP32-S3 also got this error.

@Sotatek-TuHoang, unfortunately Not!
I have used a free gpio pin to reset the ESP module if such a condition is detected. My application is not very sensitive and so reseting the ESP module works just fine.