espressif/esp-idf #11502: NimBle Can occasionally encountered not advertising data

oufei19, 2023-05-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

NimBle,advertising interval 100ms,
usage scenario:250ms updata advertising,Can occasionally encountered no advertising data
invoking function：rc = ble_gap_adv_set_data(raw_adv_data, raw_adv_len);updata advertising
I tried to call function：ble_gap_adv_stop(); ble_gap_adv_start();Have not improved.
no advertising data,As shown in the figure below
[image]

Below is a normal advertising

[image]

rahult-github, 2023-05-26:
Hi @oufei19 ,

Can you please share more information.

1. Which release version is being used  ?
2. Which chip is being used ?
3. Is the issue seen by using existing examples in IDF ? . May be bleprph ? 
4. Can you share application level changes done at your end after which issue is seen.
5. Is it possible to get and Over the Air logs ?
6. Do you have any debug enabled console logs for the run ?

oufei19, 2023-05-29:
> Hi @oufei19 ,
>
> Can you please share more information.
>
> 1. Which release version is being used  ?
> 2. Which chip is being used ?
> 3. Is the issue seen by using existing examples in IDF ? . May be [...]

idf 5.0.2
Esp32 c2
it is have  none example used C2 ，and none example in IDF about C2 update adv datas.
for debug it , I will provide the test example .,it is the same as my codes, can you help me to debug it?
[spp_server_adv_updata.zip](https://github.com/espressif/esp-idf/files/11588274/spp_server_adv_updata.zip)

rahult-github, 2023-05-30:
Hi @oufei19 ,

Thanks for the codes. The code has uart specific changes and taking the code as is causing garbage value to be shown on screen. So, went ahead and took only advertising related changes from your code.  I saw that the adv report is always shown up on nrfconnect app of the phone. 

However, please note that your original image has device name as Health Scale 5 and other details, but the adv data shared in the example file is of the below format, where address and name gets modified later on. I assume this difference is intended as you attached just sample code and not your exact working code ? 

uint8_t raw_adv_data[ADV_LENGHT_MAX] = 
{
        /* flags */
        0x02, 0x01, 0x06,
        /* service uuid */
        0x05, 0x02, 0xf0,0xff,0x12,0x18,
        /* service uuid */
        0x14, 0xff, 0x00, 0x00,
        /* mac addr */
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        /* data */
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00,0x00, 0x00, 0x00, 0x00, 0x00,
};

Since am not able to see the mentioned behaviour, it would help if you could give some more detailed logs: 

1. Please enable Debug logs for ESP chip. You can do the same via 
     Component Config  --> Log Output --> Default Log Verbosity --> Debug
Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs

2. Do you have any OTA logs that can be shared ? 
3. When you observe incomplete data, can you press on the "raw" data on the remote side to show the adv report data recieved in hex ? 

Adv report would ideally go in one packet. So, whatever data is sent, should be seen in its whole ( and not just partial )

oufei19, 2023-05-31:
the project code need run in  my hardware.so I provide the example project to debug,it is the same functions of my project. 
1.example project issue: start runing  ,it`s don`t update adv datas contiune  up 10S.after that ,the  BLE adv is normal run.However, there are cases of lost adv packets  and whe use nrf connect scan the adv ,it is  display adv interval N/A
2. I will provide a new example project to debug.and provide Debug Logs 
 
please help me to Analyze the problem
[image]
[image]
[image]
[spp_server_adv_updata.zip](https://github.com/espressif/esp-idf/files/11607635/spp_server_adv_updata.zip)
[Debug Log.txt](https://github.com/espressif/esp-idf/files/11607636/Debug.Log.txt)

rahult-github, 2023-05-31:
Hi @oufei19 ,

I have some suggestions that you can do to first correct the application.

1. The app_adv_data_init is setting both data and scan response. Please confirm the requirement. If remote device adv reports, it shows the adv data of the reports. When it performs scan request, it will get the scan response which it will then show up. So, it will now be on remote device implementation to show up the adv report data and scan response data. 

2. In your code, in a loop , i observe that you have only invoked the function to set_data. This will not cause any change. The correct flow should rather be: 
    a. Set data 
    b. Start advertising. 
    c. Stop advertising
    d. Set new data
    e. Start advertising. 

Basically ,just setting data won't have the effect, unless you stop / start advertising. 

In the log you shared, the first time advertising happens around line number 503.  

> ^[[0;32mI (2147) APP_ADV: ble_spp_server_advertise^[[0m
^[[0;32mI (2157) NimBLE: GAP procedure initiated: advertise; ^[[0m 
^[[0;32mI (2157) NimBLE: disc_mode=2^[[0m
^[[0;32mI (2167) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=160 adv_itvl_max=160^[[0m
^[[0;32mI (2187) NimBLE: 

So, the adv data set before this instance is the only one that got sent from host to controller. (Line 458 - 500 ).  

After this, as mentioned above, no matter how many times your application sets adv data, it never gets transmitted.

rahult-github, 2025-03-04:
Hi @oufei19  , this is an old issue. host will simply post the adv data information to underlying firmware which will then control sending information to remote. Can you possibly switch to latest code and see if the issue is resolved.

closed 2025-07-07 by rahult-github