espressif/esp-idf #11503: NimBle Low speed advertising Lightsleep electric current

oufei19, 2023-05-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

NimBle Advertising interval 1 second,open Power Management-Support for power management item.

rahult-github, 2023-05-26:
An example showcasing typical current consumption with configuration suggested to check is being developed. It can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/power_save) .  Currently it is done for esp32 / c3 / s3.  We will plan to add support for c2 / c6 / h2 in same example.

oufei19, 2023-05-29:
> An example showcasing typical current consumption with configuration suggested to check is being developed. It can be found [...]

Ok, when is it supported? beacause,I have a project in development that needs it .help me  please

rahult-github, 2025-03-10:
Example has already been updated to support other chips. Closing this issue.

closed 2025-03-10 by rahult-github