espressif/esp-idf #11504: UNEXPECTED RESET WHN BLE SCANNING AND HIGH DENSITY OFF DEVICES, HELP...

andurbal, 2023-05-25:
Dear Sirs

we have founded a problem that is critical in our product.
Suddlenly we have very often (20 or 30 secons) unexpected resets form the BT library. It occurs when we place our product in a high density bluetooth devices. 
We use BT Dual Mode as we need both BT Classic and BLE.

Using at this moment ESP IDf v4.4.4-439-gc25d92074e

We sart and stop BT Classic  Inquirys (3 secs scan) Stop, Wait some seconds an Start again, for looking near devices.
We also launch BLE Scan for close devices.

As said when we place the product in a hign density bluetooh devices we have started to see BT crashes very very often.

There are some logs of the crashes (not always the same message).

E (29892) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (29892) task_wdt:  - IDLE (CPU 0)
E (29892) task_wdt: Tasks currently running:
E (29892) task_wdt: CPU 0: BTC_TASK
E (29892) task_wdt: CPU 1: IDLE
E (29892) task_wdt: Aborting.

[0;31mE (24922) BT_OSI: list_append osi_calloc failed.
[0;31mE (24923) BT_HCI: host_recv_pkt_cb couldn't aquire memory for inbound data buffer.
[0;31mE (26816) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (26816) task_wdt:  - IDLE (CPU 0)[0m
[0;31mE (26816) task_wdt: Tasks currently running:[0m
[0;31mE (26816) task_wdt: CPU 0: BTC_TASK[0m
[0;31mE (26816) task_wdt: CPU 1: IDLE[0m
[0;31mE (26816) task_wdt: Aborting.[0m

I Attach the files with the crashses.

Its really urgent to find what is happenuing as this is a Stopper for our product release, and we have some thousands working.

[01 Error.txt](https://github.com/espressif/esp-idf/files/11564702/01.Error.txt)
[02 Error.txt](https://github.com/espressif/esp-idf/files/11564703/02.Error.txt)
[03 Error.txt](https://github.com/espressif/esp-idf/files/11564706/03.Error.txt)
[01BluError.txt](https://github.com/espressif/esp-idf/files/11564713/01BluError.txt)
[04 Capture.txt](https://github.com/espressif/esp-idf/files/11564708/04.Capture.txt)

Weijian-Espressif, 2023-05-26:
@andurbal 
Apparently the problem is that the chip is running out of memory.
1. Why does the memory run out, and how much memory is left when the code is initialized? Is there a task with a higher priority than the Bluetooth task working?
2. BLE duplicate scan can be turned on to check duplicate broadcast packet reporting, which reduces system memory consumption
3. If the above questions have not solved your problem, please provide the sdkconfig file

andurbal, 2023-05-28:
Thanks for you suggestions. We have tried all of them.

Finally we have found the problem and it was on our side, due that for some
reason somebody is broadcasting an invalid PDU not following Bluetooth
guidelines  and our PDU parser crashed in a loop not returning to the GAP
event handler.

I cannot share here the invalid PDU but if you are interested I can send a
PM.

Thanks

Alvin1Zhang, 2023-05-31:
Thanks for sharing the updates. Feel free to reopen.

closed 2023-05-31 by Alvin1Zhang