espressif/esp-idf #11510: [v5.1] esp_wifi_set_config() randomly returns ESP_ERR_INVALID_ARG ~20% of the time

chipweinberger, 2023-05-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

It used to always work, on v4.4

### What is the actual behavior?

20% of the time,  `ESP_ERR_INVALID_ARG` is returned. It did not do this on v4.4

### Steps to reproduce.

see debug logs

```
// create AP
wifi_config_t create_hotspot_config(pd_error_t* error)
{
    pd_err_unset(error);

    wifi_config_t config;
    memset(&config, 0, sizeof(wifi_config_t));
    {
        uint8_t channel = esp_random() % 13;
        ESP_LOGI(TAG, "AP channel %" PRIu8 "", channel);
        config.ap.channel = channel;
        config.ap.authmode = WIFI_AUTH_OPEN;
        config.ap.ssid_hidden = false;
        config.ap.max_connection = 5;
        config.ap.beacon_interval = 100;

        ESP_LOGI(TAG, "switching to 'Create Wifi AP Hotspot' mode");

        wifi_ap_config_t* ap = &config.ap;

        // set default ssid
        if (strlen(apSSID) == 0){
            eFail("apSSID not set");
            goto end;
        } 

        // set ssid
        ap->ssid_len = 32;
        memset((char*) ap->ssid, 0, 32);
        memcpy((char*) ap->ssid, apSSID, PDMIN(32, strlen(apSSID)));

        // set password
        if (strlen(apPassword) > 0) {
            config.ap.authmode = WIFI_AUTH_WPA_WPA2_PSK;
            memset(ap->password, 0, 64);
            memcpy((char*) ap->password, apPassword, PDMIN(64, strlen(apPassword)));
        }

        ESP_LOGI(TAG, "\nssid:%s \npassword:%s", apSSID, apPassword);

        pd_err_success(error);
    }
end:
    eOnFail(memset(&config, 0, sizeof(wifi_config_t)));
    return config;
}
```

```
        // create configuration
        wifi_config_t config; 
        wifi_interface_t intf;
        if (wifi_mode == PD_WIFI_MODE_AP){
            config = create_hotspot_config(error);
            intf = WIFI_IF_AP;
        } else {
            config = create_home_wifi_config(error);
            intf = WIFI_IF_STA;
        }
        if (is_fail(error)){
            eFail("create configuration failed.");
            goto end;
        }

        // set config
        if (init.esp_wifi_set_config == false) {
            ESP_LOGI(TAG, "esp_wifi_set_config(%s)", wifi_interface_str(intf));
            sTry(err =, esp_wifi_set_config,(intf,&config));
            init.esp_wifi_set_config = true;
        }
```

### Debug Logs.

```plain
(5021) pd wifi: Entering Discovery Mode
I (5021) pd wifi: wifi_start_internal(ap)
I (5021) pd wifi: esp_netif_init()
I (5021) pd wifi: esp_event_loop_create_default()
I (5021) pd wifi: esp_netif_create_default_wifi_ap()
I (5021) pd wifi: esp_wifi_init()
I (5031) pp: pp rom version: e7ae62f
I (5031) net80211: net80211 rom version: e7ae62f
I (5051) wifi_init: rx ba win: 6
I (5051) wifi_init: tcpip mbox: 32
I (5051) wifi_init: udp mbox: 6
I (5051) wifi_init: tcp mbox: 6
I (5051) wifi_init: tcp tx win: 6500
I (5061) wifi_init: tcp rx win: 6500
I (5061) wifi_init: tcp mss: 1440
I (5061) wifi_init: WiFi/LWIP prefer SPIRAM
I (5061) pd wifi: esp_event_handler_instance_register(ESP_EVENT_ANY_ID)
I (5061) pd wifi: esp_event_handler_instance_register(IP_EVENT_STA_GOT_IP)
I (5061) pd wifi: esp_wifi_set_mode(WIFI_MODE_AP)
I (5071) pd wifi: AP channel 12
I (5071) pd wifi: switching to 'Create Wifi AP Hotspot' mode
I (5071) pd wifi: 
ssid:jamcorder-dxm9h1 
password:
I (5071) pd wifi: esp_wifi_set_config(WIFI_IF_AP)
W (5081) pd issues: RAISE - [Boot] Failed to fully boot | wifi
E (5081) : e1 : [pd wifi] esp_wifi_set_config() failed. ESP_ERR_INVALID_ARG
e2 : [pd wifi] wifi_start_internal() failed.
e3 : [pd main] pd_wifi_start_ap_mode() failed
e4 : [pd main] start_wifi() failed.
```
```

KaeLL, 2023-05-26:
```c
wifi_config_t config;
memset(&config, 0, sizeof(wifi_config_t));
```

I've always been confused about why people do this instead of
```c
wifi_config_t config = {0};
```
That aside, are you sure this isn't due to
```c
// create configuration
wifi_config_t config;
```
?

chipweinberger, 2023-05-26:
the uninitialized config gets set just a couple lines later.

```
        // create configuration
        wifi_config_t config; 
        wifi_interface_t intf;
        if (wifi_mode == PD_WIFI_MODE_AP){
            config = create_hotspot_config(error);
            intf = WIFI_IF_AP;
        } else {
            config = create_home_wifi_config(error);
            intf = WIFI_IF_STA;
        }
```

igrr, 2023-05-29:
Can you also show the implementation of `create_home_wifi_config`?

As a troubleshooting step, I recommend you to print out the content of `wifi_config_t` structure right before `esp_wifi_set_config`. (Either in GDB, or to the console via ESP_LOG.) You will probably see which value of which field of the structure correlates with `esp_wifi_set_config` returning an error.

chipweinberger, 2023-06-08:
I think I know the problem. It only happens in AP mode (Wifi Hotspot), and I think due to this code

```
        uint8_t channel = esp_random() % 13;
        ESP_LOGI(TAG, "AP channel %" PRIu8 "", channel);
        config.ap.channel = channel;
```

Perhaps there is some channel that is not allowed?

chipweinberger, 2023-06-08:
So to summarize, the issue is in my code. I was trying to use channel 12, but the maximum is 11. 

It would be nice if ESP-IDF printed the reason because this function is not open source so we are only left to guess.

closed 2023-06-08 by chipweinberger

KaeLL, 2023-06-09:
Isn't channel availability dependent on the country selected when initializing the WiFi driver? I think the US has some rules against using channels 12 and 13. And what's with channel index being 0-based?

chipweinberger, 2023-06-09:
> And what's with channel index being 0-based?

I think I was wrong about that. On my home router, channels go from 1-11. 

On ESP32, they go from 0 - 11. I'm not sure what channel 0 is.