espressif/esp-idf #11511: mqtt_client: Client was not initialized

srk517, 2023-05-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello I am using the updated version of IDF and trying to connect my esp32s3 with tsl_mutual_auth example code. I am providing each and every detail in code from credentials , certificates, host name , uri and so on. But eventually esp_mqtt_client_init(&mqtt_cfg); always return NULL and it says client was not initialized. I did the same project on older version and it works. Please fix this issue I am not sure whats going on but esp_mqtt_client_init(&mqtt_cfg); has some issues or esp_mqtt_client_config_t mqtt_cfg is not responding accordingly with data. Thanks!

KaeLL, 2023-05-26:
> eventually

Care to elaborate on that by showing the code?

srk517, 2023-05-29:
Yes certainly, In function mqtt_app_start which contains esp_mqtt_client_config_t mqtt_cfg i am adding all credentials which are later assigned to esp_mqtt_client_inti , but despite taking all credentials code returns "client was not initialized". The same code works in older version of esp-idf where it is much more simple. Here is code and all struct credentials being assigned.

static void mqtt_app_start(void)
{
  usage();
  const esp_mqtt_client_config_t mqtt_cfg = {

    .broker.address.uri = AWS_IOT_ENDPOINT ,//AWS_IOT_ENDPOINT,
    .broker.address.hostname= "iot.thingsty.com",
    .broker.address.transport = MQTT_TRANSPORT_OVER_SSL,//MQTT_TRANSPORT_OVER_SSL,
    .broker.address.port=AWS_IOT_PORT,
    .broker.verification.use_global_ca_store=true,
    .broker.verification.certificate = (const char *)server_cert_pem_start,
    .broker.verification.certificate_len=server_cert_size,
    .broker.verification.psk_hint_key=NULL,
    .broker.verification.skip_cert_common_name_check=false,
    .broker.verification.alpn_protos=NULL,
    .credentials = {
      //.username="srk",  
      .client_id = "630f3f41376e947cf63a",
      .set_null_client_id=false,
      .authentication = {
        //.password = "123456",
        .certificate = (const char *)client_cert_pem_start,
        .certificate_len = client_cert_size,
        .key = (const char *)client_key_pem_start,
        .key_len=client_key_size,
        .key_password = NULL,
        .key_password_len = 0,
        .use_secure_element = false,
        .ds_data = NULL
        
      },
    },
    .session = {
            .last_will = {
               // .topic = MQTT_TOPIC,
               // .msg = NULL,
              //  .msg_len = 0,
               // .qos = 0,
               // .retain = 0
            },
            .disable_clean_session = false,
            .keepalive = 120,
            .disable_keepalive = false,
            .protocol_ver = MQTT_PROTOCOL_V_3_1,
            .message_retransmit_timeout = 0
      },
        .network = {
            .reconnect_timeout_ms = 5000,
            .timeout_ms = 5000,
            .refresh_connection_after_ms = 1000,
            .disable_auto_reconnect = false
        },
        .task = {
            .priority = 5,
            .stack_size = 8192
        },
        .buffer = {
            .size = 1024,
            .out_size = 1024
        },
        //.event_handle_cb = mqtt_event_handler_cb,
        //.transport_type = MQTT_TRANSPORT_OVER_SSL

  };
  //////////////Code ENDS///////////////////////
"//.event_handle_cb = mqtt_event_handler_cb,
        //.transport_type = MQTT_TRANSPORT_OVER_SSL"
        this line when uncommented says that these are not part of struct of client config. Please tell me what is going on. Older versions were working but has issues.

ESP-YJM, 2023-05-30:
@srk517 `event_handle_cb` is removed, you should use `esp_mqtt_client_register_event` to regist event callback. And `transport_type ` is renamed as `.broker.address.transport`. Besides, the problem that `client was not initiated`, could you please offer the whole log. The log maybe show the reason.

srk517, 2023-06-08:
Issue resolved.

Alvin1Zhang, 2023-06-08:
Thanks for sharing the updates, feel free to reopen.

closed 2023-06-08 by Alvin1Zhang

KhubaibAli95, 2024-07-09:
> Issue resolved.

Hey,
How did you resolve it? I am also trying to initialize the client using the same function of ESPIDF in Arduino ide but I could not able to do it. I am getting the same result. I am also sending all parameters like username, password, client id all. But still receiving the same error. Kindly could you please guide me how you solve it .