espressif/esp-idf #11513: WIFI STA auto record AP info to NVS, But next running the ESP32 can not read.

WangShuoran, 2023-05-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In doc show:
[image]
If the Wi-Fi NVS flash is enabled by menuconfig, all Wi-Fi configuration in this phase, or later phases, will be stored into flash. When the board powers on/reboots, you do not need to configure the Wi-Fi driver from scratch. You only need to call esp_wifi_get_xxx APIs to fetch the configuration stored in flash previously. You can also configure the Wi-Fi driver if the previous configuration is not what you want.

I want to achieve if the STA record AP info is empty, I code to set a default AP info to STA. If AP info is exist in ESP32 NVS, I will use STA record AP info to connect AP(The new AP that may be connected is obtained through the BLUFI distribution network).

Because when I use BLUFI distribution AP info to STA, the STA restart can connect to last record AP info. I think wifi set the AP info to NVS because wifi auto connect to AP.

But I try to use esp_wifi_get_config() to get config, and I find the STA info is empty:
```c
wifi_config_t wifiStaConfig = {0};
esp_err_t ret = esp_wifi_get_config(ESP_IF_WIFI_STA, &wifiStaConfig);
printf("%s\n",wifiStaConfig.sta.ssid);
printf("%s\n",wifiStaConfig.sta.password);
```

I want to know why, And I want to know the AP info is set where?

WangShuoran, 2023-05-26:
I am wrong with esp_wifi_get_config(), it can get the ESP32 storage AP info, but I want to know why I read the AP data with those code get same with STA data?
```c
wifi_config_t wifiStaConfig = {0};
esp_err_t ret = esp_wifi_get_config(ESP_IF_WIFI_STA, &wifiStaConfig);
printf("%s\n",wifiStaConfig.ap.ssid);
printf("%s\n",wifiStaConfig.ap.password);
```

kriegste, 2023-05-26:
wifi_config_t is a union and the first two fields of the (struct) members are identical (by "coincidence").

closed 2023-05-27 by WangShuoran