espressif/esp-idf #11515: Error messages when trying to send data via BLE from a third-party microcontroller

marco711, 2023-05-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**IDF version**
ESP-IDF v5.1-dev-1006-g02605f1a31-dirty 2nd stage bootloader

**SDK config file**
Attached:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/11573709/sdkconfig.zip)

**Issue**
Hello,
I'm using an ESP32-S3 module as BLE peripheral. I'm trying to use a third-party BLE module (nRF52) as central. I can pair both microcontrollers but when the BLE central (nRF52) send messages in the ESP32 terminal I get the following error messages:

```
[2023-05-26_11:16:33:33.183550] E (92019653) BLUFI_EXAMPLE: BLUFI report error, error code 0
[2023-05-26_11:16:33:33.183770] 
[2023-05-26_11:16:34:34.179400] E (92020653) BT_BTC: btc_blufi_recv_handler seq 99 is not expect 1
```

I can send data to the ESP32 via Bluetooth with a mobile phone and the EspBluFi application so I'm quite sure there's no problem on the ESP32 software. Can you help me understand the error messages displayed in the ESP32 log? This way I'll be able to correct the software on the nRF52 side.

Weijian-Espressif, 2023-05-31:
@marco711 blufi In order to prevent replay attacks, each packet contains index information, and the index is incremented by one each time a packet is sent. The error reported here is because the index of the received packet does not meet expectations

marco711, 2023-05-31:
@Weijian-Espressif in which part of the BLE packet the incrementing should be stored?

Weijian-Espressif, 2023-05-31:
@marco711 Please see the readme document in the blufi example, which contains some blufi-related documentation links

marco711, 2023-05-31:
@Weijian-Espressif I have added the incrementing packet and I see that the ESP32 receives the incremented data. Nevertheless, there is still an error:
```
[2023-05-31_11:21:26:26.799431] E (6580913) BT_BTC: btc_blufi_recv_handler seq 1 is not expect 1
[2023-05-31_11:21:26:26.799530] 
[2023-05-31_11:21:26:26.802583] E (6580913) BLUFI_EXAMPLE: BLUFI report error, error code 0
[2023-05-31_11:21:26:26.802869] 
[2023-05-31_11:21:27:27.799279] E (6581913) BT_BTC: btc_blufi_recv_handler seq 2 is not expect 1
[2023-05-31_11:21:27:27.799545] 
[2023-05-31_11:21:27:27.802352] E (6581913) BLUFI_EXAMPLE: BLUFI report error, error code 0
[2023-05-31_11:21:27:27.802432] 
[2023-05-31_11:21:28:28.897268] E (6583013) BT_BTC: btc_blufi_recv_handler seq 3 is not expect 1
[2023-05-31_11:21:28:28.897338] 
[2023-05-31_11:21:28:28.901194] E (6583013) BLUFI_EXAMPLE: BLUFI report error, error code 0
[2023-05-31_11:21:28:28.901265] 
[2023-05-31_11:21:29:29.897458] E (6584013) BT_BTC: btc_blufi_recv_handler seq 4 is not expect 1
[2023-05-31_11:21:29:29.897541] 
[2023-05-31_11:21:29:29.901322] E (6584013) BLUFI_EXAMPLE: BLUFI report error, error code 0
```
Which is the problem in this case?
I've read [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/blufi.html#frame-formats) documentation and it is not clear to me which are the possible types and subtypes so I think I'm sending frames which are not completely compliant with Blufi.

Weijian-Espressif, 2023-05-31:
[image]

@marco711 seq value is not right, should start counting from 0.

marco711, 2023-05-31:
Thanks! I was starting counting from 1 and that was the root of the problem.

closed 2023-05-31 by marco711