espressif/esp-idf #11517: CST816 Touch driver support

krupis, 2023-05-26:
### Is your feature request related to a problem?

I have been using ST7789 display from the lilygo and it works flawlesly.

The only issue is that I believe the esp-idf does not currently support the touch driver (CST816) for this display.

[image]

https://www.lilygo.cc/products/t-display-s3

I think adding support for this touch driver would be very useful for developers such as myself.

### Describe the solution you'd like.

Working touch display driver for CST816 and example project on how to get it going.

### Describe alternatives you've considered.

Using different displays.

espzav, 2023-05-29:
Hello,
the touch driver for CST816 is here: https://components.espressif.com/components/espressif/esp_lcd_touch_cst816s

krupis, 2023-05-29:
> Hello, the touch driver for CST816 is here: https://components.espressif.com/components/espressif/esp_lcd_touch_cst816s

Thanks for quick reply. Is CST816 the same as CST816S? Because the driver that you linked is for CST816S.

Regardless, I have tried to use this touch driver. My full function that initializes the LVGL Display:

```
void lvgl_setup()
{

    gpio_config_t pwr_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_POWER};
    ESP_ERROR_CHECK(gpio_config(&pwr_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_POWER, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    gpio_config_t input_conf =
        {
            .mode = GPIO_MODE_INPUT,
            .pull_up_en = GPIO_PULLUP_ENABLE,
            .pin_bit_mask = 1ULL << PIN_LCD_RD};
    ESP_ERROR_CHECK(gpio_config(&input_conf));

    gpio_config_t bk_gpio_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_BK_LIGHT};
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_BK_LIGHT, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    static lv_disp_draw_buf_t disp_buf; // contains internal graphic buffer(s) called draw buffer(s)
    static lv_disp_drv_t disp_drv;      // contains callback functions

    ESP_LOGI(TAG, "Initialize Intel 8080 bus");
    esp_lcd_i80_bus_handle_t i80_bus = NULL;
    esp_lcd_i80_bus_config_t bus_config = {
        .clk_src = LCD_CLK_SRC_DEFAULT,
        .dc_gpio_num = EXAMPLE_PIN_NUM_DC,
        .wr_gpio_num = EXAMPLE_PIN_NUM_PCLK,
        .data_gpio_nums = {
            EXAMPLE_PIN_NUM_DATA0,
            EXAMPLE_PIN_NUM_DATA1,
            EXAMPLE_PIN_NUM_DATA2,
            EXAMPLE_PIN_NUM_DATA3,
            EXAMPLE_PIN_NUM_DATA4,
            EXAMPLE_PIN_NUM_DATA5,
            EXAMPLE_PIN_NUM_DATA6,
            EXAMPLE_PIN_NUM_DATA7,
        },
        .bus_width = 8,
        .max_transfer_bytes = LVGL_LCD_BUF_SIZE * sizeof(uint16_t)
        //.psram_trans_align = EXAMPLE_PSRAM_DATA_ALIGNMENT,
        //.sram_trans_align = 4,
    };
    ESP_ERROR_CHECK(esp_lcd_new_i80_bus(&bus_config, &i80_bus));
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_i80_config_t io_config = {
        .cs_gpio_num = EXAMPLE_PIN_NUM_CS,
        .pclk_hz = EXAMPLE_LCD_PIXEL_CLOCK_HZ,
        .trans_queue_depth = 20,
        .dc_levels = {
            .dc_idle_level = 0,
            .dc_cmd_level = 0,
            .dc_dummy_level = 0,
            .dc_data_level = 1,
        },

        .on_color_trans_done = example_notify_lvgl_flush_ready,
        .user_ctx = &disp_drv,
[... 155 lines cut ...]
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        },
    };

    ESP_LOGI(TAG,"esp_lcd_touch_new_i2c_cst816s");
    esp_lcd_touch_new_i2c_cst816s(tp_io_handle, &tp_cfg, &tp);
```

I get the following errors printed in the console:
```
[1B][0;32mI (505) LVGL_SETUP: esp_lcd_new_panel_io_i2c[1B][0m
[1B][0;32mI (505) LVGL_SETUP: esp_lcd_touch_new_i2c_cst816s[1B][0m
[1B][0;32mI (505) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (515) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 [1B][0m
[1B][0;31mE (545) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (545) CST816S: read_id(174): I2C read failed[1B][0m
[1B][0;31mE (545) CST816S: esp_lcd_touch_new_i2c_cst816s(84): Read version failed[1B][0m
[1B][0;32mI (545) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (555) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;31mE (565) CST816S: Initialization failed![1B][0m
```

espzav, 2023-05-31:
Hello,
it seems, that `CST816` and `CST816S` are same. Same address and same registers. I haven't got same board and I cannot test it now. It seems, you have mistake in GPIOS on I2C. It should be like here:

```
#define EXAMPLE_I2C_SCL                 17
#define EXAMPLE_I2C_SDA                18
```

krupis, 2023-05-31:
> Hello, it seems, that `CST816` and `CST816S` are same. Same address and same registers. I haven't got same board and I cannot test it now. It seems, you have mistake in GPIOS on I2C. It should be [...]

Thanks for your response. I still havent been able to get it running. Can you confirm where did you get these I2C GPIO declaration. I have declared:
```
#define EXAMPLE_I2C_SCL                 18
#define EXAMPLE_I2C_SDA                 17
```

And that is correct according to the 
[image]

espzav, 2023-05-31:
I found it on the schematics: https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/schematic/T_Display_S3.pdf

krupis, 2023-05-31:
> I found it on the schematics: https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/schematic/T_Display_S3.pdf

Yeah you are right. In the schematic it shows the following:
[image]

but in the Readme.md it shows the following:
[image]

Lets hope to hear from someone who managed to get it to work on the esp-idf.

krupis, 2023-06-02:
After looking at the Arduino example code provided by LilyGO, I noticed that GPIO_21 is set being toggled. I have added a simillar function to the esp-idf:

```
    gpio_config_t touch_io_config =
        {
            .mode = GPIO_MODE_OUTPUT,
            .pull_up_en = GPIO_PULLUP_ENABLE,
            .pin_bit_mask = 1ULL << PIN_TOUCH_RD};
    ESP_ERROR_CHECK(gpio_config(&input_conf));

    gpio_set_level(PIN_TOUCH_RD, 0);
    vTaskDelay(500 / portTICK_PERIOD_MS);
    gpio_set_level(PIN_TOUCH_RD, 1);
```

But the results seem to be the same. I cant seem to get past the i2c touch driver initialization.

espzav, 2023-06-06:
@Lzw655 please, could you help with this? I haven't got HW for CST816 touch.