espressif/esp-idf #11518: NVS namespace deletion

KaeLL, 2023-05-26:
### Is your feature request related to a problem?

Yes. I use NVS to store a set of app configurations, and I need namespace deletion in order to facilitate versioning them. Currently, the most straightforward way to do it is by ping-ponging between 2 NVS partitions just like OTA partitions at OTA updates.

### Describe the solution you'd like.

I want to be able to import cfg's from the app the board just updated from and delete its entire namespace.

igrr, 2023-05-29:
@KaeLL Have you tried using `nvs_erase_all` function to erase the namespace? If yes, in what way does it not fit your use case?

KaeLL, 2023-05-29:
```c
void app_main( void )
{
    ESP_ERROR_CHECK( nvs_flash_erase() );
    ESP_ERROR_CHECK( nvs_flash_init() );

    for ( int i = 0; i < 256; ++i )
    {
        nvs_stats_t stats;
        ESP_ERROR_CHECK( nvs_get_stats( NULL, &stats ) );
        printf(
            "Used entries: %3zu\t"
            "Free entries: %3zu\t"
            "Total entries: %3zu\t"
            "Namespace count: %3zu\n",
            stats.used_entries,
            stats.free_entries,
            stats.total_entries,
            stats.namespace_count );
        nvs_handle_t my_handle;
        char namespace[ 16 ] = { 0 };

        sprintf( namespace, "lmao_%d", i );
        ESP_ERROR_CHECK( nvs_open( namespace, NVS_READWRITE, &my_handle ) );

        ESP_ERROR_CHECK( nvs_set_i32( my_handle, namespace, i ) );

        ESP_ERROR_CHECK( nvs_commit( my_handle ) );
        nvs_close( my_handle );

        ESP_ERROR_CHECK( nvs_open( namespace, NVS_READWRITE, &my_handle ) );
        ESP_ERROR_CHECK( nvs_erase_all( my_handle ) );
        ESP_ERROR_CHECK( nvs_commit( my_handle ) );
        nvs_close( my_handle );
    }
}
```
results in 
```
Used entries: 253       Free entries: 503       Total entries: 756      Namespace count: 253
Used entries: 254       Free entries: 502       Total entries: 756      Namespace count: 254
ESP_ERROR_CHECK failed: esp_err_t 0x1105 (ESP_ERR_NVS_NOT_ENOUGH_SPACE) at 0x400d50e2

expression: nvs_open( namespace, NVS_READWRITE, &my_handle )
```
So unless I'm missing something, `nvs_erase_all` doesn't really erases namespaces and its name is a misnomer.
And even if `nvs_erase_all` did what I wanted, what I need is a function that takes both a handle to a NVS partition and the namespace name, so as to not limit to only the default "nvs" partition. Something along the lines of:
```c
esp_err_t nvs_erase_namespace( nvs_handle_t handle, const char *namespace_name );
```
that would erase the provided namespace from the provided partition, along with all keys and values inside it.

igrr, 2023-05-29:
That's true, `nvs_erase_all` doesn't erase the namespace entry, it erases the key-value pairs associated with that namespace.

Is that a practical issue for your use case, though? (I.e. are you planning to have  more than 254 versions of the application?)

---

With regards to

> esp_err_t nvs_erase_namespace( nvs_handle_t handle, const char *namespace_name );

note that `nvs_handle_t handle` identifies the specific namespace already, so `namespace_name` is not needed. You specify the namespace name when calling `nvs_open` or `nvs_open_from_partition`. The latter allows you to open a namespace in an NVS partition other than the default one.

KaeLL, 2023-05-29:
I don't intend on having more than 254 namespaces. I intend on versioning them.
The thing here is that I use NVS to store app configuration and sensitive credentials. The idea is to use the app version as the namespace, and after updates, import non-deprecated configurations and delete the old namespace once the new app has been marked as valid.

As for the missed redundancy on my suggestion, sorry about that, brain fart.

igrr, 2023-05-30:
The use case you are describing can also be implemented without having to erase the namespace entry. Instead of using the version as the namespace name, you can use two namespaces and alternate between them.

Suppose the two namespaces are `app_data_0` and `app_data_1`.

- On normal boot: open the namespace corresponding to the current OTA slot(`app_data_0` if we are running from `ota0`, `app_data_1` if we are running from `ota1`)
- After update (say, from `ota0` to `ota1`: open the other namespace (`app_data_0`), read the old version from the key `version` in that namespace. Decide which key-value pairs to transfer into the current namespace (`app_data_1`). When done and when the new slot is marked valid, delete the key-value pairs in the old namespace (`app_data_0`). The namespace entry is not deleted, but that doesn't matter because you will reuse when updating into `ota0`.

I think the extra complexity is minimal (just the key-value pair `version` to store the app version) and you can implement this in any released IDF version, without having to wait for the new feature.

KaeLL, 2023-05-30:
That does it. Thanks.
Problem solved, but the feature wasn't added, so close the issue at your discretion.