espressif/esp-idf #11522: Bluedroid PBAP/PBA profile support

danjulio, 2023-05-27:
### Is your feature request related to a problem?

I am designing a telephony device using the Bluedroid HFP running on an ESP32.  It has a small LCD touchscreen that provides the user interface.  It would be useful for the users to be able to access the address book from the phone on that screen to both identify incoming caller names and to initiate outgoing calls from the device.  The same kind of functionality that exists in automotive head-end units.  This requires the PBA and/or PBAP profile to be supported.

### Describe the solution you'd like.

Support for the PBAP profile (and/or PBA) in the Espressif Bluedroid port.  I am interested in PCE (Client ) support but since you support AG HFP then I think you'd also have to support PSE (server) as well.

### Describe alternatives you've considered.

I looked at bStack but would be nice to use Espressif's native library.

BetterJincheng, 2023-06-19:
@danjulio Thank you very much for your suggestion. PBAP is an upper-layer protocol based on OBEX, but we do not currently support OBEX. WWe will have people supporting this feature in early 2024. At that time, I will remind you here.

Matthieu-Poulain, 2024-11-19:
> We will have people supporting this feature in early 2024

Hi, 
We are near of 2024 end, and i have not seen any PBAP or Obex feature log in the newer changes. Is the functionnality still planned and if so do you have a release time target?

Many thanks

esp-lrh, 2024-12-04:
Hi, @danjulio @Matthieu-Poulain 

The PBAP client feature will be supported soon. The merge request is currently under review and is expected to be merged this week.

As for the PBAP server, we currently have no plans to support it.

danjulio, 2024-12-04:
Thanks @esp-lrh.  I'll definitely take a look!

closed 2024-12-13 by espressif-bot