espressif/esp-idf #11523: target linux: bug in heap_caps_realloc_base

espradio, 2023-05-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

realloc is expecting the size not caps. Diff for the same.
```
diff --git a/components/heap/heap_caps_linux.c b/components/heap/heap_caps_linux.c
index 981f861674..67746132de 100644
--- a/components/heap/heap_caps_linux.c
+++ b/components/heap/heap_caps_linux.c
@@ -80,7 +80,7 @@ void *heap_caps_malloc_prefer( size_t size, size_t num, ... )

 static void *heap_caps_realloc_base( void *ptr, size_t size, uint32_t caps)
 {
-    ptr = realloc(ptr, caps);
+    ptr = realloc(ptr, size);

     if (ptr == NULL && size > 0) {
         heap_caps_alloc_failed(size, caps, __func__);
```

ESP-Marius, 2023-05-29:
Ah, you are right. Thanks for reporting this.

We'll fix it.

0xjakob, 2023-05-31:
@espradio The fix has been merged internally and will synchronize to github soon.

closed 2023-06-03 by espressif-bot