espressif/esp-idf #11525: heap corruption presumably in dhcp module

moefear85, 2023-05-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

latest master

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s2 module

### Power Supply used.

External 3.3V

### What is the expected behavior?

run

### What is the actual behavior?

crashes

### Steps to reproduce.

I simply flashed the standard wifi-sta example without any modifications, but using my own sdkconfig, where I enable heap debugging. It exposed a memory leak, see logs.
[debug.txt](https://github.com/espressif/esp-idf/files/11585267/debug.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/11585269/sdkconfig.txt)

xueyunfei998, 2023-07-07:
hi @moefear85 
To sum up your first question above.
1 The problem of stepping on memory
2 The hostname is too long, which leads to assert. 

rigth？

moefear85, 2023-07-07:
nr.2 is correct, leads to heap corruption, missing assert.

nr.2: there is bug in AP logic, so after several successful reconnects of a sta, the AP begins endlessly rejecting further reconnects for no reason (despite only 2 stations). I don't know if it is because memory issue, but it is present even when hostname is proper length.

closed 2024-01-11 by MaxwellAlan