espressif/esp-idf #11526: Issue: Smart Config Scan Error - promis scan ret -1

ABHITHLALC, 2023-05-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf -v4.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

USB

### What is the expected behavior?

I am encountering an error while attempting to use Smart Config on my ESP32 device. During the startup process, the Smart Config feature works perfectly fine. However, when I initiate the Smart Config process by pressing a button, I encounter the following error message: "E (379857) smartconfig: promis scan ret -1." This error seems to indicate an issue with the scan process during Smart Config. It states that the scan is not allowed because the station (STA) is already connecting.

I expect the Smart Config process to initiate successfully when triggered by pressing the button. The ESP32 device should be able to scan for available networks and allow me to configure the Wi-Fi connection.

### What is the actual behavior?

When I trigger the Smart Config process by pressing the button, I receive the following error message: "E (379857) smartconfig: promis scan ret -1." This error prevents the scan process during Smart Config, and I am unable to configure the Wi-Fi connection.

### Steps to reproduce.

1. Step::Initialize the ESP32 device and Wi-Fi module.
2. Step::Check if SSID and password are already stored in NVS.
3. Step::If SSID and password are present, connect to Wi-Fi using stored credentials.
4. Step::If SSID and password are not present, initiate the Smart Config process.
5. Step::Press the button for 5 seconds to trigger Smart Config.

### Debug Logs.

```plain
normal condition 
WifiLed ON .......!!!! 
WifiLed OFF .......!!!! 
WifiLed ON .......!!!! 
WifiLed OFF .......!!!! 
WifiLed ON .......!!!! 
WifiLed OFF .......!!!! 
WifiLed ON .......!!!! 
WifiLed OFF .......!!!! 
WifiLed ON .......!!!! 
WifiLed OFF .......!!!! 
I (15697) SMART_CONFIG: Entering into Smart config mode ::::Starting Smart Config Setup - Please Open APP
I (15707) wifi:wifi driver task: 3ffc2b9c, prio:23, stack:6656, core=0
I (15707) system_api: Base MAC address is not set
I (15707) system_api: read default base MAC address from EFUSE
I (15717) wifi:wifi firmware version: 63017e0
I (15717) wifi:wifi certification version: v7.0
I (15717) wifi:config NVS flash: enabled
I (15717) wifi:config nano formating: disabled
[... 370 lines cut ...]
                ESP_LOGI("NVS", "SSID and PASSWORD successfully committed to NVS data");
            }

            nvs_close(loop_Handle);
          
        }

        if (uxBits & ESPTOUCH_DONE_BIT)
        {
            ESP_LOGI(TAG, "smartconfig over");
            esp_smartconfig_stop();
            if(run_smart_config_flag)
            {
            // vTaskResume(temp_task_handle);
            // vTaskResume(main_task_handle);
            run_smart_config_flag = false;
            break;
            }
            esp_restart();
           
        }
    }
}
```
Any assistance or guidance in resolving this issue would be greatly appreciated. Thank you for your help.

Alvin1Zhang, 2025-04-09:
Thanks for reporting.

zhangyanjiaoesp, 2025-04-09:
@ABHITHLALC 
```
W (192957) wifi:sta_scan: STA is connecting, scan are not allowed!
E (192957) smartconfig: promis scan ret -1
```
When the smartconfig starts, it will scan firstly, if the device is connecting to the AP, then the scan will return fail, and show the above log. You should ensure that the STA is in a disconnected state when SmartConfig is started. 

By the way, I think call `esp_wifi_connect()` in the `WIFI_EVENT_STA_DISCONNECTED` event is not a good way, maybe this is the reason why the device was trying to connect when SmartConfig was started.

closed 2025-05-08 by Alvin1Zhang