espressif/esp-idf #11528: Why does gpio_get_level return an int and not a bool?

Gaai, 2023-05-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

```
 * @return
 *     - 0 the GPIO input level is 0
 *     - 1 the GPIO input level is 1
 *
 */
int gpio_get_level(gpio_num_t gpio_num);
```
Is there any good reason for this function returning an int rather than a bool that I am missing?

chipweinberger, 2023-05-29:
imo a book would not make sense.

the function is called "get level."

would true be high or low?

It would need to be renamed "is_level_high"

Gaai, 2023-05-29:
Different perspective I guess. I look at the possible outcomes and see only two: 0 or 1. That's a bool in my c and c++ mind. We are talking logical levels right. HIGH or LOW, true or false, 1 or 0. 

If I see a function returning an int I'd expect possible outcomes to be a range of values possibly even negative. Is this analog values you could wonder... Gets tiring pretty quickly if you need to look up every functions source code to find out what the return values actually are going to be. 

But maybe it's my lack of experience with the convenience in math you're talking about. 
✌️

Gaai, 2023-05-29:
So upon reading a little. I guess it has to do with implicit promotion rules. And confusion that might arise from that? Where if used in an expression, a bool is often implicitly promoted to int in the background. 
Or is it something else? 

Interesting read:
https://stackoverflow.com/questions/46073295/implicit-type-promotion-rules/46073296#46073296

0xjakob, 2023-05-31:
@Gaai You can have a look at [ESP-IDF-C++](https://github.com/espressif/esp-idf-cxx), where the type of [GPIOInput::get_level](https://github.com/espressif/esp-idf-cxx/blob/main/include/gpio_cxx.hpp#L318) is an [enum](https://github.com/espressif/esp-idf-cxx/blob/main/include/gpio_cxx.hpp#L74).