espressif/esp-idf #11530: adc_continuos.h no such file or directory

monurbstnc, 2023-05-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

When I try to use adc with tinyusb example I got error  
No such file or directory
   17 | #include "esp_adc/adc_continuous.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~

I have also tinyusb component in my project. Should I configure some settings? How should I solve this problem?
IDF version : 5.1

Thanks for your helps

suda-morris, 2023-05-29:
From which component are you initializing the ADC driver? Can you check if that component has added the `esp_adc` as its requirement? usually, it will look like this:

```
idf_component_register(SRCS ${srcs}
                       INCLUDE_DIRS ${includes}
                       REQUIRES driver esp_adc)
```

monurbstnc, 2023-05-29:
> From which component are you initializing the ADC driver? Can you check if that component has added the `esp_adc` as its requirement? usually, it will look like this:
>
> ``` [...]

Thank you ıt worked for me

closed 2023-05-30 by suda-morris