espressif/esp-idf #11531: Wrong JTAG PID for ESP32H2 under usb bridge v5.1

RJSDevel, 2023-05-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom

### Power Supply used.

USB

### What is the expected behavior?

Starts debug

### What is the actual behavior?

but in openOCD's script was wtritten esp_usb_jtag: VID set to 0x303a and PID to 0x1002, but chip gived PID == 0x1001

esp_usb_bridge.cfg -> espusbjtag vid_pid 0x303a 0x1002
I've changed to 0x1001 but got an error 

Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x30a
Info : esp_usb_jtag: target chip id set to 16
Warn : Transport "jtag" was already selected
Flashing C:/Users/Developer/espressif/HA_on_off_light/build/bootloader/bootloader.bin at 0x0
** program_esp input args <0x0 verify> **
Info : esp_usb_jtag: serial (60:55:F9:F7:65:8A)
Error: esp_usb_jtag: not enough data to get header
** OpenOCD init failed **
shutdown command invoked
** Flashing Failed **
-1
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

### Steps to reproduce.

1. Step
2. Step
3. Step
...

igrr, 2023-05-29:
@RJSDevel Could you please post the command line you are using to launch OpenOCD? If you are doing it from Eclipse rather than from the command line, please describe the debug configuration settings you have.

RJSDevel, 2023-05-29:
I think this is my mistake, I mixed up usb bridge and builtin USB, They have a different order in the policy from C6.
I've changed to builtin USB and it's ok

closed 2023-05-29 by RJSDevel