espressif/esp-idf #11532: Doesn't work debigging on ESP32H2 under builtin USB-JTAG - Open On-Chip Debugger v0.12.0-esp32-20230419

RJSDevel, 2023-05-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom

### Power Supply used.

USB

### What is the expected behavior?

Start from default breakpoint

### What is the actual behavior?

Just flash and reset the chip

[LOGS HERE!](https://github.com/espressif/esp-idf/files/11590143/logs.log)

Open On-Chip Debugger v0.12.0-esp32-20230419 (2023-04-18-22:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Flashing C:/Users/Developer/espressif/HA_on_off_light/build/bootloader/bootloader.bin at 0x0
** program_esp input args <0x0 verify> **
Info : esp_usb_jtag: serial (60:55:F9:F7:65:8A)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32h2.cpu tap/device found: 0x00010c25 (mfg: 0x612 (Espressif Systems), part: 0x0010, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40903105
Info : starting gdb server for esp32h2 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32h2.cpu tap/device found: 0x00010c25 (mfg: 0x612 (Espressif Systems), part: 0x0010, ver: 0x0)
Info : Reset cause (24) - (JTAG CPU reset)
Info : [esp32h2] Found 4 triggers
Info : Flash mapping 0: 0x10020 -> 0x42060020, 52 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 363 KB
Info : Auto-detected flash bank 'esp32h2.flash' size 4096 KB
Info : Using flash bank 'esp32h2.flash' size 4096 KB
** Programming Started **
Info : PROF: Erased 24576 bytes in 380.001 ms
Info : PROF: Data transferred in 271.999 ms @ 88.2356 KB/s
Info : PROF: Wrote 24576 bytes in 694.999 ms (data transfer time included)
** Programming Finished in 1874 ms **
** Verify Started **
Info : PROF: Flash verified in 207.999 ms 
** Verify OK **
** Flashing done for bootloader/bootloader.bin in 2083 ms **
Flashing C:/Users/Developer/espressif/HA_on_off_light/build/light_bulb.bin at 0x10000
** program_esp input args <0x10000 verify> **
Info : JTAG tap: esp32h2.cpu tap/device found: 0x00010c25 (mfg: 0x612 (Espressif Systems), part: 0x0010, ver: 0x0)
Info : Reset cause (24) - (JTAG CPU reset)
** Programming Started **
Info : PROF: Erased 483328 bytes in 1496 ms
Info : PROF: Data transferred in 6113.53 ms @ 77.2058 KB/s
Info : PROF: Wrote 483328 bytes in 6554.08 ms (data transfer time included)
** Programming Finished in 8313 ms **
** Verify Started **
Info : PROF: Flash verified in 684.999 ms 
** Verify OK **
** Flashing done for light_bulb.bin in 9005 ms **
Flashing C:/Users/Developer/espressif/HA_on_off_light/build/partition_table/partition-table.bin at 0x8000
** program_esp input args <0x8000 verify> **
Info : JTAG tap: esp32h2.cpu tap/device found: 0x00010c25 (mfg: 0x612 (Espressif Systems), part: 0x0010, ver: 0x0)
Info : Reset cause (24) - (JTAG CPU reset)
** Programming Started **
Info : PROF: Erased 4096 bytes in 248.001 ms
Info : PROF: Data transferred in 78.174 ms @ 51.1679 KB/s
Info : PROF: Wrote 4096 bytes in 500.568 ms (data transfer time included)
** Programming Finished in 1054 ms **
** Verify Started **
Info : PROF: Flash verified in 226.001 ms 
** Verify OK **
** Flashing done for partition_table/partition-table.bin in 1281 ms **
** Total programming time 12371 ms **
** Resetting Target **
Info : JTAG tap: esp32h2.cpu tap/device found: 0x00010c25 (mfg: 0x612 (Espressif Systems), part: 0x0010, ver: 0x0)
Info : Reset cause (24) - (JTAG CPU reset)
shutdown command invoked

### Steps to reproduce.

1. Step
2. Step
3. Step
...

gerekon, 2023-05-30:
Hi @RJSDevel!

Did you pressed `Debug` button or `Run`? In the latter case the app is just written to the target and chip is reset to run it.

kolipakakondal, 2023-05-31:
Hi @RJSDevel As @gerekon mentioned above, you can need to create a debug configuration for the project and choose Debug launch mode to start with the debugging.

Please follow the documentation from [here](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/OpenOCD%20Debugging.md)

> Please follow the below steps to create a new debug configuration.
> 
> 1. Right-click on the project
> 2. Debug As > Debug Configurations... This will launch a debug configuration window
> 3. On the left Panel, choose ESP-IDF GDB OpenOCD Debugging
> 4. Right Click and create New Configuration This will create a new debug configuration for your project
>

RJSDevel, 2023-05-31:
> Hi @RJSDevel As @gerekon mentioned above, you can need to create a debug configuration for the project and choose Debug launch mode to start with the debugging.
>
> Please follow the documentation from [...]

Thank you very much! It wasn't obviously, I'm used to that in Eclipse has default debug configuration

closed 2023-05-31 by RJSDevel