espressif/esp-idf #11537: UART parser not able to parse data correctly

krupis, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have been scratching my head regarding this issue for almost a whole day now. I cannot wrap my head around how is this even possible. I beg for advice!

In my program, I have a simple UART0 task running to receive incoming data:

```
void UART0_setup() {
    uart_config_t uart_config = {
        .baud_rate           = 115200,
        .data_bits           = UART_DATA_8_BITS,
        .parity              = UART_PARITY_DISABLE,
        .stop_bits           = UART_STOP_BITS_1,
        .flow_ctrl           = UART_HW_FLOWCTRL_DISABLE,
        //.rx_flow_ctrl_thresh = 122,
        .use_ref_tick        = false,
    };

    // Configure UART parameters
    ESP_ERROR_CHECK(uart_param_config(UART_NUM_0, &uart_config));
    // Set UART pins(TX: IO4, RX: IO5, RTS: IO18, CTS: IO19)
    //ESP_ERROR_CHECK(uart_set_pin(UART_NUM_0, 13, 26, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE));
    ESP_ERROR_CHECK(uart_driver_install(UART_NUM_0, UART0_COMMAND_LINE_MAX_SIZE, 0, 0, NULL, 0));
}

void UART0_task(void *argument)
{
	UART0_setup();
  	unsigned int char_received=EOF;
  	unsigned int char_counter=0;
  	char command_line[UART0_COMMAND_LINE_MAX_SIZE];
  	for (;;)
	{	
		int len = uart_read_bytes(UART_NUM_0, command_line, (UART0_COMMAND_LINE_MAX_SIZE - 1), 10 / portTICK_PERIOD_MS);
        if (len) {
            command_line[len] = 0;
			ParseSystemCmd(command_line, len); // Line is complete. Execute it!
			memset(&command_line, 0, sizeof(command_line));
        }
		vTaskDelay(10/portTICK_PERIOD_MS);
  }
}
```

As you can see when the command via serial is received, a ParseSystemCmd function is called. ParseSystemCmd will simply try to find a match and perform some actions. If no match is found, "System cmd not found" will be printed. See the function below:

```
bool ParseSystemCmd(char *line, uint16_t cmd_size)
{
	if (!strncmp("app_adc_calib:", line,14)){	
		if (!isdigit(line[14]))
		{
			ESP_LOGW(TAG,"Entered input is not a number");
			return 0;
		}
		char temp_buf[4];
		double temp_to_calib=0;
		for(int i = 0;i <=(strlen(line)-14);i++){
			temp_buf[i] = line[14+i];
		}
		temp_to_calib = strtof(temp_buf,NULL);
		printf("Temp to calib = %.2f \n",temp_to_calib);
		return 1;

	}

	if (!strncmp("compressor:", line,11)){	
		if (!isdigit(line[11]))
		{
			ESP_LOGW(TAG,"Entered input is not a number");
			return 0;
		}
		char temp_buf_comp[1];
		uint16_t data_to_write=0;
		temp_buf_comp[0] = line[11];
		data_to_write = atoi(temp_buf_comp);
		printf("DATA_TO_WRITE = %u \n",data_to_write);

		if(data_to_write == 0){
			gpio_set_level(GPIO_COMPRESSOR, 0);   
			printf("compressor:0\n"); 
			return true;
		}
		else if(data_to_write == 1){
			gpio_set_level(GPIO_COMPRESSOR, 1);    
			printf("compressor:1\n"); 
			return true;
		}
		else if(data_to_write != 0 || data_to_write !=1){
			printf("incorrect argument \n");
			return false;
		}
	}

	if (!strncmp("modem_version", line,13))
	{
		char Resp[AT_CMD_MAX_LEN];
		bool ret;
		ret=ModemSendATCommandRC("AT+CGMR\r", 10000, 2, 1, Resp, 20);
		if(!ret)
		{
			ESP_LOGE(TAG, "Modem FW ver: Unknown!");
			return false;
        }
        printf("%s\n", Resp);
		return true;
	}

	printf(CONF_ERROR_HEADER "System cmd not found: %s\r", line);
	memset(&line, 0, 1024);
	return false;
}
```

As you can see from above I only have a few commands implemented.

The main commands that I use are the following:
```
compressor:1
```
and
```
compressor:0
```

That will turn ON/OFF the relays. The command works without any issues. I use "Termite" to write commands to the ESP32 device:

[image]

Something very weird happens when I execute "app_adc_calib:"  command. Please see the logs below:

[image]

My UART0, after receiving app_adc_calib:15.0 command starts behaving oddly. It no longer accepts the compressor:1 or compressor:0 command.

```
	if (!strncmp("compressor:", line,11)){	
		if (!isdigit(line[11]))
		{
			ESP_LOGW(TAG,"Entered input is not a number");
			return 0;
		}
		char temp_buf_comp[1];
		uint16_t data_to_write=0;
		temp_buf_comp[0] = line[11];
		data_to_write = atoi(temp_buf_comp);
		printf("DATA_TO_WRITE = %u \n",data_to_write);

		if(data_to_write == 0){
			gpio_set_level(GPIO_COMPRESSOR, 0);   
			printf("compressor:0\n"); 
			return true;
		}
		else if(data_to_write == 1){
			gpio_set_level(GPIO_COMPRESSOR, 1);    
			printf("compressor:1\n"); 
			return true;
		}
		else if(data_to_write != 0 || data_to_write !=1){
			printf("incorrect argument \n");
			return false;
		}
	}
```

As you can see from the code above, when the UART matches the word "compressor:", it looks at the 11th byte (which is going to be 0 or 1) but for some reason it detects 15. How is that even possible? My mind is blown.

krupis, 2023-05-30:
**UPDATE**

In ParseSystemCmd function I added the following code to print out the contents of received UART buffer:

```
	for(int i = 0; i <cmd_size;i++){
		printf("Character received[%i]=%c\n",i,line[i]);
	}
```

Serial terminal output when sending commands:

```
compressor:1
Character received[0]=c
Character received[1]=o
Character received[2]=m
Character received[3]=p
Character received[4]=r
Character received[5]=e
Character received[6]=s
Character received[7]=s
Character received[8]=o
Character received[9]=r
Character received[10]=:
Character received[11]=1
Character received[12]=

DATA_TO_WRITE = 1 
compressor:1
app_adc_calib:12.0
Character received[0]=a
Character received[1]=p
Character received[2]=p
Character received[3]=_
Character received[4]=a
Character received[5]=d
Character received[6]=c
Character received[7]=_
Character received[8]=c
Character received[9]=a
Character received[10]=l
Character received[11]=i
Character received[12]=b
Character received[13]=:
Character received[14]=1
Character received[15]=2
Character received[16]=.
Character received[17]=0
Character received[18]=

Temp to calib = 12.00 
compressor:1
Character received[0]=c
Character received[1]=o
Character received[2]=m
Character received[3]=p
Character received[4]=r
Character received[5]=e
Character received[6]=s
Character received[7]=s
Character received[8]=o
Character received[9]=r
Character received[10]=:
Character received[11]=1
Character received[12]=

DATA_TO_WRITE = 12 
incorrect argument 
```

It seems that it is printing correct data, but I still cannot understand why DATA_TO_WRITE is not being set incorrectly using atoi() function

igrr, 2023-05-30:
atoi works with 0-terminated strings. The way you declare
```
		char temp_buf_comp[1];
```
it can only hold a zero-length string, since one byte is needed for the terminator.

So
```
		temp_buf_comp[0] = line[11];
```
writes the only byte in the string, leaving it unterminated, and
```
		data_to_write = atoi(temp_buf_comp);
```
tries to parse the string, which is now unterminated, so atoi just keeps reading subsequent bytes in memory (on the stack).

krupis, 2023-05-30:
> atoi works with 0-terminated strings. The way you declare
> 
> ```
> 		char temp_buf_comp[1];
> ```
> 
> it can only hold a zero-length string, since one byte is needed for the terminator.
> 
> So
> 
> ```
> 		temp_buf_comp[0] = line[11];
> ```
> 
> writes the only byte in the string, leaving it unterminated, and
> 
> ```
> 		data_to_write = atoi(temp_buf_comp);
> ```
> 
> tries to parse the string, which is now unterminated, so atoi just keeps reading subsequent bytes in memory (on the stack).
> 
> This happens to work as long as the next byte on the stack after the end of `temp_buf_comp` is zero. My guess is the next value on the stack is `temp_to_calib`, so once it gets set, you start observing this issue.

Thank you very much for pointing this out. I should have realized that myself.
So what would be correct way to do this? Since I am only interested in a single byte after "compressor:", should I manually add null terminator at ```temp_buf_comp[1]```?

```
		char temp_buf_comp[2];
		uint16_t data_to_write=0;
		temp_buf_comp[0] = line[11];
		temp_buf_comp[1] = '\0';
		data_to_write = atoi(temp_buf_comp);
		printf("DATA_TO_WRITE = %u \n",data_to_write);
```

The code above seems to fix my issue. Is that a good way to deal with it?

igrr, 2023-05-30:
I think that's okay, yes.

(Other than the fact that manual string handling using C string.h APIs is very error-prone, as you have found. On top of that you now have magic values, such as `11` or `14` scattered around your code. If you don't have a hard requirement for using this exact format of the commands, I would recommend checking out the [console component](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/console.html) which provides command line parsing and command registration functionality. There's an [example on how to start the console](https://github.com/espressif/esp-idf/blob/master/examples/system/console/basic/main/console_example_main.c) and how to [implement a command](https://github.com/espressif/esp-idf/blob/903af13e847cd301e476d8b16b4ee1c21b30b5c6/examples/system/console/advanced/components/cmd_system/cmd_system_common.c#L213-L274).)

closed 2025-01-08 by Alvin1Zhang