espressif/esp-idf #11538: Unable to receive more than one L2CAP COC packet

gomatteo, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-beta1-378-gea5e0ff298

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Olimex ESP32-GATEWAY

### Power Supply used.

USB

### What is the expected behavior?

Using the coc_plecent and coc_bleprph examples I expected to be able to send and receive more than one packet

### What is the actual behavior?

The peripheral is able to receive only the first packet before crashing

### Steps to reproduce.

1. modify coc_blecent to send more than one packet
```patch
diff --git a/main/main.c b/main/main.c
index 91fd3ae..d20e615 100644
--- a/main/main.c
+++ b/main/main.c
@@ -38,13 +38,13 @@ static int blecent_l2cap_coc_event_cb(struct ble_l2cap_event *event, void *arg);
  */
 static void
 blecent_l2cap_coc_send_data(struct ble_l2cap_chan *chan)
 {
     struct os_mbuf *sdu_rx_data;
     int rc = 0;
-    int len = 512;
+    int len = 32;
     uint8_t value[len];

     for (int i = 0; i < len; i++) {
         value[i] = i;
     }

@@ -111,12 +111,14 @@ blecent_l2cap_coc_event_cb(struct ble_l2cap_event *event, void *arg)
                        " peer_mps: %d, peer_mtu: %d\n",
                        event->connect.conn_handle, event->connect.chan,
                        chan_info.psm, chan_info.scid, chan_info.dcid,
                        chan_info.our_l2cap_mtu, chan_info.our_coc_mtu,
                        chan_info.peer_l2cap_mtu, chan_info.peer_coc_mtu);
         blecent_l2cap_coc_send_data(event->connect.chan);
+        blecent_l2cap_coc_send_data(event->connect.chan);
+        blecent_l2cap_coc_send_data(event->connect.chan);
         return 0;

     case BLE_L2CAP_EVENT_COC_DISCONNECTED:
         console_printf("LE CoC disconnected, chan: %p\n",
                        event->disconnect.chan);
         return 0;
```

### Debug Logs.

```plain
--------- log of coc_ble coc_bleprph ---------

I (940) NimBLE: GAP procedure initiated: advertise;
I (940) NimBLE: disc_mode=2
I (940) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (950) NimBLE:

I (2590) NimBLE: connection established; status=0
I (2590) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
I (2590) NimBLE: 3c:71:bf:0c:9d:62
I (2590) NimBLE:  our_id_addr_type=0 our_id_addr=
I (2600) NimBLE: 3c:71:bf:0c:9d:62
I (2600) NimBLE:  peer_ota_addr_type=0 peer_ota_addr=
I (2610) NimBLE: 3c:71:bf:0c:9f:86
I (2610) NimBLE:  peer_id_addr_type=0 peer_id_addr=
I (2620) NimBLE: 3c:71:bf:0c:9f:86
I (2620) NimBLE:  conn_itvl=40 conn_latency=0 supervision_timeout=256 encrypted=0 authenticated=0 bonded=0

I (2630) NimBLE:

I (2640) NimBLE: handle=0 our_ota_addr_type=0 our_ota_addr=
I (2640) NimBLE: 3c:71:bf:0c:9d:62
I (2650) NimBLE:  our_id_addr_type=0 our_id_addr=
I (2650) NimBLE: 3c:71:bf:0c:9d:62
I (2660) NimBLE:  peer_ota_addr_type=0 peer_ota_addr=
I (2660) NimBLE: 3c:71:bf:0c:9f:86
I (2670) NimBLE:  peer_id_addr_type=0 peer_id_addr=
I (2670) NimBLE: 3c:71:bf:0c:9f:86
I (2680) NimBLE:  conn_itvl=40 conn_latency=0 supervision_timeout=256 encrypted=0 authenticated=0 bonded=0

LE CoC accepting, chan: 0x3ffc3634, peer_mtu 512
LE COC connected, conn: 0, chan: 0x3ffc3634, psm: 0x1002, scid: 0x0040, dcid: 0x0040, our_mps: 248, our_mtu: 512, peer_mps: 248, peer_mtu: 512
I (2850) NimBLE: Data received :
0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e521f  PS      : 0x00060630  A0      : 0x800d7f45  A1      : 0x3ffcc020
0x400e521f: ble_l2cap_coc_rx_fn at /home/matteo/esp/esp-idf_v5/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap_coc.c:200
A2      : 0x3ffc3644  A3      : 0x3ffc3634  A4      : 0x00000022  A5      : 0x3ffcc054
A6      : 0x3ffc97e0  A7      : 0xb33fffff  A8      : 0x800d7592  A9      : 0x3ffcc010
A10     : 0x00000000  A11     : 0x3ffc97e0  A12     : 0x3ffc990c  A13     : 0x3ffcc054
A14     : 0x3ffc459c  A15     : 0xb33fffff  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

Backtrace: 0x400e521c:0x3ffcc020 0x400d7f42:0x3ffcc050 0x400d759d:0x3ffcc080 0x400d75af:0x3ffcc0a0 0x40090929:0x3ffcc0c0 0x40083c9d:0x3ffcc0e0 0x400d677a:0x3ffcc100 0x40095199:0x3ffcc120
0x400e521c: ble_l2cap_coc_rx_fn at /home/matteo/esp/esp-idf_v5/components/bt/host/nimble/nimble/nimble/host/src/ble_l2cap_coc.c:197
0x400d7f42: ble_hs_hci_evt_acl_process at /home/matteo/esp/esp-idf_v5/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_hci_evt.c:1063
0x400d759d: ble_hs_process_rx_data_queue at /home/matteo/esp/esp-idf_v5/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:241
0x400d75af: ble_hs_event_rx_data at /home/matteo/esp/esp-idf_v5/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:538
0x40090929: npl_freertos_event_run at /home/matteo/esp/esp-idf_v5/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:450
0x40083c9d: ble_npl_event_run at /home/matteo/esp/esp-idf_v5/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at /home/matteo/esp/esp-idf_v5/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:246
0x400d677a: bleprph_host_task at /home/matteo/p2/main/main.c:420
0x40095199: vPortTaskWrapper at /home/matteo/esp/esp-idf_v5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 2a606548aff290da

Rebooting...
```

dfacchin, 2023-05-31:
I'm planning to use L2CAP CoC transmission, @gomatteo or anyone that finds a solution please share how. Thanks

IshaESP, 2023-06-06:
Hi,
This issue was reproduced at out end. The crash was on this [line](https://github.com/espressif/esp-nimble/blob/ca668cf0aa11704f3fde4b641cb2a30758792975/nimble/host/src/ble_l2cap_coc.c#L200) as rx->sdu is NULL. rx->sdu is allocated during connection establishment procedure. Once data is received, this [line](https://github.com/espressif/esp-nimble/blob/ca668cf0aa11704f3fde4b641cb2a30758792975/nimble/host/src/ble_l2cap_coc.c#L278) makes it NULL. As the same behavior exists in upstream code as well, we are checking on it.

IshaESP, 2023-06-14:
Hey,
Can you please try out the below patch and see if it works?
To apply these patches:
1. cd esp-idf
2. cd components/bt/host/nimble/nimble/
3. git apply nimble.patch
4. cd esp-idf
5. git apply idf.patch
[patches.zip](https://github.com/espressif/esp-idf/files/11742139/l2cap_more_pkts.zip)

gomatteo, 2023-06-19:
Hi, I applied the patch.
I did a quick test and it seems to work. I will do more in-depth tests in the next few days.
Thank you!

closed 2023-08-04 by rahult-github