espressif/esp-idf #11539: 5.1-beta1 Beta does not include a fix #56123 from issue #10595

MacWyznawca, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-beta1-dirty

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

Battery

### What is the expected behavior?

With ULP-RISC-V ADC Example currewnt draw < 100 uA

### What is the actual behavior?

Current > 1 mA

### Steps to reproduce.

Compile and run:
ULP-RISC-V ADC Example

### More Information.

After add patch `adc_power_fix_56123.patch.txt` current consumption is as it should be.

igrr, 2023-05-30:
Thanks for the reminder @MacWyznawca. The fix has already been backported to release/v5.1 branch in commit https://github.com/espressif/esp-idf/commit/ed32d7a267b41d1d38106303850e54befde52e0c. This happened after the v5.1-beta1 was handed over to QA, so the fix is not included into the release. The fix will be included into v5.1-beta2 or v5.1-rc1, whichever comes next. Note that there is a side effect of this fix which was discovered later, that the ADC sampling rate becomes lower after the fix. We'll try to get that fixed before the final v5.1 release, as well.

AxelLin, 2023-10-03:
> Will keep open until [3e26e76](https://github.com/espressif/esp-idf/commit/3e26e7630e28009434009064532cfdf54638ab9b) is backported, as well.

v5.1 fix: 6f673e95c9583e3603a5c3674f0d704e0c38fdb0

AxelLin, 2023-12-10:
> Will keep open until [3e26e76](https://github.com/espressif/esp-idf/commit/3e26e7630e28009434009064532cfdf54638ab9b) is backported, as well.

This should be closed now.

closed 2023-12-11 by Alvin1Zhang