espressif/esp-idf #11540: fatal error: semaphore.h

RuslanSpartak, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-823-g903af13e84-dirty

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board with esp32

### Power Supply used.

External 5V

### What is the expected behavior?

Project is built

### What is the actual behavior?

Can't build the project

### Steps to reproduce.

1. Just try to build project

### More Information.

[image]

igrr, 2023-05-30:
@RuslanSpartak Could you please clarify this part of the issue report:
> Steps to reproduce.
> Just try to build project

Which project do you get this error with? Can the error be reproduced using one of the example projects in ESP-IDF? If the error can only be reproduced with a custom project, please attach enough information to allow us to reproduce the issue.
For example, you may attach the relevant part of ota.cpp.

Also this part: 

> Custom board with esp32

Is that really the case? The logs mention RISC-V toolchain, which is not supposed to be used for the ESP32.

0xjakob, 2023-05-31:
@RuslanSpartak would you mind trying the following patch?
[0001-cxx-made-pthread-a-public-dependency-of-cxx.zip](https://github.com/espressif/esp-idf/files/11608612/0001-cxx-made-pthread-a-public-dependency-of-cxx.zip)

We have an issue that looks like it's related and we're working on fixing it.

RuslanSpartak, 2023-05-31:
> @RuslanSpartak Could you please clarify this part of the issue report:
> 
> > Steps to reproduce.
> > Just try to build project
> I try to build project with latest master sdk and I have such kind of issue
> Which project do you get this error with? Can the error be reproduced using one of the example projects in ESP-IDF? If the error can only be reproduced with a custom project, please attach enough information to allow us to reproduce the issue. For example, you may attach the relevant part of ota.cpp.
> it is master branch sdk
> Also this part:
> 
> > Custom board with esp32
> 
> Is that really the case? The logs mention RISC-V toolchain, which is not supposed to be used for the ESP32. 
> It is esp32-c6

RuslanSpartak, 2023-05-31:
> @RuslanSpartak would you mind trying the following patch? [0001-cxx-made-pthread-a-public-dependency-of-cxx.zip](https://github.com/espressif/esp-idf/files/11608612/0001-cxx-made-pthread-a-public-dependency-of-cxx.zip)
> 
> We have an issue that looks like it's related and we're working on fixing it.

Could you tell me how I need to use that file?

0xjakob, 2023-05-31:
@RuslanSpartak The comments from @igrr are still valid, actually. When investigating issue reports, more specific information is very helpful for us to re-create the issue on our side. It's coincidence that we discovered the same (or similar looking) error in a different context just today and could come back quickly here.

> Could you tell me how I need to use that file?

Unzip it. Then use `git apply 0001-cxx-made-pthread-a-public-dependency-of-cxx.patch` (on the command line) inside the IDF repository to add the patch to IDF. Unfortunately, I'm not very familiar how to do that on Windows.

Alternatively, you can go to https://github.com/espressif/esp-idf/blob/master/components/cxx/CMakeLists.txt#L4 (in your IDF repository) and change the line to:  `REQUIRES pthread)`.

RuslanSpartak, 2023-05-31:
> REQUIRES

Thanks. I know how to do it. It worked. do you woek with [igrr](https://github.com/igrr)?

RuslanSpartak, 2023-05-31:
> @RuslanSpartak Could you please clarify this part of the issue report:
>
>
> Which project do you get this error with? Can the error be reproduced using one of the example projects in ESP-IDF? If the [...]

Fix from [0xjakob](https://github.com/0xjakob) is working.

closed 2023-05-31 by RuslanSpartak