espressif/esp-idf #11542: Use slave i2c 10 bit

playmiel, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've used this program to receive i2c data with a 10-bit address, but I'm not receiving anything at all, even though everything is initialized correctly.

```
#include <Arduino.h>
#include "driver/i2c.h"

#define I2C_SLAVE_NUM I2C_NUM_1   // I2C port number for slave dev
#define I2C_SLAVE_SCL_IO  32   // gpio number for I2C slave clock
#define I2C_SLAVE_SDA_IO 33    // gpio number for I2C slave data 
#define I2C_SLAVE_FREQ_HZ 32000    // I2C slave clock frequency
#define RX_BUF_SIZE 1024   // Size of the RX buffer
#define TX_BUF_SIZE 1024   // Size of the TX buffer
#define HIGH_ADDR 0x78
#define LOW_ADDR 0xF0
#define  SLAVE_ADDR ((HIGH_ADDR << 8) | LOW_ADDR)

void setup() {
  Serial.begin(115200); // Initialize serial communication
while(!Serial); // Wait for the serial connection to be established
delay(15000);
Serial.println("I2C Slave");
  i2c_config_t conf;
  conf.mode = I2C_MODE_SLAVE;
  conf.sda_io_num = I2C_SLAVE_SDA_IO;
  conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
  conf.scl_io_num = I2C_SLAVE_SCL_IO;
  conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
  conf.slave.addr_10bit_en = 1; // Enable 10-bit slave address
  conf.slave.slave_addr = SLAVE_ADDR; // 10-bit address
esp_err_t ret;

ret = i2c_param_config(I2C_SLAVE_NUM, &conf);
if(ret != ESP_OK){
  Serial.println("Failed to configure I2C");
  Serial.println(esp_err_to_name(ret)); // Print the error name
}else{
  Serial.println("I2C configured successfully");
}

ret = i2c_driver_install(I2C_SLAVE_NUM, conf.mode, RX_BUF_SIZE, TX_BUF_SIZE, ESP_INTR_FLAG_IRAM);
if(ret != ESP_OK){
  Serial.println("Failed to install I2C driver");
  Serial.println(esp_err_to_name(ret)); // Print the error name
}else{
  Serial.println("I2C driver installed successfully");
}
}

void loop() {
  uint8_t data[RX_BUF_SIZE];
  int size = i2c_slave_read_buffer(I2C_SLAVE_NUM, data, RX_BUF_SIZE, 1000 / portTICK_RATE_MS);
  if (size > 0) {
    for(int i = 0; i < size; i++) {
      Serial.print(data[i], HEX); // Print each byte in HEX
      Serial.print(" "); // Print a space between each byte
    }
    Serial.println(); // Print a newline at the end

    // Echo back the data
    i2c_slave_write_buffer(I2C_SLAVE_NUM, data, size, 1000 / portTICK_RATE_MS);
  }
}

```
as the 10-bit i2c is not widely used, I couldn't see if anyone had managed to get it to work in slave mode.

mythbuster5, 2023-05-31:
Yes, 10-bit address as a i2c  new feature is already in progress in our new i2c driver. So, hoping that you can use that early. Now I think this issue can be closed 😀

mythbuster5, 2025-01-13:
10-bit is online now,

closed 2025-01-13 by mythbuster5