espressif/esp-idf #11543: Fix for data corruption in Mbedtls bignum not backported to ESP32S3

QuentinFarizon, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitM1

### Power Supply used.

USB

### What is the expected behavior?

I think https://github.com/espressif/esp-idf/commit/84b0254fbf4cd06ed74a8e27106c91cbf06a3ee0 should be backported to ESP-32S3 (components/mbedtls/port/esp32s3/bignum.c), and from what I see, other targets too (h2, s2, ...)

### What is the actual behavior?

Fix only done for ESP32

### Steps to reproduce.

1. Step
2. Step
3. Step
...

igrr, 2023-05-30:
@QuentinFarizon Do you observe some related issue on the ESP32-S3?

As far as I can tell, the linked commit is specific to ESP32 architecture, where the MPI peripheral is on a dedicated CPU interface (called DPORT). On newer chips, MPI and other cryptographic peripherals are on the regular peripheral bus and are not subject to issues with speculative execution which DPORT has.

laukik-hase, 2023-06-19:
Hello, @QuentinFarizon!

Can you reproduce the issues fixed by the said commit on ESP32-S3? If yes, could you please provide the following:
- Example to reproduce the issue (or a minimal code snippet)
- Project sdkconfig
- Debug logs

QuentinFarizon, 2023-06-19:
Hello, sorry for the late response

I did not see an specific issue on ESP32-S3, I was just curious to know if it was intended that it was not backported. If it is indeed backported, we can close this issue

laukik-hase, 2023-06-19:
As @igrr mentioned, the issue was specific to ESP32 and hence, there was no need to backport it.

Closing the issue, thank you!

closed 2023-06-19 by laukik-hase