espressif/esp-idf #11544: Can't reuse a esp_http_client_handle_t after some calls errored

QuentinFarizon, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I create a esp_http_client_handle_t once, and then reuses it to make calls to esp_http_client_perform (after changing data with esp_http_client_set_post_field)

When there are network errors (I unplug my internet box), I get ESP_ERR_HTTP_EAGAIN as expected.   
But when network is back, I continue to get ESP_ERR_HTTP_EAGAIN on this https client. 
I would expect the client to reopen the connection transparently, and to start performing OK when network is back.

https client created after network is back perform correctly.

Is that expected ? If I need to detect network error or connection closed, or other, what should be the strategy to reopen the connection, or to re-create the https client ?

### What is the actual behavior?

I can't reuse an esp_http_client_handle_t after some calls errored

### Steps to reproduce.

```
// Called once at device initialization
void logs_http_init(void)
{
    esp_http_client_config_t http_client_config = {
        .url = url,
        .event_handler = _http_event_handler,
        .auth_type = HTTP_AUTH_TYPE_BASIC,
        .cert_pem = (char *) server_root_cert_pem_start,
    };
    client = esp_http_client_init(&http_client_config);
    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_header(client, "Content-Type", "application/json");
}

// Called periodically
bool send_logs_http(char *pvParameters)
{
    esp_http_client_set_post_field(client, (char *)pvParameters, strlen(pvParameters));
    esp_err_t err = esp_http_client_perform(client);
 ```

Alvin1Zhang, 2023-05-31:
Thanks for reporting.

hmalpani, 2023-07-03:
Hello @QuentinFarizon 
Apologies for the delayed reply.
It is an expected behavior. You can check for network errors using the event callbacks. It should look like this:
```
static void wifi_event_handler(void *arg, esp_event_base_t event_base,
                               int32_t event_id, void *event_data)
{
    if (event_id == WIFI_EVENT_AP_STACONNECTED) {
        wifi_event_ap_staconnected_t *event = (wifi_event_ap_staconnected_t *)event_data;
        ESP_LOGI(TAG, "station " MACSTR " join, AID=%d",
                 MAC2STR(event->mac), event->aid);
    } else if (event_id == WIFI_EVENT_AP_STADISCONNECTED) {
        // Clean-up the esp_http_client handler
    }
}
```

You can also use `esp_http_client_cancel_request()` API to use the same client handler in case any network error occurs.

hmalpani, 2023-07-12:
Hello @QuentinFarizon 
Have you tried the above suggestion? Did it work for you?

hmalpani, 2023-07-17:
Closing this issue. Please feel free to reopen.

closed 2023-07-17 by hmalpani