espressif/esp-idf #11545: Why is the ECCN for the ESP32-S3-MINI1N8 Module in Mouser 5A002.A?

NaivelyWritten, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This triggers export control reviews whenever we buy it and it is very cumbersome to deal with. Other modules does not have this classification and the same module is classified as 5A992.C on Digi-Key so I believe this is an error.

huming2207, 2023-05-31:
I guess this is because of the encryption algorithms. Probably when Espressif designs their chip, they bought those hardware crypto peripherals/IP cores from some US providers so then they need to follow the US regulations??

NaivelyWritten, 2023-05-31:
> I guess this is because of the encryption algorithms. Probably when Espressif designs their chip, they bought those hardware crypto peripherals/IP cores from some US providers so then they need to follow the US regulations??

**the same module is classified as 5A992.C on Digi-Key** and also the almost all dev kits are 5A992.C, so this has to be a mistake from Espressif. Mouser told me in the emails that the classification is provided by Espressif.

igrr, 2023-05-31:
Hi @NaivelyWritten, since this is an issue tracker of ESP-IDF project (not a discussion or support forum for every Espressif product), I'm afraid you won't find the answer here. Please use https://www.espressif.com/en/contact-us/sales-questions for this question.

NaivelyWritten, 2023-05-31:
Hi @igrr could you relay this question to the sales team? Last time I used that form nobody got in touch with me so I don't really trust that.

AxelLin, 2023-05-31:
@NaivelyWritten 
I post the same question on  esp32 forum, you may close this issue.
https://www.esp32.com/viewtopic.php?f=2&t=33907

Hope someone from espressif will clarify this.

igrr, 2023-05-31:
@NaivelyWritten Sorry that no one got back to you on that previous query. I'll follow up on that with the relevant team. And I've sent the link to this topic and to the forum post to the right people, they will take a look.

closed 2023-05-31 by igrr

NaivelyWritten, 2023-05-31:
Thank you @AxelLin and @igrr !

AxelLin, 2023-08-01:
@igrr 
There is no follow up about this question on the forum (https://www.esp32.com/viewtopic.php?f=2&t=33907) .
So how is the status to clarify the ECCN for esp32S3? (It's ok if you reply on the forum)

AxelLin, 2024-01-05:
@igrr
Sorry to bother you on this closed issue, but it has been a while and there is no response from the forum.
How is the status of ECCN for esp32S3?

igrr, 2024-01-24:
For anyone who happens to find his issue later, the original issue was resolved and the updated ECCN is 5A992.c, see the linked forum post.