espressif/esp-idf #11547: Inconsistent exception when appending file

marklangster, 2023-05-30:
The vfprintf function sometimes causes a exception when appending a file in V4.4

The stack trace return is:
`0x401b0539: _vfprintf_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1436

0x401a6471: _fopen_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:158 (discriminator 5)
`

Code to reproduce:

`FILE *file = fopen(path, "a");
fprintf(file, "%s", data);`

igrr, 2023-05-30:
Hi @marklangster,
Could you please provide more information so that we can help you troubleshoot the issue?

* ESP-IDF version (is it really v4.4 and not one of the bugfix releases, e.g. v4.4.3?)
* sdkconfig file, as an attachment
* Minimal code to reproduce the issue. (The two lines you have posted can't be compiled, because `path` and `data` aren't defined. If I make reasonable assumptions about what the `path` and `data` might be, then the same code exists in [numerous](https://github.com/espressif/esp-idf/blob/903af13e847cd301e476d8b16b4ee1c21b30b5c6/examples/storage/sd_card/sdmmc/main/sd_card_example_main.c#L28-L33) [places](https://github.com/espressif/esp-idf/blob/903af13e847cd301e476d8b16b4ee1c21b30b5c6/components/fatfs/test_apps/test_fatfs_common/test_fatfs_common.c#L263-L266) in our test suite, and doesn't fail, so clearly the scenario is not as simple as two lines of code.)
* The filesystem you are using (FAT in Flash, FAT on an SD card, SPIFFS, something else)
* Complete log output you get from the application. [Panic handler output](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#panic-handler) contains a lot of information, out of which you have shown only two lines. The reset of the information (starting from "Guru meditation error") can help understand the nature of the issue.

We ask for this information in the [issue template](https://github.com/espressif/esp-idf/blob/master/.github/ISSUE_TEMPLATE/02_runtime_bug.yml) which you can fill in when opening the issue.

marklangster, 2023-06-05:
@igrr 

Please note, I am using littlefs on top of a SPIFFS partition although that doesn't seem to be an issue as this seems to be an issue with the write command. I am trying without littlefs as a sanity check though. Also, this issue is very inconsistent but causes a panic and restart. Over the course of a 3-4 days, the issue may happen about once per day. Can you also point me to the implementation of the vfprintf function? I would like to see the line that is breaking.

1. IDF version is 4.4.3. OS - Mac OS. Built with command line idf.py. ESP-WROOM-32E module.
3. Attached [sdkconfig.zip](https://github.com/espressif/esp-idf/files/11649433/sdkconfig.zip)
4. Simplified code below. Please note that the failure point is AFTER a check for a successful opening of the file for appending.
5. SPIFFS with LittleFS

```
#include <esp_littlefs.h>
#include <esp_log.h>
#include <sys/stat.h>

#define BASE_PATH           "/store"
#define STRING_SIZE         30
#define PARTITION           "data_store"
#define TAG                 "DATA"

char path[30];

void initDataStorage(){
    esp_vfs_littlefs_conf_t conf = {
            .base_path = BASE_PATH,
            .partition_label = PARTITION,
            .format_if_mount_failed = true,
            .dont_mount = false
    };
    esp_err_t err = esp_vfs_littlefs_register(&conf);
    if (err != ESP_OK)
    {
        if (err == ESP_FAIL)
        {
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
        }
        else if (err == ESP_ERR_NOT_FOUND)
        {
            ESP_LOGE(TAG, "Failed to find LittleFS partition");
        }
        else
        {
            ESP_LOGE(TAG, "Failed to initialize LittleFS (%s)", esp_err_to_name(err));
        }
        return;
    }
    size_t total = 0, used = 0;
    err = esp_littlefs_info(conf.partition_label, &total, &used);
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to get LittleFS partition information (%s)", esp_err_to_name(err));
    }
    else
    {
        ESP_LOGI(TAG, "Partition size: total: %d, used: %d", total, used);
    }
}

void createNewFile(){
    uint8_t count;
    for (count = 0; count< 255; count++){
        struct stat st;
        snprintf(path, STRING_SIZE, "%s/s_%d.json", BASE_PATH, count);
        if (stat(path, &st) != 0){
           break;
        }
    }

    FILE *file = fopen(path, "w");
    ESP_LOGI(TAG, "File open");

    if (file == NULL)
    {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    fclose(file);

}

void store(char* data){
    FILE *file = fopen(path, "a");
    if (file == NULL)
    {
        ESP_LOGE(TAG, "Failed to open file for appending");
        return;
    }
   
    fprintf(file, ",%s", data); //Breaking line
    

    fclose(file);
    ESP_LOGI(TAG, "File at %s appended", path);

}

```

**Stack Trace**

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400014fd  PS      : 0x00060330  A0      : 0x801b053c  A1      : 0x3ffcdb70  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffcdb30  
A10     : 0x3ffcdb50  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x3ffcdd30  
A14     : 0x0000002d  A15     : 0x3ffcdd30  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fa:0x3ffcdb70 0x401b0539:0x3ffcdb80 0x401a6471:0x3ffcde90 0x400d9474:0x3ffcdee0 0x400d9537:0x3ffcdf20 0x400d98ae:0x3ffcdfb0 0x400d8a3e:0x3ffcdfd0 0x400d8e0d:0x3ffce010 0x40096241:0x3ffce040
0x401b0539: _vfprintf_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1436

0x401a6471: _fopen_r at /Users/brnomac003/.gitlab-runner/builds/qR2TxTby/0/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fopen.c:158 (discriminator 5)

```

marklangster, 2023-06-05:
File appends are occurring about once every 10 minutes

marklangster, 2023-06-07:
@igrr Can you point me in the direction of the vfprintf implementation?

AxelLin, 2024-03-26:
@marklangster Do you still hit this issue in recent esp-idf and littlefs releases?

marklangster, 2024-03-26:
No. This was actually an issue with a memory leak in another library. I'm not sure why it manifested this way but it has been fixed.

closed 2024-03-26 by marklangster