espressif/esp-idf #11548: I2C STOP not being applied correctly (SCL stuck LOW) after upgrade from IDF v4.4.x to IDF v5.0.x

abellofiore, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Since upgrading to IDF v5.0.1 (and v5.0.2) from IDF v4.4.3, I am encountering situations where the ESP32-C3 is NOT terminating a bus transaction with a STOP when a NAK is encountered (using "driver/i2c.c").  I can freely switch between IDF builds to prove this.

My setup is an ESP32-C3 with a single I2C device (Pericom PI4IOE5V6416 I/O expander, I2C address=0x20) on the bus.  Both SDA and SCL have external 4.7K pull-ups.

The issue can be reproduced by performing an I2C bus scan using the code in the "i2c_tools" example project; this performs a write to each I2C address without sending any data and looking for an ACK/NAK (it also happens after ANY access to a valid I2C device, but this example will suffice):

i.e.:

```
#include "driver/i2c.h"

static int do_i2cdetect_cmd(int argc, char **argv)
{
    int i2c_master_port = I2C_NUM_0;
    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = I2C_MASTER_SDA_IO,
        .scl_io_num = I2C_MASTER_SCL_IO,
        .sda_pullup_en = GPIO_PULLUP_DISABLE,
        .scl_pullup_en = GPIO_PULLUP_DISABLE,
        .master.clk_speed = 400000,
    };
    i2c_driver_install(i2c_master_port, conf.mode, I2C_MASTER_RX_BUF_DISABLE, I2C_MASTER_TX_BUF_DISABLE, 0);
    i2c_param_config(i2c_master_port, &conf);
    uint8_t address;
    printf("     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f\r\n");
    for (int i = 0; i < 128; i += 16) {
        printf("%02x: ", i);
        for (int j = 0; j < 16; j++) {
            fflush(stdout);
            address = i + j;
            i2c_cmd_handle_t cmd = i2c_cmd_link_create();
            i2c_master_start(cmd);
            i2c_master_write_byte(cmd, (address << 1) | WRITE_BIT, ACK_CHECK_EN);
            i2c_master_stop(cmd);
            esp_err_t ret = i2c_master_cmd_begin(i2c_port, cmd, 50 / portTICK_PERIOD_MS);
            i2c_cmd_link_delete(cmd);
            if (ret == ESP_OK) {
                printf("%02x ", address);
            } else if (ret == ESP_ERR_TIMEOUT) {
                printf("UU ");
            } else {
                printf("-- ");
            }
        }
        printf("\r\n");
    }

    i2c_driver_delete(i2c_port);
    return 0;
}
```

I perform a full I2C bus scan using the above code.  There is only a single I2C device on the bus (@ 0x20).

**When performed in a build using IDF v5.0.2 (or .v5.0.1), the scan of the next empty address (0x21) after the last populated address (0x20) will NOT have its STOP applied by the ESP32-C3; this leaves the SCL line in a LOW state until the next transaction with corrupts it.**  

- This is NOT due to the device at 0x20 performing a clock stretch:  I can lift the CLK line connection to device 0x20 and the line STILL remains LOW, driven by the ESP32-C3:

[image]

**When performed in a build using IDF v4.4.3, the transaction is as expected:**

[image]

**So far, the only way I have found around this in IDF v.5.0.x is to check if the SCL line is LOW after EVERY I2C transaction, and, if so, perform a single STOP operation.  This works, but it's incredibly hacky.  According to the Technical manual, the ESP32-C3 should automatically perform a STOP when encountering a NAK on a write and the ACK_CHECK_EN flag is set.  Help!**

mythbuster5, 2023-05-31:
Could you please try applying this commit and try again? https://github.com/espressif/esp-idf/commit/a0f8434f9382016db6f3b102051569b5cbaea2c0

AxelLin, 2023-05-31:
> Could you please try applying this commit and try again? [a0f8434](https://github.com/espressif/esp-idf/commit/a0f8434f9382016db6f3b102051569b5cbaea2c0)

That does not explain why IDF v4.4.3 works (which does not include above fix).

mythbuster5, 2023-05-31:
> > Could you please try applying this commit and try again? [a0f8434](https://github.com/espressif/esp-idf/commit/a0f8434f9382016db6f3b102051569b5cbaea2c0)
> 
> That does not explain why IDF v4.4.3 works (which does not include above fix).

Because 5.0 contains other new changes... 5.0 improved the SCL clock frequency and adjust sampling.

abellofiore, 2023-05-31:
> Could you please try applying this commit and try again? [a0f8434](https://github.com/espressif/esp-idf/commit/a0f8434f9382016db6f3b102051569b5cbaea2c0)

Thank you.  Yes, the one-line timing change fixes below fixes the issue I've been seeing:

[image]

Do you have any timeframe for a 5.0.x release including it?
I'm not sure how to incorporate this change nicely as I'm trying NOT to keep the v5.0.1 IDF under my source control; it seems I'd have to pull a lot of the IDF files into my project's source control to implement this.  Do you have any suggestions?

AxelLin, 2023-06-01:
> Do you have any timeframe for a 5.0.x release including it?

ed70f1c66b31429b230ed194f1cd7f693c16332e is already in release/v5.0 branch.

abellofiore, 2023-06-01:
> 

Forgive me, but I'm using the Windows IDF installer and the latest version of v5 branch is v5.0.2.  Am I missing something?

[image]

DanielKorz, 2023-06-01:
I can't really believe that I have come to see this issue on my system just today. And it gets mentioned.... yesterday. I would have been looking for solutions for hours.

Thanks!

mythbuster5, 2023-06-02:
> > 
> 
> Forgive me, but I'm using the Windows IDF installer and the latest available version of v5 branch is v5.0.2 which does NOT have this change. Are the Windows installer releases not up tp date or am I missing something?
> 
> Once v5.0 was released, why are changes still being made to it? Shouldn't this have been in a bug fix release v5.0.3?
> 
> [image]

a lazy way is to copy-paste some codes into it. Could you please have a try?

mythbuster5, 2023-06-02:
BTW, I saw this issue is very similar to what you have fixed @o-marshmallow , and I'm going to be on leave for few weeks, could you please also take a look? Thank you! in [a0f8434](https://github.com/espressif/esp-idf/commit/a0f8434f9382016db6f3b102051569b5cbaea2c0)

mythbuster5, 2023-06-02:
> > Do you have any timeframe for a 5.0.x release including it?
> 
> [ed70f1c](https://github.com/espressif/esp-idf/commit/ed70f1c66b31429b230ed194f1cd7f693c16332e) is already in release/v5.0 branch.

on 5.0 branch not equal to v5.0.1 and v5.0.2 has same commit. I checked this commit is not there.

AxelLin, 2023-11-18:
@abellofiore Could you check if this is fixed in v5.0.4?

mythbuster5, 2025-01-13:
The issue is solved, I will close this issue.

closed 2025-01-13 by mythbuster5