espressif/esp-idf #11549: IEEE 802.15.4 doesn't receive packets after calling esp_ieee802154_receive()

spark404, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-823-g903af13e84

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

I expected the call to esp_ieee802154_receive() to enable the driver to receive all packets on the network on a particular channel. This should trigger the corresponding hooks like esp_ieee802154_receive_done and esp_ieee802154_receive_failed but neither are called.

### What is the actual behavior?

The expected behaviour is a continuous stream of packets passed to the esp_ieee802154_receive_done handler.

It's important to note here that this does work if the esp_ieee802154_transmit(..) is called before calling esp_ieee802154_receive().

### Steps to reproduce.

Use the following main.c 

```
#include <string.h>
#include <nvs.h>
#include <nvs_flash.h>
#include <esp_ieee802154.h>
#include <esp_log.h>
#include <esp_phy_init.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"

#define TAG "main"
#define RADIO_TAG "ieee802154"

#define PANID 0x4242
#define CHANNEL 15

#define SHORT_NOT_CONFIGURED 0xFFFE

void esp_ieee802154_receive_done(uint8_t* frame, esp_ieee802154_frame_info_t* frame_info) {
    ESP_EARLY_LOGI(RADIO_TAG, "rx OK, received %d bytes", frame[0]);
}

void esp_ieee802154_receive_failed(uint16_t error) {
    ESP_EARLY_LOGI(RADIO_TAG, "rx failed, error %d", error);
}

void esp_ieee802154_receive_sfd_done(void) {
    ESP_EARLY_LOGI(RADIO_TAG, "rx sfd done, Radio state: %d", esp_ieee802154_get_state());
}

void app_main() {
    ESP_LOGI(TAG, "Initializing NVS from flash...");
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);

    ESP_ERROR_CHECK(esp_ieee802154_enable());
    ESP_ERROR_CHECK(esp_ieee802154_set_promiscuous(true));
    ESP_ERROR_CHECK(esp_ieee802154_set_rx_when_idle(true));

    ESP_ERROR_CHECK(esp_ieee802154_set_panid(PANID));
    ESP_ERROR_CHECK(esp_ieee802154_set_coordinator(false));

    ESP_ERROR_CHECK(esp_ieee802154_set_channel(CHANNEL));

    esp_phy_calibration_data_t cal_data;
    ESP_ERROR_CHECK(esp_phy_load_cal_data_from_nvs(&cal_data));

    // Set long address to the mac address (with 0xff padding at the end)
    // Set short address to unconfigured
    uint8_t long_address[8];
    memcpy(&long_address, cal_data.mac, 6);
    long_address[6] = 0xff;
    long_address[7] = 0xfe;
    esp_ieee802154_set_extended_address(long_address);
    esp_ieee802154_set_short_address(SHORT_NOT_CONFIGURED);

//#define MAKE_IT_WORK
#ifdef MAKE_IT_WORK
    // basically bogus data
    long_address[0] = 8;
    esp_ieee802154_transmit(long_address, false);
#endif

    uint8_t radio_long_address[8];
    esp_ieee802154_get_extended_address(radio_long_address);
    ESP_LOGI(TAG, "Receiver ready, panId=0x%04x, channel=%d, long=%02x:%02x:%02x:%02x:%02x:%02x:%02x:%02x, short=%04x",
             esp_ieee802154_get_panid(), esp_ieee802154_get_channel(),
             radio_long_address[0], radio_long_address[1], radio_long_address[2], radio_long_address[3],
             radio_long_address[4], radio_long_address[5], radio_long_address[6], radio_long_address[7],
             esp_ieee802154_get_short_address());

    ESP_ERROR_CHECK(esp_ieee802154_receive());

    // All done, the rest is up to handlers
    while (true) {
        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
}
```

On an active Zigbee network the output with the MAKE_IT_WORK define enabled is:

```
I (293) main_task: Calling app_main()
I (293) main: Initializing NVS from flash...
I (303) phy_init: phy_version 202,b4b3263,May 17 2023,20:14:14
I (353) main: Receiver ready, panId=0x4242, channel=15, long=40:4c:ca:40:28:dc:ff:fe, short=fffe
I (1143) ieee802154: rx sfd done, Radio state: 2
I (1143) ieee802154: rx OK, received 50 bytes
I (1203) ieee802154: rx sfd done, Radio state: 2
I (1203) ieee802154: rx OK, received 53 bytes
I (1203) ieee802154: rx sfd done, Radio state: 2
I (1233) ieee802154: rx sfd done, Radio state: 2
I (1233) ieee802154: rx OK, received 50 bytes
I (1243) ieee802154: rx sfd done, Radio state: 2
I (1243) ieee802154: rx OK, received 53 bytes
```

On the same Zigbee network without  the MAKE_IT_WORK define

```
I (293) main_task: Calling app_main()
I (293) main: Initializing NVS from flash...
I (303) phy_init: phy_version 202,b4b3263,May 17 2023,20:14:14
I (353) main: Receiver ready, panId=0x4242, channel=15, long=40:4c:ca:40:28:dc:ff:fe, short=fffe
I (9943) ieee802154: rx sfd done, Radio state: 2
```

zwx1995esp, 2023-05-31:
Hi @spark404 I have already tried your main.c code, but I did not reproduce this issue.

I use your main.c, without macro `MAKE_IT_WORK` defined, as a receiver. And using another board to transmit some packets, the receiver can receive all of the packet.

So, could you please try using another board to send some packets to see if the receiver can receive those packets?(using another board to send packets, using `esp_ieee802154_transmit` function to send some specific packets).

spark404, 2023-05-31:
Hi @zwx1995esp, thank you for assisting with this issue. I followed your recommendation, however I still encounter the issue I mentioned. For reference I've uploaded the two project that I use for verification:

https://github.com/spark404/ieee802154-receiver
https://github.com/spark404/ieee802154-sender

The output from the sender is as follows:
```
--- WARNING: Serial ports accessed as /dev/tty.* will hang gdb if launched.
--- Using /dev/cu.usbmodem11201 instead...
--- idf_monitor on /dev/cu.usbmodem11201 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
cpu_start: Compile time:     May 31 2023 19:41:26ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400294a6
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd44
load:0x4086e610,len:0x2db8
load:0x40875720,len:0x181c
entry 0x4086c41a
I (23) boot: ESP-IDF v5.2-dev-823-g903af13e84 2nd stage bootloader
I (23) boot: compile time May 31 2023 19:41:31
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 40MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00096000
I (80) boot:  3 zb_storage       Unknown data     01 81 000a6000 00004000
I (88) boot:  4 zb_fct           Unknown data     01 81 000aa000 00000400
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=42020020 size=09590h ( 38288) map
I (115) esp_image: segment 1: paddr=000195b8 vaddr=40800000 size=06a60h ( 27232) load
I (122) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=191a8h (102824) map
I (145) esp_image: segment 3: paddr=000391d0 vaddr=40806a60 size=06984h ( 27012) load
I (155) boot: Loaded app from partition at offset 0x10000
I (155) boot: Disabling RNG early entropy source...
I (166) cpu_start: Unicore app
I (167) cpu_start: Pro cpu up.
W (175) clk: esp_perip_clk_init() has not been implemented yet
I (182) cpu_start: Pro cpu start user code
I (182) cpu_start: cpu freq: 160000000 Hz
I (182) cpu_start: Application information:
I (185) cpu_start: Project name:     ieee802154-sender
I (191) cpu_start: App version:      0.1.0
I (196) cpu_start: Compile time:     May 31 2023 19:41:26
I (202) cpu_start: ELF file SHA256:  c89fda4a0...
I (207) cpu_start: ESP-IDF:          v5.2-dev-823-g903af13e84
I (213) cpu_start: Min chip rev:     v0.0
I (218) cpu_start: Max chip rev:     v0.99
I (223) cpu_start: Chip rev:         v0.0
I (228) heap_init: Initializing. RAM available for dynamic allocation:
I (235) heap_init: At 4080EFD0 len 0006D640 (437 KiB): D/IRAM
I (241) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (248) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (255) spi_flash: detected chip: generic
I (259) spi_flash: flash io: dio
I (263) sleep: Configure to isolate all GPIO pins in sleep state
I (270) sleep: Enable automatic switching of GPIO sleep configuration
I (277) coexist: coex firmware version: ebddf30
I (282) coexist: coexist rom version 5b8dcfa
I (287) app_start: Starting scheduler on CPU0
I (292) main_task: Started on CPU0
I (292) main_task: Calling app_main()
I (292) main: Initializing NVS from flash...
[... 122 lines cut ...]
I (278) coexist: coex firmware version: ebddf30
I (283) coexist: coexist rom version 5b8dcfa
I (288) app_start: Starting scheduler on CPU0
I (293) main_task: Started on CPU0
I (293) main_task: Calling app_main()
I (293) main: Initializing NVS from flash...
I (313) phy_init: phy_version 202,b4b3263,May 17 2023,20:14:14
I (343) main: Receiver ready, panId=0x4242, channel=11, long=40:4c:ca:40:2a:24:ff:fe, short=fffe
```

with the define enabled I see the following
```
I (313) phy_init: phy_version 202,b4b3263,May 17 2023,20:14:14
I (353) main: Receiver ready, panId=0x4242, channel=11, long=40:4c:ca:40:2a:24:ff:fe, short=fffe
I (633) ieee802154: rx sfd done, Radio state: 2
I (633) ieee802154: rx OK, received 103 bytes
I (1633) ieee802154: rx sfd done, Radio state: 2
I (1633) ieee802154: rx OK, received 103 bytes
I (2633) ieee802154: rx sfd done, Radio state: 2
I (2633) ieee802154: rx OK, received 103 bytes
I (3633) ieee802154: rx sfd done, Radio state: 2
I (3633) ieee802154: rx OK, received 103 bytes
```

I tried switching the roles from between the two boards, but that didn't help. The board I'm using is the ESP32-C6-DevKitC-1. Unfortunately I have no other boards to test this with.

zwx1995esp, 2023-06-01:
Hi I add some debug info function for getting more infomations for the reason why your board can not receive any messages.
https://github.com/zwx1995esp/ieee802154-sender/tree/add_debug_info

Could you please try the code with these two steps and send the two logs to me?

(Tips, the log may be too long, so you may need to choose a clear channel, which may not have any IEEE802154 packets(or just very little packets) transmitting on.)

Step 1, enable `PHY_DEBUG` and `MAKE_IT_WORK`, build and flash the firmware, waiting for the whole log printed done, save the whole logs, name it with `normal`.

Step 2, only enable `PHY_DEBUG`, disable `MAKE_IT_WORK`, also build, flash, save logs, name it with `abnormal`.

spark404, 2023-06-02:
Thanks for putting in this effort! I've attached the output files.
[esp32c6-phydebug-abnormal.txt](https://github.com/espressif/esp-idf/files/11634933/esp32c6-phydebug-abnormal.txt)
[esp32c6-phydebug-normal.txt](https://github.com/espressif/esp-idf/files/11634934/esp32c6-phydebug-normal.txt)

zwx1995esp, 2023-06-02:
Ok, I will ask for the PHY group for some help on your logs, maybe need some times.

zwx1995esp, 2023-06-05:
Hi, I have already synced with PHY group, and they said, IEEE802154 has already entered rx mode after the first log print,  so could you please confirm that if there are some codes modified in your `esp-idf`?

 you can also add some logs for this API [`ieee802154_receive`](https://github.com/espressif/esp-idf/blob/master/components/ieee802154/driver/esp_ieee802154_dev.c#L694) to see if this function is called twice in your code.

spark404, 2023-06-07:
Hello @zwx1995esp thank you for keeping track of this. I've ran a test with a log statement on the receive function. Attached are the logs. 

[output_not_working_with_receivecheck.txt](https://github.com/espressif/esp-idf/files/11679690/output_not_working_with_receivecheck.txt)

I noticed something that might be related. On the receiver application I'm setting the address with:
    uint8_t long_address[8];
    // memcpy(&long_address, cal_data.mac, 6);
    esp_read_mac(long_address, ESP_MAC_IEEE802154);
    long_address[6] = 0xff;
    long_address[7] = 0xff;
    esp_ieee802154_set_extended_address(long_address);
    esp_ieee802154_set_short_address(0x1111);

When I send a message from a sender application to that particular combination of pan and long address it isn't received unless I activate promiscuous mode with esp_ieee802154_set_promiscuous(true). I've tried with regular device connected to the same pan and there the packet is received.

zwx1995esp, 2023-06-08:
But, I see your code, `ESP_ERROR_CHECK(esp_ieee802154_set_promiscuous(true));` you have already set the promiscuous mode `true`, so why you need to set it again?

zwx1995esp, 2023-06-08:
And could you please push your idf branch to your remote repository and give me a link? I want to try to reproduce with your code base.

spark404, 2023-06-09:
@zwx1995esp I've updated the receiver and sender code I used to demonstrate the case more clearly I hope this helps. This a small summary of where I am with the debugging. I'll update make the esp-idf branch and update the ticket.

The sender app sends a series of packets for testing
https://github.com/spark404/ieee802154-sender

The receiver app prints debug output for every packet it receives
https://github.com/spark404/ieee802154-receiver

The log output and packet capture for the sequence is attached.
[logs_and_capture.zip](https://github.com/espressif/esp-idf/files/11705411/logs_and_capture.zip)

What seems to be the current case:
If the MAKE_IT_WORK define is not enabled the receiver receives nothing.
If MAKE_IT_WORK is defined the receiver received the broadcast message, the message sent to the short address. It doesn't receive the message sent to the long address.

spark404, 2023-06-09:
I created the debug repository here: https://github.com/spark404/esp-idf/tree/ieee802154-debug

Attached the output from the receiver app with the debug statements (MAKE_IT_WORK enabled):
[receiver_with_debug.txt](https://github.com/espressif/esp-idf/files/11705584/receiver_with_debug.txt)

spark404, 2023-06-09:
I think I solved at least part of my problem. The call to set the extended address needs to be in reversed order: 

```
    uint8_t eui64[8] = {0};
    esp_read_mac(eui64, ESP_MAC_IEEE802154);
    uint8_t eui64_rev[8] = {0};
    for (int i=0; i<8; i++) {
        eui64_rev[7-i] = eui64[i];
    }
    esp_ieee802154_set_extended_address(eui64_rev);
```

If I do that the directed frames arrive as expected.

zwx1995esp, 2023-06-14:
Sorry for the delay, due to some urgent breaks me.

Your issue seems due to the bit order, the extended address is not set rightly as expected. So some frames may be filtered by the hardware.
 
But you enable the promiscuous mode.... If you enable the promiscuous mode, I don't think some frame will be filtered by our hardware.

Our in your apps, maybe some other logic before the examples you shared to me....
It doesn't make sense if you enable the promiscuous mode but some frames are filtered....

nopnop2002, 2023-12-16:
@spark404 

Your code was able to communicate with the Digi Xbee S2C 802.15.4 firmware.

spark404, 2024-01-06:
I've tried the demo projects I've created with the latest version of esp-idf and everything seems to be working. I verified by removing the workarounds in some other code an no issues anymore.

```
I (192) cpu_start: Application information:
I (195) cpu_start: Project name:     ieee802154-receiver
I (201) cpu_start: App version:      0.1.0
I (205) cpu_start: Compile time:     Jan  6 2024 21:38:50
I (211) cpu_start: ELF file SHA256:  0ab02c395...
I (217) cpu_start: ESP-IDF:          v5.3-dev-1288-g5524b692ee
I (223) cpu_start: Min chip rev:     v0.0
I (228) cpu_start: Max chip rev:     v0.99
I (233) cpu_start: Chip rev:         v0.0
```

Many thanks for all the support on this!

closed 2024-01-06 by spark404