espressif/esp-idf #11550: esp32-c6 WiFi-6 drops connection in light sleep

roblatour, 2023-05-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-beta1-378-gea5e0ff298

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-c6 devkitC-1 v1.2 

### Power Supply used.

USB

### What is the expected behavior?

ESP should retain Wi-Fi 6 connect through light sleep.

### What is the actual behavior?

ESP is losing Wi-Fi 6 connect through light sleep ( in general when light sleep > 3 minutes).

### Steps to reproduce.

Code is here:
https://github.com/roblatour/WeatherStation

Run with the following options in general_user_settings.h 
WORKAROUND 0
GENERAL_USER_SETTINGS_REPORTING_FREQUENCY_IN_MINUTES 15

Here is an example of the log when light sleep is set to 3 minutes:

Note the second and third cycle times are much less than the first cycle thanks to Wi-Fi 6 and the retained session.

[image]

In contrast, here is an example of the (partial) log when light sleep is set to 15 minutes - the full log's text is shown in the next section below:

Note the second cycle time is much greater here than in the first example due to the need to reconnect.

[image]

### Debug Logs.

```plain
log showing activity associated with at 15 minute light sleep

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xa98
[... 176 lines cut ...]
I (6469) Weather Station: MQTT_EVENT_CONNECTED
I (6474) Weather Station: MQTT connected (attempt 2 of 3)
I (6474) Weather Station: publish: WeatherStation/temperature 26.04
I (6478) Weather Station: publish: WeatherStation/humidity 32.522
I (6484) Weather Station: publish: WeatherStation/pressure 1007.6
I (6503) Weather Station: MQTT publishing complete
I (6559) Weather Station: publish via PWSWeather is switched off
W (6559) Weather Station: cycle 2 processing time: 2.967039 seconds
I (6561) Weather Station: begin light sleep for 900 seconds
```

### More Information.

I have tested with each of:
    esp_wifi_set_ps(WIFI_PS_MAX_MODEM); 
    esp_wifi_set_ps(WIFI_PS_MIN_MODEM);
    esp_wifi_set_ps(WIFI_PS_NONE);
    and with not setting the ps option at all
same results in all cases.

Wi-Fi connection is established using my country code (CA for Canada)

The problem appears to happen less often with shorter sleep periods (for example once every 3 minutes or less), but definitely happen with longer sleep periods (for example 15 minutes).

The problem does **not** occur if instead of a 15 minute light sleep, the code uses a 15 minute VTaskDelay.

igrr, 2023-05-30:
Regarding https://github.com/roblatour/WeatherStation/blob/cdf04e0e0bd931095254a0b11b61f24d7053ff7d/main/main.c#LL845C37-L845C37,
please do note that we don't support entering light sleep manually when communication peripherals (such as Wi-Fi) are working.

Quoting the [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/sleep_modes.html#wi-fi-bluetooth-and-sleep-modes),

> Before entering Deep-sleep or Light-sleep modes, the application must disable Wi-Fi and Bluetooth using the appropriate calls

Basically, communication drivers make no attempt to preserve the connection if the drivers weren't stopped (e.g. `esp_wifi_stop`) before entering light sleep.

Note that there is another mechanism, [_automatic_ light sleep](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/power_management.html), which allows the system to enter light sleep whenever all tasks are idle. In automatic light sleep more communication drivers _can_ preserve the connection, and will automatically wake up the chip when appropriate, for example to receive a beacon frame. Note that this mechanism is already enabled by default [in the itwt example](https://github.com/espressif/esp-idf/blob/903af13e847cd301e476d8b16b4ee1c21b30b5c6/examples/wifi/itwt/sdkconfig.defaults#L3-L6).

To conclude, if you need to keep Wi-Fi connection (especially with Wi-Fi 6 and ITWT), do not use `esp_light_sleep_start`) and let the system use the automatic light sleep mode, instead. In this case association with the AP will be maintained.

roblatour, 2023-05-31:
Thank you, I had missed that.

I am currently in the process of updating the program to use automatic light sleep. 

When I have it updated, I will run some power comparisons for using automatic light sleep to keep the WiFi 6 connection alive, vs using deep sleep and having the program need to reconnect and post them on my github page referenced above - this to help anyone who stumbles across this post in the future.

As an aside, in working my way thru the itwt example (https://github.com/espressif/esp-idf/blob/release/v5.1/examples/wifi/itwt/main/itwt.c ) I notice the following code which I though a little odd:

`static void connect_handler(void *arg, esp_event_base_t event_base,
                            int32_t event_id, void *event_data)
{
    ESP_LOGI(TAG, "sta connect to %s", DEFAULT_SSID);
    esp_wifi_connect();
}
`
More specifically, the that connect handler calls esp_wifi_connect().  In my code I have taken this line out and it works fine.  Is there a reason that it is there?

esp-lis, 2023-05-31:
@roblatour `esp_wifi_connect` may be called elsewhere in your code, need to check it.

dgduncan, 2023-05-31:
@roblatour More often then not, `esp_wifi_connect` is often used in the handlers to deal with reconnections; however, in this case I believe the name `connect_handler` is a bit of a misnomer as it has nothing to do with "connecting" but instead catching the event when the WIFI_STATION actually starts and is ready to connect to the AP.

closed 2023-05-31 by roblatour