espressif/esp-idf #11551: Is there a tutorial on how to match NVS multi partition and NVS encrypted partition

WangShuoran, 2023-05-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is there a tutorial on how to match NVS multi partition and NVS encrypted partition?

My requirement is that multiple partitions of NVS can be encrypted (even if they all use the same key).

By the way, in the NVS document, is it recommended that we not store data to the default NVS partition, but use a new NVS to store persistence data.

mahavirj, 2023-06-07:
@WangShuoran 

Following example shows initialization of default as well as custom NVS partition with encryption enabled case:

https://github.com/espressif/esp-idf/blob/dc016f59877d13e6e7d4fc193aa5aa764547f16d/examples/security/flash_encryption/main/flash_encrypt_main.c#L74-L88

WangShuoran, 2023-06-14:
@mahavirj 
Thank you for your quickly reply.
If I want to use pre-generated key partition and Flash encrypted, How to use pre-generated key partition tools and Flash encrypted?

In the detail, I have the following questions:
1. Do I use the above function steps to first burn the NVS key partition and the NVS data partition that has been encrypted by the NVS key into the system, then burn the BIN file that has been configured for Flash encryption and NVS encryption through menuconfig into the system, and then restart the ESP32 chip?
2. If there is unencrypted data in the NVS data partition, if we finish the above steps, ESP32 reboot to system run to nvs_flash_secure_init_partition() or nvs_flash_secure_init() and nvs_flash_init() function will report an error? Or will it automatically encrypt the original NVS partition, or delete the original data?

以下是中文，中文是我的母语：
1. 我使用上面功能的步骤是不是1先把NVS密钥分区和已经被NVS密钥加密的NVS数据分区先烧录到系统中，然后再把已经通过menuconfig配置Flash加密和NVS加密的BIN文件烧录到系统中，再重启ESP32芯片？
2. 如果NVS数据分区存在未加密数据时，如果我们完成上面的步骤后，ESP32重启到系统运行到nvs_flash_secure_init_partition()或nvs_flash_secure_init()和nvs_flash_init()函数中会报错吗？还是会自动加密原来的NVS分区，或者删除原有数据呢？

laukik-hase, 2023-06-19:
Hello, @WangShuoran!

1. For generating the NVS keys partition and the encrypted NVS partition, you can use the NVS Partition Generator Utility. Refer to the NVS Partition Generator Utility - [Generate Encryption Keys Partition](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_gen.html#generate-encryption-keys-partition) and [Generate Encrypted NVS Partition](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_gen.html#generate-encrypted-nvs-partition) docs for more information. You will then need to write these files to the device flash according to your partition table.
2. When NVS Encryption is turned on, any unencrypted data present in the NVS partition will not be encrypted automatically - this data will be treated as garbage (and an error will be reported) when an NVS read operation is attempted. It expects encrypted data to be present in the partition.

CC: @mahavirj

WangShuoran, 2023-06-20:
Hello, @laukik-hase !
Thank you for your quickly reply.
So, if I want to burn the encrypted data to the ESP32, I need to burning the key of NVS and encrypted data to the patition. Is this right?

And I am what to use Flash encryption function in ESP32, If I what to use spiffs and before Flash encryption burning, does the spiffs partition can auto encryption?

laukik-hase, 2023-06-20:
Hello, @WangShuoran!

1. Yes, you need to flash the NVS keys partition as well as the encrypted NVS partition to the device.
2. SPIFFS does not support encrypted flash; you may take a look at [`littleFS`](https://github.com/joltwallet/esp_littlefs) or NVS as an alternative.

WangShuoran, 2023-06-20:
Hello, @laukik-hase !
Thank you for your quickly reply.
Now, I find the CONFIG_SECURE_UART_ROM_DL_MODE menuconfig to setting UART Secure Download mode(SECURE_ENABLE_SECURE_ROM_DL_MODE).
https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/api-reference/kconfig.html#config-secure-uart-rom-dl-mode

How do I Flash APP BIN file with this config. Does I need to use Pre burn key to encrypted the BIN file? Or only burn the unencrypted BIN file?
But the plaintext BIN file is prohibit in DIS_DOWNLOAD_MANUAL_ENCRYPT. I do not know how to burn BIN file.

WangShuoran, 2023-06-22:
Hello, @laukik-hase !
I find the ESP-IOT-SOLUTION VFS page, it tell me the FAT can use encryption, how to use it? I can not find the doc the achieve it.
[image]

laukik-hase, 2023-07-05:
Hello, @WangShuoran!

Sorry for the late response. FatFs does support encryption - the image will be encrypted by the bootloader during the first boot when the flash encryption process starts. To enable the same, you must add the `encrypted` flag in your example partition table for the FatFS partition.

mahavirj, 2023-07-17:
We will soon be adding an example for the storage with flash encryption use-case. I am closing this ticket as NVS related questions have been answered here. Thanks.

closed 2023-07-17 by mahavirj

WangShuoran, 2023-11-06:
@mahavirj Where is the example?