espressif/esp-idf #11552: OTA HTTP API in URL redirection set a lot cert_pem to http_config

WangShuoran, 2023-05-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In the OTA HTTP Doc, at esp_https_ota_begin and esp_https_ota function doc show:

This API supports URL redirection, but if CA cert of URLs differ then it should be appended to cert_pem member of config.

I do not know how to set 2 or more cert_pem to the http_config struct.

I don't think the http_config can set 2 or more cert_pem. And if my URLs in redirect have many root pem. How can I do?

ESP-YJM, 2023-06-01:
You can refer to this link https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-reference/protocols/esp_crt_bundle.html#esp-x509-certificate-bundle for introduction.

mahavirj, 2023-06-07:
@WangShuoran 

As @ESP-YJM mentioned above, simplest option is to use ESP certificate bundle feature that shall take care of including in entire Root certificate bundle in the firmware and hence application need not worry about the individual certs for each endpoint.

Alternatively, you may also use custom bundle feature having specific certificate only, to save on the flash footprint, please have a look at the example `examples/protocols/https_x509_bundle` for this.

For some reason if you do not plan to use the certificate bundle feature then you can just concatenate multiple certs to single file and then embed the file in the application, use it through the `cert_pem` member.

Hope this helps!

closed 2023-06-07 by mahavirj

WangShuoran, 2023-06-08:
If I want to use global CA store to add all of I need CA certificate, when URL redirection Is it feasible?