espressif/esp-idf #11554: Add option to put LEDC into IRAM

klew, 2023-05-31:
### Is your feature request related to a problem?

I'm working on LED matrix display in which I can only select LEDs from single row at a time, so I have to switch display between rows frequently - I use it GPTimer in ISR context.
On the other hand, display brightness is controlled by PWM conntected to one GPIO. I use LEDC driver to control it.

I had to put function that display a row at LED matrix in ISR context, so it won't glitch i.e. when I perform flash write operations.

However I had to disable PWM when I configure display for another row. Otherwise, I see dimmed LEDs in one row below current new line. This is related to HW design of this LED matrix, which updates column index of enabled LED and in last operation, it switches selected row.

Working solution for my problem is to put LEDC functions (at least ledc_stop and ledc_update_duty) in IRAM (otherwise it can crash, when cache is disabled i.e. during flash write).
I did it by manually editing linker.lf and by adding there "ledc (noflash)".

Currently there is no option to put LEDC into IRAM via configuration. I also didn't find any way how to force linker to do it without modification of esp-idf itself.

### Describe the solution you'd like.

Add option to put LEDC into IRAM (at least ledc_stop and ledc_update_duty functions).

igrr, 2023-05-31:
> I did it by manually editing linker.lf and by adding there "ledc (noflash)".
> I also didn't find any way how to force linker to do it without modification of esp-idf itself.

You can add a linker script rule not only in the component where the object file is located, but also from a component in your application. See, for example, how the GPIO driver is moved into IRAM in this example's `main` component:

https://github.com/espressif/esp-idf/blob/903af13e847cd301e476d8b16b4ee1c21b30b5c6/examples/peripherals/spi_master/hd_eeprom/components/eeprom/linker.lf#L10-L19

klew, 2023-05-31:
> You can add a linker script rule not only in the component where the object file is located, but also from a component in your application. See, for example, how the GPIO driver is moved into IRAM in this example's `main` component:

Thanks for hint, it worked!
I was thinking about it earlier, but I found some forum post where it was stated that there was some error reported when duplicated archive name is found (it was related to esp-idf v3.x). Then I read https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/linker-script-generation.html which says that "placement is specified at component level" and "In cases where multiple fragments of the same type and name are encountered, an exception is thrown", so I assumed that I can't override default placement in other place then "driver" component itself.

So basically, this solves my case, however I still think that having configuration option to place LEDC (or part of it) in IRAM may be beneficial for other users as well. Feel free to keep this issue open and add such option, or to close it with mentioned above solution.

closed 2023-06-16 by espressif-bot