espressif/esp-idf #11555: Encounter build errors during migration to v5.0

junyoun-kim, 2023-05-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2-227-gde419dbe0b

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build fail.

### What is the actual behavior?

During migration to esp_idf 5.0 release, I encountered two build errors.

Could you check it there is a bug or guide me if I do wrong?

1. At esp_macros.h

```
In file included from /home/junyoun3/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:76,
                 from /home/junyoun3/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from /home/junyoun3/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from /home/junyoun3/esp/esp-idf/components/esp_system/include/esp_task.h:24,
                 from /home/junyoun3/esp/esp-idf/components/lwip/port/esp32/include/lwipopts.h:23,
                 from /home/junyoun3/esp/esp-idf/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from /home/junyoun3/esp/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:42,
                 from /home/junyoun3/esp/esp-idf/components/lwip/include/lwip/sockets.h:8,
                 from /home/junyoun3/esp/esp-idf/components/lwip/port/esp32/include/sys/socket.h:33,
                 from /home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/port/net/mbedtls/iot_net_mbedtls.c:24:
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:51:23: error: expected declaration specifiers or '...' before numeric constant
   51 | #define foo_no_args() 2
      |                       ^
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:45:64: note: in expansion of macro 'foo_no_args'
   45 | #define CHOOSE_MACRO_VA_ARG_INN(one, two, MACRO1, MACRO2, ...) MACRO1
      |                                                                ^~~~~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:46:71: note: in expansion of macro 'CHOOSE_MACRO_VA_ARG_INN'
   46 | #define CHOOSE_MACRO_VA_ARG(MACRO_WITH_ARGS, MACRO_WITH_NO_ARGS, ...) CHOOSE_MACRO_VA_ARG_INN(0, ##__VA_ARGS__, MACRO_WITH_ARGS, MACRO_WITH_NO_ARGS, 0)
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:55:18: note: in expansion of macro 'CHOOSE_MACRO_VA_ARG'
   55 | #define foo(...) CHOOSE_MACRO_VA_ARG(foo_args, foo_no_args, ##__VA_ARGS__)(__VA_ARGS__)
      |                  ^~~~~~~~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:58:19: note: in expansion of macro 'foo'
   58 | ESP_STATIC_ASSERT(foo() == 2, "CHOOSE_MACRO_VA_ARG() result does not match for 0 arguments");
      |                   ^~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:58:31: error: expected declaration specifiers or '...' before string constant
   58 | ESP_STATIC_ASSERT(foo() == 2, "CHOOSE_MACRO_VA_ARG() result does not match for 0 arguments");
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:50:23: error: expected declaration specifiers or '...' before numeric constant
   50 | #define foo_args(...) 1
      |                       ^
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:45:64: note: in expansion of macro 'foo_args'
   45 | #define CHOOSE_MACRO_VA_ARG_INN(one, two, MACRO1, MACRO2, ...) MACRO1
      |                                                                ^~~~~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:46:71: note: in expansion of macro 'CHOOSE_MACRO_VA_ARG_INN'
   46 | #define CHOOSE_MACRO_VA_ARG(MACRO_WITH_ARGS, MACRO_WITH_NO_ARGS, ...) CHOOSE_MACRO_VA_ARG_INN(0, ##__VA_ARGS__, MACRO_WITH_ARGS, MACRO_WITH_NO_ARGS, 0)
      |                                                                       ^~~~~~~~~~~~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:55:18: note: in expansion of macro 'CHOOSE_MACRO_VA_ARG'
   55 | #define foo(...) CHOOSE_MACRO_VA_ARG(foo_args, foo_no_args, ##__VA_ARGS__)(__VA_ARGS__)
      |                  ^~~~~~~~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:59:19: note: in expansion of macro 'foo'
   59 | ESP_STATIC_ASSERT(foo(42) == 1, "CHOOSE_MACRO_VA_ARG() result does not match for 1 argument");
      |                   ^~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_macros.h:59:33: error: expected declaration specifiers or '...' before string constant
   59 | ESP_STATIC_ASSERT(foo(42) == 1, "CHOOSE_MACRO_VA_ARG() result does not match for 1 argument");
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```

2. No static_assert function.

```
In file included from /home/junyoun3/esp/esp-idf/components/soc/esp32/include/soc/soc.h:11,
                 from /home/junyoun3/esp/esp-idf/components/soc/esp32/include/soc/efuse_reg.h:10,
                 from /home/junyoun3/esp/esp-idf/components/soc/include/soc/efuse_periph.h:8,
                 from /home/junyoun3/esp/esp-idf/components/hal/esp32/include/hal/efuse_ll.h:11,
                 from /home/junyoun3/esp/esp-idf/components/bootloader_support/include/esp_flash_encrypt.h:15,
                 from /home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/port/bsp/esp32/iot_bsp_fs_esp32.c:23:
/home/junyoun3/esp/esp-idf/components/hal/esp32/include/hal/efuse_ll.h: In function 'efuse_ll_get_flash_crypt_cnt':
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_assert.h:12:27: error: implicit declaration of function 'static_assert' [-Werror=implicit-function-declaration]
   12 | #define ESP_STATIC_ASSERT static_assert
      |                           ^~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/esp_common/include/esp_assert.h:18:13: note: in expansion of macro 'ESP_STATIC_ASSERT'
   18 |             ESP_STATIC_ASSERT(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
      |             ^~~~~~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/soc/esp32/include/soc/soc.h:34:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
   34 | #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
      |                                      ^~~~~~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/soc/esp32/include/soc/soc.h:77:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
   77 |             ASSERT_IF_DPORT_REG((_r), REG_GET_FIELD);                                                                  \
      |             ^~~~~~~~~~~~~~~~~~~
/home/junyoun3/esp/esp-idf/components/hal/esp32/include/hal/efuse_ll.h:26:12: note: in expansion of macro 'REG_GET_FIELD'
   26 |     return REG_GET_FIELD(EFUSE_BLK0_RDATA0_REG, EFUSE_RD_FLASH_CRYPT_CNT);

```

### Steps to reproduce.

1. Step
2. Step
3. Step
...

igrr, 2023-05-31:
Could you please post the full compiler command line used to compile these two files, to help our analysis? You can find the compiler command line either immediately above the error message, or in build/compile_commands.json file (look for iot_bsp_fs_esp32.c and iot_net_mbedtls.c)

junyoun-kim, 2023-05-31:
Here are what you request.

- command line for iot_net_mbedtls.c

/home/junyoun3/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/example/esp32/build/config -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/include/os -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/include/mqtt -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/include/external -I/home/junyoun3/esp/esp-idf/components/freertos/include/freertos -I/home/junyoun3/esp/esp-idf/components/nvs_flash/include -I/home/junyoun3/esp/esp-idf/components/spi_flash/include -I/home/junyoun3/esp/esp-idf/components/bootloader_support/include -I/home/junyoun3/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/include -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/include/bsp -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/include/bsp/esp32 -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/port/net -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/port/net/mbedtls -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/port/http/socket -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/deps/libsodium/libsodium/src/libsodium/include -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/deps/libsodium/libsodium/src/libsodium/include/sodium -I/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/deps/libsodium/port/include -I/home/junyoun3/esp/esp-idf/components/newlib/platform_include -I/home/junyoun3/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/junyoun3/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/junyoun3/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/junyoun3/esp/esp-idf/components/freertos/esp_additions/include -I/home/junyoun3/esp/esp-idf/components/esp_hw_support/include -I/home/junyoun3/esp/esp-idf/components/esp_hw_support/include/soc -I/home/junyoun3/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/junyoun3/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/junyoun3/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/home/junyoun3/esp/esp-idf/components/heap/include -I/home/junyoun3/esp/esp-idf/components/log/include -I/home/junyoun3/esp/esp-idf/components/soc/include -I/home/junyoun3/esp/esp-idf/components/soc/esp32/. -I/home/junyoun3/esp/esp-idf/components/soc/esp32/include -I/home/junyoun3/esp/esp-idf/components/hal/esp32/include -I/home/junyoun3/esp/esp-idf/components/hal/include -I/home/junyoun3/esp/esp-idf/components/hal/platform_port/include -I/home/junyoun3/esp/esp-idf/components/esp_rom/include -I/home/junyoun3/esp/esp-idf/components/esp_rom/include/esp32 -I/home/junyoun3/esp/esp-idf/components/esp_rom/esp32 -I/home/junyoun3/esp/esp-idf/components/esp_common/include -I/home/junyoun3/esp/esp-idf/components/esp_system/include -I/home/junyoun3/esp/esp-idf/components/esp_system/port/soc -I/home/junyoun3/esp/esp-idf/components/esp_system/port/include/private -I/home/junyoun3/esp/esp-idf/components/xtensa/include -I/home/junyoun3/esp/esp-idf/components/xtensa/esp32/include -I/home/junyoun3/esp/esp-idf/components/lwip/include -I/home/junyoun3/esp/esp-idf/components/lwip/include/apps -I/home/junyoun3/esp/esp-idf/components/lwip/include/apps/sntp -I/home/junyoun3/esp/esp-idf/components/lwip/lwip/src/include -I/home/junyoun3/esp/esp-idf/components/lwip/port/esp32/include -I/home/junyoun3/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/junyoun3/esp/esp-idf/components/esp_partition/include -I/home/junyoun3/esp/esp-idf/components/esp_wifi/include -I/home/junyoun3/esp/esp-idf/components/esp_event/include -I/home/junyoun3/esp/esp-idf/components/esp_phy/include -I/home/junyoun3/esp/esp-idf/components/esp_phy/esp32/include -I/home/junyoun3/esp/esp-idf/components/esp_netif/include -I/home/junyoun3/esp/esp-idf/components/mbedtls/port/include -I/home/junyoun3/esp/esp-idf/components/mbedtls/mbedtls/library -I/home/junyoun3/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/junyoun3/esp/esp-idf/components/json/cJSON -mlongcalls -Wno-frame-address  -DCONFIG_STDK_IOT_CORE -DCONFIG_STDK_IOT_CORE_EASYSETUP_DISCOVERY_SSID -DCONFIG_STDK_IOT_CORE_EASYSETUP_HTTP -DCONFIG_STDK_IOT_CORE_EASYSETUP_HTTP_USE_SOCKET_API -DCONFIG_STDK_IOT_CORE_LOG_LEVEL_ERROR -DCONFIG_STDK_IOT_CORE_LOG_LEVEL_WARN -DCONFIG_STDK_IOT_CORE_LOG_LEVEL_INFO -DCONFIG_STDK_IOT_CORE_LOG_FILE -DCONFIG_STDK_IOT_CORE_LOG_FILE_RAM_ONLY -DCONFIG_STDK_IOT_CORE_LOG_FILE_RAM_BUF_SIZE=8192 -DCONFIG_STDK_IOT_CORE_BSP_SUPPORT_ESP32 -DCONFIG_STDK_IOT_CORE_OS_SUPPORT_FREERTOS -DCONFIG_STDK_IOT_CORE_USE_MBEDTLS -DCONFIG_STDK_IOT_CORE_CRYPTO_SUPPORT_ED25519 -DCONFIG_STDK_IOT_CORE_SECURITY_BACKEND_SOFTWARE -DCONFIG_STDK_IOT_CORE_FS_HW_ENCRYPTION -DCONFIG_STDK_IOT_CORE_NET_MBEDTLS -include /home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/include/iot_common.h -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/example/esp32=. -fmacro-prefix-map=/home/junyo [... cut ...] -old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0.2-227-gde419dbe0b\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -std=gnu99 -MD -MT esp-idf/main/iotcore/src/CMakeFiles/iotcore.dir/port/bsp/esp32/iot_bsp_fs_esp32.c.obj -MF esp-idf/main/iotcore/src/CMakeFiles/iotcore.dir/port/bsp/esp32/iot_bsp_fs_esp32.c.obj.d -o esp-idf/main/iotcore/src/CMakeFiles/iotcore.dir/port/bsp/esp32/iot_bsp_fs_esp32.c.obj -c /home/junyoun3/ProjectRepos/STDK_refs_internal/iot-core/src/port/bsp/esp32/iot_bsp_fs_esp32.c

igrr, 2023-05-31:
> `-std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER="v5.0.2-227-gde419dbe0b" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -std=gnu99`

Looks like there are both `-std=gnu17` and `-std=gnu99` flags passed to the compiler. Could you please check if you are by any chance adding `-std=gnu99` somewhere in your component CMakeLists.txt file?

junyoun-kim, 2023-06-08:
@igrr Thank you very much. Looks like that was the problem. After I delete setting 99 standard option in my cmake, it solved above errors.

closed 2023-06-08 by junyoun-kim