espressif/esp-idf #11556: Add LWIP ALTCP TLS Support

HamzaHajeir, 2023-05-31:
### Is your feature request related to a problem?

I'm maintaining an Arduino-based library called [H4AsyncTCP](https://github.com/HamzaHajeir/H4AsyncTCP) after its author who has passed away a year ago, alongwith his H4 Project.

We've succeeded to reach a stable development of TCP over LwIP Raw APIs over his work.

Going through solid and secured IoT systems, often requires running over TLS layer, which is provided by LwIP through its introduced [Application Layered TCP](https://www.nongnu.org/lwip/2_1_x/group__altcp__api.html) with minimum code changes.

 The default in ESP-IDF doesn't add the related files to the precompiled libraries, which results in 'undefined reference to altcp_xxx' linking errors even with corresponding macros defined at building.

### Describe the solution you'd like.

Add LWIP ALTCP and TLS into support, to what 
- Add `LWIP_ALTCP` and `LWIP_ALTCP_TLS` as configurations (such as `CONFIG_LWIP_XXX`, and to SDKConfig.
- Add Configuration-based addition of sources and include directories in the lwip [CMakeLists file](https://github.com/espressif/esp-idf/blob/master/components/lwip/CMakeLists.txt#L19), here's a "just working" [example](https://github.com/HamzaHajeir/esp32-arduino-lib-builder/commit/552ab457bf34f73a31531fa8b861c4c98422e15a#diff-8d131133b6a4762fe1845209fed75c45602b9fad090a69ac755674f019d98139) you might source from.

### Additional context.

The H4AsyncTCP library is fully Async, which implements the preferred way of programming especially for resource constrained devices, and it has reached a solid point though.

Having it works on Raw LwIP APIs is something valuable, because LwIP is mature and fast, fits into place.
Especially when it offers seamless shift to support TLS, it's the only choice I can see.

---- 

After building the libraries by esp32-arduino-lib-builder, I've successfully shifted to ALTCP and have run TCP with no issues, (yet to come for TLS development).

HamzaHajeir, 2023-06-22:
> @HamzaHajeir <https://github.com/HamzaHajeir> Note, that you can add
> ALTCP to lwip build on the project level makefile (with no changes to IDF)
>
> project(...)
> set(src_dir $ENV{IDF_PATH}/components/lwip/lwip/src/)set(srcs
>         "${src_dir}/core/altcp_alloc.c"
>         "${src_dir}/core/altcp_tcp.c"
>         "${src_dir}/core/altcp.c"
>         "${src_dir}/apps/altcp_tls/altcp_tls_mbedtls_mem.c"
>         "${src_dir}/apps/altcp_tls/altcp_tls_mbedtls.c")
> idf_component_get_property(lwip lwip COMPONENT_LIB)target_sources(${lwip} PRIVATE "${srcs}")target_include_directories(${lwip} PUBLIC "${src_dir}/apps/altcp_tls")target_compile_definitions(${lwip} PRIVATE -DLWIP_ALTCP=1 -DLWIP_ALTCP_TLS=1)

I agree that this is a way to go at this time.

What I'm seeking is making ALTCP_TLS_MBEDTLS available for Arduino folks.

The idf libs are being built by espressif/esp32_arduino_lib_builder tool
for being used by Arduino users.

it's way easier and straightforward for it's maintainer to tweak config
files to support it than overriding lwip component.

That being said, I really can't stand for asking the maintainer to include
the modified lwip as a custom component, rather maintaining the copy.

Best regards,
Hamzah Hajeir

david-cermak, 2023-06-23:
Although adding these few lines to IDF config is a very simple change, I'm a bit reluctant to do so, as it might give a false impression that the `ALTCP` is a supported and well tested feature while it's actually not the case. Quite the opposite, we claim that raw lwip API is available, but not tested and often discourage people from using it. There are two potential issues with this API IMO: 1) it might be used incorrectly wrtt thread safety; 2) It's usually not very clear if we're using plain TCP or TLS communication (i.e. customer might think they're using TLS, but per some mis-configuration ALTCP could be simply replaced with TCP)

Let us keep this issue open to see if there're any other comments, some other use cases or inputs from users.

Anyway, you can use https://github.com/espressif/esp-idf/issues/11556#issuecomment-1600628194 without modifying IDF components, sources or maintaining a separate copy of lwip. You just place this in a project side configuration (in your case Arduino lib builder)

HamzaHajeir, 2023-06-23:
Hi david,
I see your concerns.

The LwIP user would (and should) really know what its restrictions, LwIP has [documented](https://www.nongnu.org/lwip/2_1_x/pitfalls.html) the raw APIs issue in OS Mode.

We've successfully implemented raw APIs by one of the mentioned methods: LwIP core thread locking, which is correctly supported in esp-idf port.

On top of it, We've successfully converted to ALTCP, although it required applying patches/commits that resides in the LwIP master branch, but that's another story I'll point it to esp-lwip maintainers beside few patches I've applied and will submit to LwIP at soonest.

Now we've a working [library](https://github.com/HamzaHajeir/H4AsyncTCP) utilizing ALTCP that can behave as TCP or TLS/TCP dynamically. You may check it out.

Per the concern about ALTCP misconfiguration that drops the security down to TCP. A misconfiguration would result in `NULL` being returned as a Protocol Control Block `pcb`, wherein ALTCP uses allocators of two types: `altcp_tcp_allocator` and `altcp_tls_allocator` being fed to `altcp_new`, nevertheless the explicit APIs `altcp_tls_new` or `altcp_tcp_new`.

david-cermak, 2023-10-10:
Hi @HamzaHajeir 

Thanks for pointing out your solution.It seems though that the integration to IDF still missing few steps and, as mentioned above, I've raised some concerns.
The actual integration should be easy and should happen in the *"user space"* only, without any need of incorporating patches to a released IDF (as noted in https://github.com/espressif/esp-idf/issues/11556#issuecomment-1603744257)

So we will not add ALTCP into lwip's Kconfig for now.

closed 2023-10-10 by david-cermak

HamzaHajeir, 2023-10-10:
Hi @david-cermak 

First things first, I'm already in the preparation process for patching esp-lwip, that syncs with lwip-master, provides altcp_tls_mbedtls fixes, and supports IDF v5 (mbedtls v3 dependency). I'm sending PRs soon.

I've been running ALTCL TLS for +2 months without issues, I can provide a PlatformIO environment to test out TLS webserver and TLS MQTT/HTTP clients with simple preprocessor edits.

Per Kconfig, the configuration most probably will disable ALTCP and the further TLS and configs by default, thus there's no concerns regarding the IDF users.

Finally, as Arduino users will gain great benefits with ALTCP support, I appreciate your reconsideration. I can help with the support.

Best,
Hamza Hajeir