espressif/esp-idf #11558: SPIRAM corruption in light sleep mode

Erlkoenig90, 2023-05-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

both v5.0.2 and master/903af13e847cd301e476d8b16b4ee1c21b30b5c6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC

### Power Supply used.

USB

### What is the expected behavior?

When using sleep mode, SPIRAM contents should stay intact.

### What is the actual behavior?

When the CPU is in light-sleep for > ~10min, individual bits in SPIRAM are flipped. If data structures / pointers are affected, this leads to random crashes/errors.

### Steps to reproduce.

1. Compile the [test case application](https://github.com/Erlkoenig90/esp32s3-spiram-corrupt).
2. Flash on ESP32S3-DevKit-C
3. Run for 10-30min

This error occurs when memory in SPIRAM is allocated via `heap_caps_malloc (bufSize, MALLOC_CAP_SPIRAM)`, some data is written to memory, sleep mode is entered for some time (10s to 10min), and then memory is read again. Some of the read bytes have one bit flipped.

The following config options are set to reproduce this:
`ESPTOOLPY_FLASHSIZE_8MB`
`SPIRAM`
`SPIRAM_MODE_OCT`
`SPIRAM_USE_CAPS_ALLOC`
`PM_ENABLE`
`FREERTOS_USE_TICKLESS_IDLE`
`ESP_DEFAULT_CPU_FREQ_MHZ_240`
`COMPILER_OPTIMIZATION_PERF`

However, when `ESP_SLEEP_PSRAM_LEAKAGE_WORKAROUND` and `ESP_SLEEP_FLASH_LEAKAGE_WORKAROUND` are set, this issue does **not** occur.

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x1718
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc08
load:0x403cc700,len:0x2f08
[... 96 lines cut ...]
E (603680) main: Memory check failed at addr=0x3c030fe8 i=   1768, got=f8, expect=e8, xor=10, mismatches=0, label=loop
E (603690) main: Memory check failed at addr=0x3c03102e i=   1838, got=2c, expect=2e, xor=02, mismatches=0, label=loop
E (603700) main: Memory check failed at addr=0x3c03103d i=   1853, got=35, expect=3d, xor=08, mismatches=0, label=loop
E (603720) main: Memory check failed at addr=0x3c031120 i=   2080, got=21, expect=20, xor=01, mismatches=0, label=loop
E (603730) main: Memory check failed at addr=0x3c03113e i=   2110, got=36, expect=3e, xor=08, mismatches=0, label=loop
E (603740) main: Memory check failed at addr=0x3c031178 i=   2168, got=58, expect=78, xor=20, mismatches=0, label=loop
E (603750) main: Memory check failed at addr=0x3c031194 i=   2196, got=95, expect=94, xor=01, mismatches=0, label=loop
E (603760) main: Memory check failed at addr=0x3c0311aa i=   2218, got=2a, expect=aa, xor=80, mismatches=0, label=loop
E (603770) main: Memory check failed at addr=0x3c0311e2 i=   2274, got=e0, expect=e2, xor=02, mismatches=0, label=loop
E (603780) main: Memory check failed at addr=0x3c031209 i=   2313, got=0d, expect=09, xor=04, mismatches=0, label=loop
E (603800) main: Memory check failed at addr=0x3c03120e i=   2318, got=0f, expect=0e, xor=01, mismatches=0, label=loop
E (603810) main: Memory check failed at addr=0x3c031222 i=   2338, got=02, expect=22, xor=20, mismatches=0, label=loop
E (603820) main: Memory check failed at addr=0x3c03122d i=   2349, got=4d, expect=2d, xor=60, mismatches=0, label=loop
E (603830) main: Memory check failed at addr=0x3c031256 i=   2390, got=57, expect=56, xor=01, mismatches=0, label=loop
```

### More Information.

It took me a long time to narrow this issue down. I experienced random crashes in a complex application after waking up from sleep mode which turned out to be caused by the memory corruption. I accidentally disabled `ESP_SLEEP_PSRAM_LEAKAGE_WORKAROUND` and `ESP_SLEEP_FLASH_LEAKAGE_WORKAROUND`. I didn't realize that those are necessary to use SPIRAM and sleep mode. The documentation for those settings states that they can be used to reduce power consumption in sleep mode, but not that they are needed to keep SPIRAM data valid.

I am not sure whether this is normal/intended behavior. Should SPIRAM work normally even without `ESP_SLEEP_PSRAM_LEAKAGE_WORKAROUND` and `ESP_SLEEP_FLASH_LEAKAGE_WORKAROUND`, just with higher power consumption?

Perhaps this is just a documentation issue? Should the documentation for those config settings state that they are necessary for using sleep mode? Maybe the doc for [SPIRAM](https://docs.espressif.com/projects/esp-idf/en/v5.0.2/esp32s3/api-guides/external-ram.html) and [sleep modes](https://docs.espressif.com/projects/esp-idf/en/v5.0.2/esp32s3/api-reference/system/sleep_modes.html) should have a warning that those settings are needed?

Would it make sense to run a memory check after waking up from sleep mode to verify SPIRAM contents is still valid (keeping a checksum in IRAM)?

huming2207, 2023-06-01:
I think we had a similar issue before, and I think enabling the `ESP_SLEEP_PSRAM_LEAKAGE_WORKAROUND` is a proper fix unless you have an external PSRAM and external pullup. Otherwise, two things will happen:

1. The SPI flash is sharing the same SPI master with the PSRAM and the microcontroller need to talk to it as well. If the CS/CE# pin of PSRAM is floating when the ESP32 is preparing itself to go to sleep or wake up, garbage data may be written into the PSRAM, which causes corruption;
3. In regards to the power consumption, at least for ESP-PSRAM64H and APS12808L-3OBM, both datasheets mentioned: "When CE#=1, the chip is in standby state". Thus if you disable `ESP_SLEEP_PSRAM_LEAKAGE_WORKAROUND` and the CS/CE# pin does not have an external pull-up, it will probably waste more power; while if you enable it, it won't waste power. Indeed pulling up the CE# pin during sleep may also cause a minor leakage, it should be far less than leaving PSRAM awake, probably a few uA vs a few mA.

See: 

- https://www.espressif.com/sites/default/files/documentation/esp-psram64_esp-psram64h_datasheet_en.pdf
- https://au.mouser.com/datasheet/2/1127/APM_PSRAM_OPI_Xccela_APS12808L_3OBMx_v1_1b_PKG-1954845.pdf

Erlkoenig90, 2023-06-01:
Hi, thanks for the information. Then I will keep `ESP_SLEEP_PSRAM_LEAKAGE_WORKAROUND` enabled.

Which PSRAM chip is included in the `ESP32-S3R8` chip, and does it have pull-up resistors? We use the `ESP32-S3-WROOM-1-N16R8` module on a custom PCB, and the module datasheet does not show any pull-up on `SPICS0`/`SPICS1`, but perhaps the `ESP32-S3R8` contains one internally?

closed 2023-06-29 by espressif-bot