espressif/esp-idf #11561: Provide the linux-i686 build for the esp32ulp-elf "Toolchain for ESP32 ULP coprocessor"

usr97629238, 2023-05-31:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

esp32ulp-elf https://github.com/espressif/binutils-gdb and https://github.com/espressif/esp-idf/blob/master/tools/tools.json provide the esp32ulp-elf build for linux-i686

### What is the actual behavior?

ERROR: tool esp32ulp-elf does not have versions compatible with platform linux-i686   

### Steps to reproduce.

1. Install esp-idf on linux-i686
2. Run ./install.sh esp32

### Build or installation Logs.

```plain
Detecting the Python interpreter                                                                                                               Checking "python" ...                                                                                                                          Checking "python3" ...                                                                                                                         Python 3.7.3                                                                                                                                   "python3" has been detected                                                                                                                    Installing ESP-IDF tools                                                                                                                       Current system platform: linux-i686                                                                                                            Selected targets are: esp32, esp32c3                                                                                                           Installing tools: xtensa-esp-elf-gdb, xtensa-esp32-elf, esp32ulp-elf
Skipping xtensa-esp-elf-gdb@11.2_20220823 (already installed)
Skipping xtensa-esp32-elf@esp-2021r2-patch5-8.4.0 (already installed) 
ERROR: tool esp32ulp-elf does not have versions compatible with platform linux-i686
```

### More Information.

Please provide the linux-i686 build for the esp32ulp-elf https://github.com/espressif/binutils-gdb, and update the https://github.com/espressif/esp-idf/blob/master/tools/tools.json accordingly.
Or at least provide the clear config/make configuration to build it.
Strangely, all the tools in the toolkit have their linux-i686 version except for this.

Trying to build it myself, there are numerous issues:
- injecting the link into the `$HOME/esp/esp-idf/tools/tools.json` (would be gone after an update)
- the build configuration isn't provided with the release https://github.com/espressif/binutils-gdb/releases/tag/esp32ulp-elf-v2.35_20220830, there are some hints here https://github.com/espressif/binutils-esp32ulp/issues/1 and there https://github.com/espressif/binutils-esp32ulp/issues/19, but they are incomplete, cause despite building it from the release source https://github.com/espressif/binutils-gdb/releases/tag/esp32ulp-elf-v2.35_20220830 (and not from the head) the resulting files differ from the released by espressif and now I'm stuck at the ` . $HOME/esp/esp-idf/export.sh`:
```
Adding ESP-IDF tools to PATH...
WARNING: tool esp32ulp-elf version 2.35_20220830 is installed, but has reported version Copyright
ERROR: tool esp32ulp-elf has no installed versions. Please run '/home/user/esp/esp-idf/install.sh' to install it. 
```

As the built file's output lacks the version printout:
`./esp32ulp-elf-as --version`
```
GNU assembler (GNU Binutils)
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `esp32ulp-elf'.
```

While the version check in the script $HOME/esp/esp-idf/tools/idf_tools.py awaits something along the lines of:
```
GNU assembler (crosstool-NG esp-2021r2-patch5) 2.35.1.20201223
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `riscv32-esp-elf'.
```

And that requires either hacking the `$HOME/esp/esp-idf/tools/idf_tools.py` to override this check or finding out how to enable the version printout in that "trillion lines of code" binutils' source.

usr97629238, 2023-06-02:
The lack of a linux-i686 build for one tool in the toolkit is a bug, not a feature request.

Lapshin, 2023-06-06:
@usr97629238 , thank you for bug report. This will be done with the next release.

To build the toolchain by yourself you can use the command (replace `gcc` with yours if need):
```
./configure \
  --host=`gcc -dumpmachine`\
  --target=esp32ulp-elf \
  --prefix="${INSTALL_DIR}" \
  --disable-doc --disable-gdb --disable-libdecnumber --disable-readline --disable-sim
make
make install
```

Also, see the build routine here https://github.com/espressif/binutils-gdb/blob/esp32ulp-elf-v2.35_20220830/.gitlab-ci.yml#L131

antmak, 2023-06-14:
Hi @usr97629238. Thank you, fixed. A fix will be in master, v4.4 and other branches soon. 

The i686 distribution has been added to the release https://github.com/espressif/binutils-gdb/releases/tag/esp32ulp-elf-v2.35_20220830  
An IDF fix
```patch
diff --git a/tools/tools.json b/tools/tools.json
index 612b9353d8..43079c09fa 100644
--- a/tools/tools.json
+++ b/tools/tools.json
@@ -542,6 +542,11 @@
             "size": 9591791,
             "url": "https://github.com/espressif/binutils-gdb/releases/download/esp32ulp-elf-v2.35_20220830/esp32ulp-elf-2.35_20220830-linux-armhf.tar.gz"
           },
+          "linux-i686": {
+            "sha256": "beb9b6737c975369b6959007739c88f44eb5afbb220f40737071540b2c1a9064",
+            "size": 10596533,
+            "url": "https://github.com/espressif/binutils-gdb/releases/download/esp32ulp-elf-v2.35_20220830/esp32ulp-elf-2.35_20220830-linux-i686.tar.gz"
+          },
           "macos": {
             "sha256": "5a952087b621ced16af1e375feac1371a61cb51ab7e7b44cbefb5afda2d573de",
             "size": 16028909,

```

closed 2023-06-14 by espressif-bot