espressif/esp-idf #11567: ESP32-S3 Module 256MBit PSRAM Runtime Problem

suniltop, 2023-06-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I have several esp32-s3-wroom-1u module. 
When i try to use psram (in package), software detects psram but it fails to run app code and hangs at that point and then wdt reset the system.

Also, GPIO35,36 and 37 is unconnected

Log

> ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x17d8
load:0x403c9700,len:0xe88
load:0x403cc700,len:0x3000
entry 0x403c9930
I (25) boot: ESP-IDF v5.0 2nd stage bootloader
I (25) boot: compile time 09:42:10
I (25) boot: chip revision: v0.1
I (26) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=089e4h ( 35300) map
I (117) esp_image: segment 1: paddr=00018a0c vaddr=3fc92600 size=03208h ( 12808) load
I (122) esp_image: segment 2: paddr=0001bc1c vaddr=40374000 size=043fch ( 17404) load
I (132) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19808h (104456) map
I (153) esp_image: segment 4: paddr=00039830 vaddr=403783fc size=0a194h ( 41364) load
I (161) esp_image: segment 5: paddr=000439cc vaddr=50000000 size=00010h (    16) load
I (168) boot: Loaded app from partition at offset 0x10000
I (168) boot: Disabling RNG early entropy source...
I (182) octal_psram: vendor id    : 0x00 (UNKNOWN)
I (182) octal_psram: dev id       : 0x03 (generation 4)
I (182) octal_psram: density      : 0x07 (256 Mbit)
I (187) octal_psram: good-die     : 0x01 (Pass)
I (192) octal_psram: Latency      : 0x01 (Fixed)
I (198) octal_psram: VCC          : 0x01 (3V)
I (203) octal_psram: SRF          : 0x00 (Slow Refresh)
I (209) octal_psram: BurstType    : 0x01 ( Wrap)
I (214) octal_psram: BurstLen     : 0x03 (1024 Byte)
I (220) octal_psram: Readlatency  : 0x07 (20 cycles@Fixed)
I (226) octal_psram: DriveStrength: 0x03 (1/8)
W (231) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
W (242) MSPI Timing: tuning fail, best point is fallen back to index 1
I (248) esp_psram: Found 32MB PSRAM device
I (253) esp_psram: Speed: 80MHz
W (257) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 31MB is mapped
I (266) cpu_start: Pro cpu up.
I (270) cpu_start: Starting app cpu, entry point is 0x40375354
0x40375354: call_start_cpu1 at C:/espressif/frameworks/esp-idf-v5.0/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: A

Icarus113, 2023-06-05:
May I know which IDF version are you using? And do you know the chip integrated? Is it ESP32-S3R2 or ESP32-S3R8?

suniltop, 2023-06-05:
Hi Icarus,

i am using branch: release/5.0  commit:7f5ecbe533b2666df6b10658a023b5f637320696
But i am not sure, which versions chip i use, is there any way how to sure =

`$ git log
commit 7f5ecbe533b2666df6b10658a023b5f637320696 (HEAD, tag: v5.0)
Merge: 5afd0f5a3d 277e4d5471
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Thu Dec 1 20:37:52 2022 +0800

    Merge branch 'bugfix/wafer_version_minor_s3_v5.0' into 'release/v5.0'

    esp32s3: fixed bug chip v0.0 detected as vX.8 (v5.0)

    See merge request espressif/esp-idf!21351
`

suniltop, 2023-06-05:
Hi,

I have made a tests with different line mode already but no success.
I really wonder that is my module is fake or orginal because i cant find any ESP32-S3 module with 256MBit PSRAM on Espressif documents and website

Also what does it mean **WRT the chip** ?

Icarus113, 2023-06-06:
Hi @suniltop ,

> because i cant find any ESP32-S3 module with 256MBit PSRAM on Espressif documents and website

I suspect that this is because your branch doesn't have this [fix](https://github.com/espressif/esp-idf/commit/753ed2697fd5c3efb09998e38151175ba62a5ae1). 
```
I (182) octal_psram: vendor id : 0x00 (UNKNOWN)
I (182) octal_psram: dev id : 0x03 (generation 4)
I (182) octal_psram: density : 0x07 (256 Mbit)
I (187) octal_psram: good-die : 0x01 (Pass)
I (192) octal_psram: Latency : 0x01 (Fixed)
I (198) octal_psram: VCC : 0x01 (3V)
I (203) octal_psram: SRF : 0x00 (Slow Refresh)
I (209) octal_psram: BurstType : 0x01 ( Wrap)
I (214) octal_psram: BurstLen : 0x03 (1024 Byte)
I (220) octal_psram: Readlatency : 0x07 (20 cycles@Fixed)
I (226) octal_psram: DriveStrength: 0x03 (1/8)
```
It fails to detect the chip because the vendor id is an unknown one. But because the lack of the fix I mentioned, the program still executes without returning an error immediately. So the `256 Mbit` is not valid.

> Also what does it mean WRT the chip ?

Sorry for the misleading, I was referring to the IC package (contains the SoC) embedded on the module (e.g. ESP32-S3-WROOM-1U-N8). The ESP32S3 uses SiP PSRAM so the PSRAM is integrated into the package. So

- ESP32-S3-WROOM-1U-N8, the chip on this module is without the PSRAM.
- ESP32-S3-WROOM-1U-N8R2, the chip on this module is with a quad PSRAM.
- ESP32-S3-WROOM-1U-N8R8, the chip on this module is with an octal PSRAM.

Besides, you might need to check that the IO 26-37 are not multiplexed as these pins might influence the communication to/from the PSRAM.

suniltop, 2023-06-06:
Hi @Icarus113 

Firstly, I have updated esp-idf to 5.0.2 via espressif ide and then made a same tests.

> ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375adc
0x40375adc: esp_restart_noos at C:/espressif/frameworks/esp-idf-v5.0.2/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

>SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x17bc
load:0x403c9700,len:0xd7c
load:0x403cc700,len:0x300c
entry 0x403c992c
I (29) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (29) boot: compile time 13:28:44
I (29) boot: chip revision: v0.1
I (32) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32s3: Boot SPI Speed : 80MHz
I (42) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 16MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=099fch ( 39420) map
I (109) esp_image: segment 1: paddr=00019a24 vaddr=3fc91e00 size=027a0h ( 10144) load
I (113) esp_image: segment 2: paddr=0001c1cc vaddr=40374000 size=03e4ch ( 15948) load
I (123) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1a154h (106836) map
I (144) esp_image: segment 4: paddr=0003a17c vaddr=40377e4c size=09ee8h ( 40680) load
I (158) boot: Loaded app from partition at offset 0x10000
I (158) boot: Disabling RNG early entropy source...
**_E (169) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (170) cpu_start: Failed to init external RAM!_**

>abort() was called at PC 0x4037540d on core 0
0x4037540d: call_start_cpu0 at C:/espressif/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:394 (discriminator 3)

>Backtrace: 0x40375e6e:0x3fceb220 0x4037a881:0x3fceb240 0x4037fc26:0x3fceb260 0x4037540d:0x3fceb2d0 0x403cdb56:0x3fceb330 0x403cdf5d:0x3fceb380 0x403c9995:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED
0x40375e6e: panic_abort at C:/espressif/frameworks/esp-idf-v5.0.2/components/esp_system/panic.c:423

So, i think module has not got any in package psram, am i right ? 
if yes, when if i disable psram from sdkconfig at project then it said that 2MB PSRAM found so i am confused.

> ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x17bc
load:0x403c9700,len:0xd7c
load:0x403cc700,len:0x300c
entry 0x403c992c
I (25) boot: ESP-IDF v5.0.2-dirty 2nd stage bootloader
I (25) boot: compile time 13:28:44
I (25) boot: chip revision: v0.1
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : QIO
I (42) boot.esp32s3: SPI Flash Size : 16MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0948ch ( 38028) map
I (104) esp_image: segment 1: paddr=000194b4 vaddr=3fc91200 size=02358h (  9048) load
I (108) esp_image: segment 2: paddr=0001b814 vaddr=40374000 size=04804h ( 18436) load
I (119) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=19a8ch (105100) map
I (139) esp_image: segment 4: paddr=00039ab4 vaddr=40378804 size=089e8h ( 35304) load
I (152) boot: Loaded app from partition at offset 0x10000
I (152) boot: Disabling RNG early entropy source...
I (163) cpu_start: Pro cpu up.
I (164) cpu_start: Starting app cpu, entry point is 0x4037518c
0x4037518c: call_start_cpu1 at C:/espressif/frameworks/esp-idf-v5.0.2/components/esp_system/port/cpu_start.c:141

>I (0) cpu_start: App cpu up.
I (178) cpu_start: Pro cpu start user code
I (178) cpu_start: cpu freq: 240000000 Hz
I (178) cpu_start: Application information:
I (181) cpu_start: Project name:     hello_world_S3
I (186) cpu_start: App version:      1
I (191) cpu_start: Compile time:     Jun  6 2023 14:23:32
I (197) cpu_start: ELF file SHA256:  dab247f56fcbda81...
I (203) cpu_start: ESP-IDF:          v5.0.2-dirty
I (208) cpu_start: Min chip rev:     v0.0
I (213) cpu_start: Max chip rev:     v0.99 
I (218) cpu_start: Chip rev:         v0.1
I (223) heap_init: Initializing. RAM available for dynamic allocation:
I (230) heap_init: At 3FC93F78 len 00055798 (341 KiB): DRAM
I (236) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (243) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (249) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (256) spi_flash: detected chip: gd
I (260) spi_flash: flash io: qio
I (264) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!
**_This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision v0.1, 16MB external flash
16MB external PSRAM_**
Minimum free heap size: 391384 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...

Icarus113, 2023-06-06:
> when if i disable psram from sdkconfig at project then it said that 16MB PSRAM found so i am confused.

> This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision v0.1, 16MB external flash
16MB external PSRAM

Hmmm this is very weird. As I don't remember we have a log in the hello_world example for the PSRAM related info. See [this hello_world_main.c](https://github.com/espressif/esp-idf/blob/5181de8ac5ec5e18f04f634da8ce173b7ef5ab73/examples/get-started/hello_world/main/hello_world_main.c), it is the v5.0.2 one, and there's no PSRAM related logs.

suniltop, 2023-06-06:
Hi @Icarus113, 

I am sorry for this log that is my fault.
At this point what do you think ?
I think module does not has a any PSRAM thus, if yes, maybe we can close this issue.

Icarus113, 2023-06-07:
As you've tried with different line modes, I think this module possibly is ESP32-S3-WROOM-1U-N8 and this module is without PSRAM. 

Feel free to close this one and reopen it if there's any updates. @suniltop

closed 2023-06-07 by suniltop