espressif/esp-idf #11570: esp_hid_device example crash

swortain, 2023-06-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

run as the example code expected

### What is the actual behavior?

repeat crash after PC pairing with ESP32S3

### Steps to reproduce.

1. create a project in PlatformIO
2. copy all the file from src in esp_hid_device to src in PIO
3. burn the example and paired with PC

### Build or installation Logs.

```plain
# Serial LOG
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375ca9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1658
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2d9c
entry 0x403c9900
I (29) boot: ESP-IDF 5.0.2 2nd stage bootloader
I (29) boot: compile time 19:12:36
I (29) boot: chip revision: v0.1
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 8MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=23138h (143672) map
I (122) esp_image: segment 1: paddr=00033160 vaddr=3fc96500 size=035bch ( 13756) load
I (125) esp_image: segment 2: paddr=00036724 vaddr=40374000 size=098f4h ( 39156) load
I (137) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=7eafch (518908) map
I (230) esp_image: segment 4: paddr=000beb24 vaddr=4037d8f4 size=08bd8h ( 35800) load
I (246) boot: Loaded app from partition at offset 0x10000
[... 78 lines cut ...]
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 11.2.0+2022r1
 - toolchain-xtensa-esp32s3 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies       
Building in debug mode
Retrieving maximum program size .pio\build\ESP32-S3\firmware.elf
Checking size .pio\build\ESP32-S3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   9.2% (used 30052 bytes from 327680 bytes)  
Flash: [=======   ]  71.6% (used 751033 bytes from 1048576 bytes)
============================================================================================================== [SUCCESS] Took 26.36 seconds ==============================================================================================================
```

### More Information.

It's test on ESP32S3 with PlatformIO(v6.3.1)

esp-cjh, 2023-06-05:
Hi, @swortain 
What is the brand and model of the PC you paired with? Can the issue be reproduced with ESP-IDF?

swortain, 2023-06-05:
Hi, It's [ROG幻15 GU502 |](https://rog.asus.com.cn/laptops/rog-zephyrus/rog-zephyrus-m15-series/spec/). I'm not sure because I don't have ESP-IDF with esptool...but in theory platformIO is the same with origin ESP-IDF in my oponion?

esp-cjh, 2023-06-09:
Hi, @swortain Is this an inevitable question? Is it easy to reproduce on your end? I will give you a patch with debug print to find out the cause.

swortain, 2023-06-09:
>

swortain, 2023-06-09:
> 

sure，can you send me using email jianshan_54@126.com

zhp0406, 2023-11-17:
@swortain 
Is this issue still present in the latest version?

swortain, 2023-11-17:
en....I'm not sure, I reinstall my Windows and the crash disappeared....

closed 2023-11-17 by swortain