espressif/esp-idf #11573: Change USB mode programmatically.

sblantipodi, 2023-06-01:
### Is your feature request related to a problem?

Is it possible to change the USB Mode programmatically?

it would be cool to have a function like the ESP.restart() that works like
ESP.restartInCdcMode()
or
ESP.restartInTinyUsbMode()

Dazza0, 2023-06-20:
@sblantipodi Just to clarify, do you mean you want to change from USB Host mode to USB Device mode (or vice versa) at run time?

sblantipodi, 2023-06-20:
Let's use the Arduino IDE words...
ESP32-S3 has 2 major `USB MODE`:

- USB-OTG (TinyUSB)
- Hardware CDC and JTAG

Both modes let us read and print on Serial.
Serial.read() and Serial.write()

Web Tools installers needs CDC/JTAG to work properly, some firmware like `Luciferin` needs TinyUSB to avoid problems with ISR and RMT ISR 
https://github.com/espressif/arduino-esp32/issues/8284#issuecomment-1587014409

this is why I think that a function that permit to change the USB mode programmatically is really needed.

roma-jam, 2025-06-25:
@sblantipodi 

> How can I restart/switch programmatically from TinyUSB to USB-Serial-JTAG and viceversa?

You need to configure USB PHY back to the `USB_PHY_CTRL_SERIAL_JTAG` controller for that. 

Related topics: 
- https://github.com/espressif/esp-idf/issues/15912
- https://github.com/espressif/esp-idf/pull/15932