espressif/esp-idf #11577: [idf.py monitor] stopped working after trying out GDBStubs, prints OKOKOK

chipweinberger, 2023-06-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Dev Kit C

### Power Supply used.

USB

### What is the expected behavior?

The monitor should show the logs.

### What is the actual behavior?

- The monitor does not output any logs.
- addresses (e.g. 0x4038324e) are still translated by the monitor though (weird)
- the monitor logs this weird message when trying to quit with ctrl-c
- VSC shows the logs, but just printf `OKOKOKOK` when you try to interact over the console

```
^C^]/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Volumes/User/MBP-Google-Drive/jamcorder/firmware/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin/xtensa-esp-elf-gdb-3.11.ulvx4gr
^[warning: Invalid baud rate 460800.  Maximum value is 230400.
/dev/cu.usbserial-10: Invalid argument.
Quit
Quit
Quit

```

**idf.py monitor**
[image]

**VSC**
[image]

### Steps to reproduce.

1. enable GDBStubs at runtime, CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME=y
2. launch idf.py monitor
3. try to use the GDBStubs, though im not sure I ever got it working
4. try to exit the terminal with `ctlr-c`, `ctrl-[`, etc, nothing worked. I had to kill the entire Terminal
5. set `CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME=n`, and reflash
6. monitor is still broken!

chipweinberger, 2023-06-02:
The problem was `CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME=n`.

You need to use `# CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME is not set`

@igrr, can ESP-IDF replace all it's uses of `#ifdef` with `#if`? When you set something to `=y` in `sdkconfig.defaults`, setting it to `=n` in `sdkconfig.prod` will not work correctly.

igrr, 2023-06-02:
> The problem was CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME=n.

> You need to use # CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME is not set

Both =n and "is not set" are accepted in sdkconfig files. Could you please describe the steps to reproduce the issue? (Keep in mind that adding an option to sdkconfig.defaults when sdkconfig has already been generated doesn't change anything in sdkconfig.)

> can ESP-IDF replace all it's uses of #ifdef with #if? If a setting it set to =y in sdkconfig.defaults, setting it to =n in sdkconfig.prod will not work correctly.

Could you please expand that thought a bit? Are you suggesting that your sdkconfig.h file has an option macro defined while the same option is set to =n in sdkconfig file?

chipweinberger, 2023-06-02:
**Repro steps:**

1. set `CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME=n` in sdkconfig
2. build, flash, run

**Explanation:**

If you ctrl-f for CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME, you can see ESP-IDF mixes `#ifdef` and `#if` for the config, so feature get turned halfway on.

igrr, 2023-06-02:
> so =n turns the feature halfway on ('n' still counts as defined).

That is definitely not the case. `sdkconfig.h` (note `.h`!) file only contains defines for the options for which are enabled. No matter if you put `=n` or `not set` into `sdkconfig`, the corresponding option won't be defined in `sdkconfig.h`.

(If you see that happening, please attach both sdkconfig and sdkconfig.h, we'll take a look.)

> set CONFIG_ESP_SYSTEM_GDBSTUB_RUNTIME=n in sdkconfig

This option is a `switch`, so you can't disable it by setting it to `n`. Instead you need to enable another option from the same switch, e.g. `CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT=y`.

chipweinberger, 2023-06-02:
I forgot to mention I am using this PR: https://github.com/espressif/esp-idf/pull/11569

> That is definitely not the case.

If so, then I have no explanation for what fixed it. Perhaps something about flashing it again. Either way, this was a real issue I hit after enable GDBStubs.

igrr, 2023-06-02:
I think we'll close the issue for the lack of the way to reproduce it. If you run into something like that again, please keep sdkconfig and sdkconfig.h so that we can investigate. Otherwise it's pretty hard to tell what was the cause of the issue.

closed 2023-06-02 by chipweinberger