espressif/esp-idf #11578: idf-5.02 errors on macos Ventura 13.2 Apple M1 for esp32c6

kkimes2020, 2023-06-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-823-g903af13e84

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Expect "idf.py build" in my "hello-world" subdirectory of $HOME/esp/ to complete without error.

### What is the actual behavior?

Exits with error.

### Steps to reproduce.

1. Install esp-idf for esp32c6 per the programming guide on espressif website.  This completes without error.
2. Type ". ./export.sh" in the $HOME/esp/esp-idf directory.
3. In $HOME/esp "mkdir hello"
4. "cp -r esp-idf/examples/hello-world hello-world"
5. "idf.py set-target esp32c6" no errors
6. "idf.py build" terminates with error cut and pasted above.
7. 
[idf_py_stdout_output_67580.txt](https://github.com/espressif/esp-idf/files/11631782/idf_py_stdout_output_67580.txt)
[idf_py_stderr_output_67580.txt](https://github.com/espressif/esp-idf/files/11631785/idf_py_stderr_output_67580.txt)

### Build or installation Logs.

```plain
[idf_py_stdout_output_67580.txt](https://github.com/espressif/esp-idf/files/11631782/idf_py_stdout_output_67580.txt)
[idf_py_stderr_output_67580.txt](https://github.com/espressif/esp-idf/files/11631785/idf_py_stderr_output_67580.txt)
```

### More Information.

Have also, tried installing using VSCode and Eclipse similar or more confusing behavior occurred.

dobairoland, 2023-06-02:
I copy here the relevant part of the logs.

```
ImportError: dlopen(/Users/usr/.espressif/python_env/idf5.2_py3.12_env/lib/python3.12/site-packages/_cffi_backend.cpython-312-darwin.so, 0x0002): symbol not found in flat namespace '_ffi_type_double'
thread '<unnamed>' panicked at 'Python API call failed', /Users/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pyo3-0.18.3/src/err/mod.rs:790:5
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Traceback (most recent call last):
  File "/Users/usr/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 25, in <module>
    from cryptography import x509
  File "/Users/usr/.espressif/python_env/idf5.2_py3.12_env/lib/python3.12/site-packages/cryptography/x509/__init__.py", line 7, in <module>
    from cryptography.x509 import certificate_transparency
  File "/Users/usr/.espressif/python_env/idf5.2_py3.12_env/lib/python3.12/site-packages/cryptography/x509/certificate_transparency.py", line 11, in <module>
    from cryptography.hazmat.bindings._rust import x509 as rust_x509
pyo3_runtime.PanicException: Python API call failed
ninja: build stopped: subcommand failed.
```

Can you run the following please and see if it reveals anything else:
```
RUST_BACKTRACE=1 idf.py build
```

Also please share the output of `python -m pip freeze`.

kkimes2020, 2023-06-02:
I ran that.  Here is the stdout log, stderr is empty.

dobairoland, 2023-06-05:
Please try to remove the content of `/Users/usr/.espressif/python_env/idf5.2_py3.12_env` and re-run the install scripts. Maybe that will help.

kkimes2020, 2023-06-06:
I didn't notice the python -m pip freeze

03:18[hello]> python -m pip freeze

bitstring==4.0.2

CacheControl==0.13.0

certifi==2023.5.7

cffi==1.15.1

charset-normalizer==3.1.0

click==8.1.3

colorama==0.4.6

construct==2.10.68

contextlib2==21.6.0

cryptography==41.0.1

ecdsa==0.18.0

esp-coredump==1.5.1

esp-idf-kconfig==1.1.0

esp-idf-monitor==1.0.4

esp-idf-size==0.3.1

esptool==4.6.1

filelock==3.12.0

freertos-gdb==1.0.2

future==0.18.3

idf-component-manager==1.2.3

idna==3.4

kconfiglib==14.1.0

msgpack==1.0.5

packaging==23.1

pycparser==2.21

pyelftools==0.29

pygdbmi==0.9.0.2

pyparsing==3.0.9

pyserial==3.5

PyYAML==6.0

reedsolo==1.7.0

requests==2.31.0

requests-file==1.5.1

requests-toolbelt==1.0.0

schema==0.7.5

six==1.16.0

tqdm==4.65.0

urllib3==1.26.16

03:19[hello]> rm -rf /Users/usr/.espressif/python_env/idf5.2_py3.12_env

03:20[hello]> RUST_BACKTRACE=1 idf.py build

Please use idf.py only in an ESP-IDF shell environment.

03:20[hello]>

kkimes2020, 2023-06-06:
Ran ./install.sh esp32c3 esp32c6
Then, closed terminal.  Opened new terminal emulator and ran . ./export.sh
I was concerned about the state of environment.
In /esp/hello/, ran
idf.py clean
then,
idf.py set-target esp32c6
RUST_BACKTRACE=1 idf.py build
[idf_py_stdout_output_44956.txt](https://github.com/espressif/esp-idf/files/11659669/idf_py_stdout_output_44956.txt)

kkimes2020, 2023-06-07:
In $HOME/esp/hello/,
idf.py clean
idf.py set-target esp32c3
RUST_BACKTRACE=full idf.py build
[idf_py_stdout_output_55139.txt](https://github.com/espressif/esp-idf/files/11671285/idf_py_stdout_output_55139.txt)

dobairoland, 2023-06-08:
Please try the following and share the output from the terminal (not the files which are missing some information):

```
. $IDF_PATH/export.sh
python -m pip uninstall cryptography cffi
python -m pip --no-cache-dir install cryptography cffi
cd your_project
idf.py build
```

kkimes2020, 2023-06-09:
I cloned ninja from github.com, built it and copied it to /usr/local/bin.  My "idf.py build" for esp32c3 and esp32c6 now completes without error.

[idf_py_stdout_50069.txt](https://github.com/espressif/esp-idf/files/11701978/idf_py_stdout_50069.txt)

closed 2023-06-09 by kkimes2020

dobairoland, 2023-06-09:
I'm happy that your issue has been solved. 

But honestly I don't understand how ninja and cryptography relates to each other. To me, the issue looked liked as a broken cryptography wheel linked to libraries not present on your computer. 

Anyway, this seems to be an isolated issue and no further action is required.

kkimes2020, 2023-06-11:
I don't understand it at all.  But, then I don't understand a lot.  Like just about everything "freedesktop.org" aka Redhat has done since the 1990's.  Dbus, gvfs, NetworkManager, LoginMaager, systemd...  I don't use homebrew anymore because, it creates conflicts after a short while.  Anaconda either.

I don't understand what RUST_BACKTRACE=1, is.  Is there Rust language code in esp-idf?  

I really don't want to have to have gigabytes of binaries in large, complex ".rustup" and ".cargo" directories in order to maintain things.  Tools downloading binaries to a machine is scary and rust does it insecurely.   Aside from rust's only advantage being manual memory management disguised as new theory and complex constructs in the language.  But, when do we get to pick things, these days?

With esp-idf I understand that I need crosscompilers but, it gets really hard to know what's going on with all these python build tools on top of ninja with cmake.

kkimes2020, 2023-06-12:
Thanks for your time and help.