espressif/esp-idf #11585: add an esp-idf example for conserving power with the esp32-c6 using Wi-Fi 6 TWT after deepsleep

roblatour, 2023-06-03:
### Is your feature request related to a problem?

would like to minimize power consumption for battery supplied projects

### Describe the solution you'd like.

In the testing of my esp32-c6 the wifi-6 connection would only stay connected over the long term if the board went into automatic light sleep, which uses a bunch of current, for the entire sleep period - which is a far cry from what the people are talking about here: https://www.youtube.com/watch?v=NaoozZDanuM&t=105s

The existing itwt example does not illustrate how to initially set the Wi-Fi 6 connection, go into deep sleep for a given period of time, wake up and be able to use the Wi-Fi 6 connection without needing to reconnect. 

As such, I would like to see an example where the esp32-c6 initially sets the Wi-Fi 6 connection, go into deep sleep for a given period of time (for example 15 minutes), wake up and be able to use the Wi-Fi 6 connection without needing to reconnect. 

### Describe alternatives you've considered.

I have coded three other alternatives:

1. using deepsleep and doing a full start and connect when the esp32 wakes up,
2.  using automatic light sleep, and
3.  using manual light sleep.

These are described https://github.com/roblatour/WeatherStation with the power stats and the source code also available from that same site.

In support of this, I have also written a TWT calculator here:
https://github.com/roblatour/TWTWakeDurationCalculator

igrr, 2023-06-04:
Could you please describe how you are measuring the current consumption of ESP32-C6? The current consumption you are measuring in automatic light sleep mode is significantly higher than expected. This might indicate that either there is something preventing the system from staying in low-power mode for a long enough period, or you are measuring the total current consumption of the development board, not just that of the C6 module.

As one more data point, could you please try measuring the current consumption (using the same method) of the itwt example in IDF? Then we can compare the numbers and that might help narrow down the issue.

As to your original request: in deep sleep mode, Wi-Fi connection cannot be preserved. ESP32-C6 has been designed to preserve Wi-Fi connection in light sleep mode. With all the current consumption optimizations the current consumed in light sleep mode should be around 35 uA. Average current consumption while maintaining Wi-Fi connection will depend on the negotiated wake-up interval.

To conclude, the feature request will probably be closed as "won't do", but we can try to help you troubleshoot the high current consumption of C6 in automatic light sleep mode that you are seeing.

roblatour, 2023-06-04:
To measure the power consumption I am using a I am using a Nordic Power Profiler II
https://www.nordicsemi.com/Products/Development-hardware/Power-Profiler-Kit-2

The numbers reported on my github page is the average for the hour.  So that includes the peak power usage for a few seconds from when the esp32 is connected and doing work (which alone averages out at about 65 mA in that short time) , along with the much longer sleep time (which for example averages itself alone at about 891 uA).  Here's a couple screen shots:  https://ibb.co/3Ngdcrs https://ibb.co/vPnWp5b  to help illustrate.

If you want you're welcome to download the capture files. I've uploaded them here: https://6ec1f0a2f74d4d0c2019-591364a760543a57f40bab2c37672676.ssl.cf5.rackcdn.com/power_monitoring_results.zip

I will measure the itwt example, but it will need to be much later today.

igrr, 2023-06-04:
> Also, of note, for these measurements I am powering the board via the 5v VIN pin on the board. 

I would recommend measuring current across the removable jumper located next to the module. That excludes the current used by the usb-serial bridge chip as well as the quiescent current of the LDO. (I think there are plenty of LDOs with much lower quiescent current than the one used on the C6 devkit.)

> it gives you the ability to turn the device off and wake it back up and still be connected

In principle, a Wi-Fi 6 client could indeed "turn itself off", and later wake up to receive the beacon. However for practical reasons ESP32-C6 doesn't support Wi-Fi connection persistence in deep sleep. There are a few reasons for this. First, in deep sleep the chip can retain much less memory than in light sleep, which makes preserving the state of the software stack challenging. Second is that deep sleep wake-up on the C6 can easily take several hundred milliseconds, and the time depends on various factors: program size, Flash frequency, and so on. Compensating for this wake-up time becomes pretty difficult. Finally, the programming model would be more complicated: deciding when it is possible to enter deep sleep and for what interval is difficult in the user application, since the Wi-Fi stack state needs to be taken into account. On the other hand, if the system would enter deep sleep automatically (as it does with light sleep) then the program would unexpectedly lose its state (deep sleep reset is very similar to power on reset.)

If everything is configured properly, you should see around 35 uA current consumption during light sleep intervals. Currently the current is around 900uA, a fair bit more than expected.

roblatour, 2023-06-04:
Thank you.

Regarding:

> I would recommend measuring current across the removable jumper located next to the module

There are two pins on that jumper.  Is the one closest to the middle of the board the + pin, or is the + pin the one closest to the edge of the board?   Also, is the other ground, or do I use a ground pin on the edge of the board?

Regarding:

> "for practical reasons ESP32-C6 doesn't support Wi-Fi connection persistence in deep sleep

thank you, that clarifies things quite a bit.  I was chasing that as really the main way to save power using the chip.  I guess I focus on using the pins you suggest and seeing how that works?

igrr, 2023-06-04:
The jumper is between the 3.3V rail and the 3V3 pin of the module. You can find the schematics and the layout [at the bottom of this page](https://espressif-docs.readthedocs-hosted.com/projects/esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html#current-measurement). The pin closer to the side of the board is connected to the module, the one closer to the LDO is connected to the LDO.

roblatour, 2023-06-04:
so can I power it by supplying the positive of 3.3v power supply to the jumper pin closest to the LDO, and the  negative of the power supply to one of the ground pins on the edge of the esp32-c6 board?

roblatour, 2023-06-05:
ok - knowing now that the ESP32-C6 doesn't support Wi-Fi connection persistence in deep sleep, this issue can be closed.

closed 2023-06-05 by roblatour