espressif/esp-idf #11587: Faulty includes in core.h

HeCoding180, 2023-06-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF version: v5.0.2

The "esp-idf-v5.0.2/components/xtensa/esp32s3/include/xtensa/config/core.h" file contains the following faulty includes:
"../hal.h"
"../xtensa-versions.h"

After going through the files a bit I found another directory, which seems like it might be the right one:
"esp-idf-v5.0.2/components/xtensa/include/xtensa"
This directory contains the "hal.h" file as well as the "xtensa-versions.h"

zikalino, 2023-06-05:
@HeCoding180 do you actually encounter any problem when building your application?

HeCoding180, 2023-06-08:
I have not tried that yet, since it's really annoying and atm, I'm having other issues

zikalino, 2023-10-18:
@HeCoding180 I will close this ticket for now. Please reopen if you have more information / still experiencing any issue related to this.

closed 2023-10-18 by zikalino