espressif/esp-idf #11591: console: "help <command>" in addition to plain "help"

crackwitz, 2023-06-05:
### Is your feature request related to a problem?

I have a lot of commands. Calling "help" confronts the user with a listing several pages long.

### Describe the solution you'd like.

I'd like `help_command(int argc, char **argv)` to accept an optional first argument.

Given that first argument, the `help` command will print just that command's `help` string.

### Describe alternatives you've considered.

I've considered implementing `help` myself but that'll probably have no access to the list of registered commands and their argtables. Tthe `s_cmd_list` in `help_command` in `components/console/commands.c` isn't exposed.

### Additional context.

I'd be willing to propose the code for `help_command`. I don't anticipate it being more than a fistful of lines. It'd merely have to filter the iteration, skipping each command that doesn't match the query.

Would be nice if prefix/infix matches were allowed. `strstr()` seems useful.

igrr, 2023-06-05:
@crackwitz sounds like a good idea! You are welcome to open a pull request.

jnorgan, 2023-07-24:
I also had this same need and had done something similar. Not doing any wildcarding, but for my purposes, full command name was sufficient.

crackwitz, 2023-07-25:
I hope to find some time to get this done tomorrow.

jnorgan, 2023-07-25:
@crackwitz Here's the PR I submitted: https://github.com/espressif/esp-idf/pull/11945

jnorgan, 2023-08-25:
Commit https://github.com/espressif/esp-idf/commit/0b595f3c77d52a334e80d918a08040982bb60cfc seems to resolve this issue.

ESP-Marius, 2023-08-29:
As already mentioned, the merged PR seems to address this.

Closing the issue for now, feel free to re-open if you feel there is still something left unresolved.